beegee-tokyo/SX126x-Arduino #3: SX126x-Arduino + Ethernet

brolly759, 2019-10-25:
I have successfully used your library to communicate 2 modules together but seem to have an issue when I try to use Ethernet at the same time. 

Here is the Ethernet, ESP32 I am using:
https://www.amazon.com/Olimex-ESP32-POE-ISO-Espressif-Ethernet-Isolation/dp/B07R1ZMSG7/ref=sr_1_2?keywords=esp32+poe&qid=1572020502&sr=8-2

From what I can tell I need to use a different SPI pins, here is my configuration:

// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 5;  // LORA RESET
int PIN_LORA_NSS = 15;  // LORA SPI CS
int PIN_LORA_SCLK = 0;  // LORA SPI CLK
int PIN_LORA_MISO = 2;  // LORA SPI MISO
int PIN_LORA_DIO_1 = 32; //21; // LORA DIO_1
int PIN_LORA_BUSY = 16; //22;  // LORA SPI BUSY
int PIN_LORA_MOSI = 4;  // LORA SPI MOSI
int RADIO_TXEN = -1; // 26;   // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;  //27;   // LORA ANTENNA RX ENABLE

What I would really like to test is to remove somehow all the events / IRQ's and just check for a message coming in manually. 

Side note, would be nice if this was ported to AVR also :)

brolly759, 2019-10-25:
to update, I got them both to work but I do see some timing issues. My original issue was int RADIO_TXEN and RADIO_RXEN set to 26,27 instead of -1. Now I just need to figure out the timing issue. Here is my code:
`
#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT
#define ETH_PHY_POWER 12

#include <Arduino.h>
#include <SX126x-Arduino.h>
#include <SPI.h>
#include <ETH.h>

static bool eth_connected = false;

hw_config hwConfig;
/*
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 4;  // LORA RESET
int PIN_LORA_NSS = 3;	// LORA SPI CS
int PIN_LORA_SCLK = 18;  // LORA SPI CLK
int PIN_LORA_MISO = 19;  // LORA SPI MISO
int PIN_LORA_DIO_1 = 32; //21; // LORA DIO_1
int PIN_LORA_BUSY = 33; //22;  // LORA SPI BUSY
int PIN_LORA_MOSI = 23;  // LORA SPI MOSI
int RADIO_TXEN = 2; // 26;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = 1;  //27;	 // LORA ANTENNA RX ENABLE

*/

// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 5;  // LORA RESET
int PIN_LORA_NSS = 15;  // LORA SPI CS
int PIN_LORA_SCLK = 0;  // LORA SPI CLK
int PIN_LORA_MISO = 2;  // LORA SPI MISO
int PIN_LORA_DIO_1 = 32; //21; // LORA DIO_1
int PIN_LORA_BUSY = 16; //22;  // LORA SPI BUSY
int PIN_LORA_MOSI = 4;  // LORA SPI MOSI
int RADIO_TXEN = -1; // 26;   // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;  //27;   // LORA ANTENNA RX ENABLE

// Function declarations
void OnTxDone(void);
void OnRxDone(uint8_t *payload, uint16_t size, int16_t rssi, int8_t snr);
void OnTxTimeout(void);
void OnRxTimeout(void);
void OnRxError(void);
void OnCadDone(bool cadResult);

//#define LED_BUILTIN 

// Define LoRa parameters
#define RF_FREQUENCY 915000000  // Hz
#define TX_OUTPUT_POWER 22		// dBm
#define LORA_BANDWIDTH 0		// [0: 125 kHz, 1: 250 kHz, 2: 500 kHz, 3: Reserved]
#define LORA_SPREADING_FACTOR 7 // [SF7..SF12]
#define LORA_CODINGRATE 1		// [1: 4/5, 2: 4/6,  3: 4/7,  4: 4/8]
#define LORA_PREAMBLE_LENGTH 8  // Same for Tx and Rx
#define LORA_SYMBOL_TIMEOUT 0   // Symbols
#define LORA_FIX_LENGTH_PAYLOAD_ON false
#define LORA_IQ_INVERSION_ON false
#define RX_TIMEOUT_VALUE 3000
#define TX_TIMEOUT_VALUE 3000

#define BUFFER_SIZE 64 // Define the payload size here

static RadioEvents_t RadioEvents;
static uint16_t BufferSize = BUFFER_SIZE;
static uint8_t RcvBuffer[BUFFER_SIZE];
static uint8_t TxdBuffer[BUFFER_SIZE];
static bool isMaster = false;
const uint8_t PingMsg[] = "PING";
const uint8_t PongMsg[] = "PONG";
[... 358 lines cut ...]
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}`

beegee-tokyo, 2019-10-31:
One problem could be in    
```cpp
void testClient(const char * host, uint16_t port)
```
You might need to move this to a separate task. Because while you are connecting to the client, the main loop() is blocked and the LoRa events are not handled.

Either put the LoRa stuff into a independant running task or the ETH stuff.

beegee-tokyo, 2019-10-31:
Regarding AVR port, I use the Ticker library for timeouts and this is not available for AVR platform as far as I know.

brolly759, 2019-10-31:
You can close this for now, I changed the code a little and it seems to be working now so its not a hardware issue, I just need to code it better .

beegee-tokyo, 2019-11-01:
Good.

closed 2019-11-01 by beegee-tokyo