beegee-tokyo/SX126x-Arduino #4: SX126x

laughingboyjj, 2019-10-31:
Hi , @beegee-tokyo I am using your lib, and I get better results in the initial stage. However, in the function of using CAD, and the official document has always been the error of packet loss, do you have time to write a CAD routine? Thank you very much

beegee-tokyo, 2019-10-31:
@laughingboyjj     
The [PingPong ](https://github.com/beegee-tokyo/SX126x-Arduino/tree/master/examples/PingPong)example uses CAD to determine if the channel is free before sending.

laughingboyjj, 2019-10-31:
I use the CAD function to start the receive mode when the CAD Detect is generated; the relevant parameter is LORA_SYMBOL_TIMEOUT, when set to 0, it can be received but the packet loss rate is very high; when the setting is greater than 0, Rxtimeout will be automatically generated; I have not engaged Good for this CAD function

beegee-tokyo, 2019-10-31:
You can set LORA_SYMBOL_TIMEOUT with 
```cpp
	/*!
 * \brief Set the number of symbol the radio will wait to validate a reception
 *
 * \param [in]  SymbNum          number of LoRa symbols
 */
	void SX126xSetLoRaSymbNumTimeout(uint8_t SymbNum);
```

laughingboyjj, 2019-10-31:
of course! This applies to the receiver, waiting for the correct symbol delay to receive the problem, if the setting is greater than 0, it should be more stable, but the result is rxtimeout

laughingboyjj, 2019-10-31:
If you have time, please test the receiving mode of the CAD function of SX126x, not only me, I believe can help a lot of people.

beegee-tokyo, 2019-10-31:
I think the problem is how you start CAD. There are two CAD modes according to the SX1262 datasheet.    
**CAD_ONLY**    
- The chip performs the CAD operation in LoRa®. Once done and whatever the
activity on the channel, the chip goes back to STBY_RC mode.    

**CAD_RX**    
- The chip performs a CAD operation and if an activity is detected, it stays in RX until
a packet is detected or the timer reaches the timeout defined by
cadTimeout * 15.625 us    

How are you setting the CAD parameters. Maybe 
```cpp
SX126xSetCadParams(LORA_CAD_08_SYMBOL, LORA_SPREADING_FACTOR + 13, 10, 
					LORA_CAD_RX, RX_TIMEOUT);
SX126xSetDioIrqParams(IRQ_CAD_DONE | IRQ_CAD_ACTIVITY_DETECTED,
					IRQ_CAD_DONE | IRQ_CAD_ACTIVITY_DETECTED,
					IRQ_RADIO_NONE, IRQ_RADIO_NONE);
Radio.StartCad();
```
will work.
It setup CAD to go into RX mode after symbol detection (**LORA_CAD_RX**), starts the CAD, and goes into RX mode when symbols are detected and should trigger an interrupt after a package was received or timeout after (RX_TIMEOUT * 15.625) ms.

Sorry, cannot test right now.

laughingboyjj, 2019-10-31:
Anyway, thank you very much! I will continue, waiting for my good news.

beegee-tokyo, 2019-10-31:
I think I misunderstand what you are trying to do.

Is it correct that you:
Want to start CAD in CAD_RX mode so that if CAD detects symbols it goes into RX mode and receives that package. Then you want to read the received package from the SX126x.

Like a one-channel gateway would do?

laughingboyjj, 2019-10-31:
Yes, your guess is correct, but the 126x CAD function is unstable, I can't proceed to the next step.

beegee-tokyo, 2019-11-01:
@laughingboyjj 
Maybe you know this document already
https://www.semtech.com/uploads/documents/SX126X_CAD_performance_evaluation_V1.0.pdf
Quite interesting stuff about CAD

laughingboyjj, 2019-11-01:
Yes, I have seen it! According to the parameter settings inside, still failed to solve the problem

laughingboyjj, 2019-11-01:
My code is already on my My Repositories->LoRa. If you have time, you can take a look

beegee-tokyo, 2019-11-01:
I spent some time and tried to get more information from the datasheet.
It is a little bit confusing.
CAD is waiting for ??? amount of time for symbols. 
CAD can be set to two different exit modes
- CAD_ONLY => After CAD is done (with whatever result) the SX126x goes into STBY_RC mode 
- CAD_RX => After CAD is done and symbols were detected, it stays in RX mode, I guess to receive the package. But that is nowhere explained.    
So if I use CAD_ONLY, the CAD_DONE event is triggered when symbols are detected. But by the time I go into RX mode again to receive the package, it is too late, because some symbols are already lost, so the package will not be detected anymore.
And if I use CAD_RX, I would expect that after CAD_DONE event is done, a second event should be triggered after the complete package was received or a timeout occurred. But that is not explained anywhere as well. 

Looking into my library, I think what you are trying to do is not possible at the moment. I will have to detect the CAD mode and act different on CAD IRQs if it is CAD_ONLY or CAD_RX.

I saw as well that Semtech updated there example code. I need as well to go through these changes and update my sources accordingly.

laughingboyjj, 2019-11-01:
Thank you very much, thank you for providing these ideas; I will also do it at the same time.

beegee-tokyo, 2019-11-03:
I went through the Semtech updates and implemented them. But nothing is related to CAD.

I will raise a question regarding CAD_RX mode on the Semtech help forum tomorrow. Usually the guys are quite helpful. 

In the meantime (the Halloween holidays gave me some spare time) I adapted the [ESP-1ch-Gateway-v6.0](https://github.com/things4u/ESP-1ch-Gateway-v6.0) to work with the SX1262. It is limited to RX continuous, CAD and frequency hopping is not supported. Downlink is not working yet, could be a problem with RX window timing or the frequency used.

laughingboyjj, 2019-11-03:
These are all good news. I read the library functions on the official website these two days, and I'm still trying to solve the CAD problem

beegee-tokyo, 2019-11-04:
If you want to try, I published my gateway code. The application is based on the [ESP-1ch-Gateway-v6.0](https://github.com/things4u/ESP-1ch-Gateway-v6.0/tree/master/ESP-sc-gway) code.

You can find it in [SX1262-SC-GW](https://github.com/beegee-tokyo/SX1262-SC-GW)

laughingboyjj, 2019-11-04:
This is the cad routine code given by semtech official, you can refer to it, but I can't implement it well on your lib, I am researching you to do lib 
https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R000000HSjg/kmXX1tZ43BMdqACJW11_TwYQr0v8wVp6IhAeK0YfzR0

beegee-tokyo, 2019-11-04:
The Semtech cad routine (translated into Arduino to work with my library) did not work first. But studying it gave me some input. I think there are some critical parts.
- **```LORA_PREAMBLE_LENGTH 8```** MUST be the same as the sender.
- **```LORA_SYMBOL_TIMEOUT 2```** the lower I set this the less false detections I got
- after each triggered event they call **```Radio.Standby();```**

I changed my test code and it works now. It detects the channel activity and receives the packet. There are some false detections, but I think that is natural.

What is missing is to get the SF of the packet after CAD detected it. No idea how to do that.

Try attached code, I didn't change anything in the library itself.
What you need to check:
- lines 9 to 17 => adapt to your HW setup
- line 60 => set to the frequency you are using
- line 65 => set to the preample length you are using (LoRaWan packages seem to use 8)

[main.zip](https://github.com/beegee-tokyo/SX126x-Arduino/files/3803491/main.zip)

laughingboyjj, 2019-11-05:
I found two interesting phenomena last night. First, when cadResult is true, it will not receive, it will repeat the cycle of cadResult. Second, different sf of the same frequency can also produce cad_detect; maybe this chip The functionality of CAD is not as stable as the documentation says.

beegee-tokyo, 2019-11-05:
I didn't see the first problem. I receive the package 99% of the tiime when cadResult is true with the example code.

The second one is expected. Because the CAD should look for symbols independent of the SF of the incoming package. What I am missing is how I can determine which SF to use to receive. I will try to understand how [ESP-1ch-Gateway-v6.0](https://github.com/things4u/ESP-1ch-Gateway-v6.0/tree/master/ESP-sc-gway) is using CAD to receive packages coming on the same frequency but with different SF. There is a solution, because if I use the ESP-1ch-Gateway with a SX1272/RFM95 module it receives packages independant of the SF the sender uses. It somehow detects from CAD result the SF, sets the RX parameters to the expected SF and is able to receive.

laughingboyjj, 2019-11-05:
This information has given me a lot of help, thank you very much, I am different from you in the use of USE_DIO2_ANT_SWITCH, USE_DIO3_TCXO, USE_DIO3_ANT_SWITCH, I will continue to study, will tell you when the results come out

beegee-tokyo, 2019-11-05:
May I ask what module you are using?
I am using eByte E22-900M22S modules mainly, but I am expecting some Dorji DRF1962T modules soon. The eByte module is more difficult because you need 2 pins to control the antenna (which makes no sense to me). I hope the Dorji modules are easier to use.

laughingboyjj, 2019-11-05:
I am using a board drawn by the company's hardware engineers. I think using a board with uncertainties has caused me some troubles to some extent, but it doesn't matter. The chip uses the SX1261, and the hardware does not have TCXO.
[image]

beegee-tokyo, 2019-11-05:
Hmmm, interesting design. Looks same as the eByte module you are using 2 signals to control the RF antenna switch. You can use the **```RADIO_TXEN```** and **```RADIO_RXEN```** in my hwConfig to control them.
[image]

But if your schematic is correct, you are using **```DIO3```** as power supply for the TCXO, so **```hwConfig.USE_DIO3_TCXO```** should be set **```TRUE```**
[image]

laughingboyjj, 2019-11-05:
Yes, those I have set up, the reason is another point, if #define LORA_SYMBOL_TIMEOUT 0; my experimental phenomenon is acceptable, although occasionally CAD_Detect occurs in the endless, when it is greater than 0, it will continue to generate RxTimeout

laughingboyjj, 2019-11-05:
OK , I have solved this problem. When Radio.Rx(0), LORA_SYMBOL_TIMEOUT works. If it exceeds this set time, it will generate rxtimeout.

beegee-tokyo, 2019-11-05:
Good to know.

Good luck with your project!

laughingboyjj, 2019-11-07:
With your help, my project has made rapid progress; but I still have a point to discuss with you, is to enter the CAD_Detect receiving time, what is the basis for us to set up? I get a single channel gateway in single Rx mode, using LORA_SYMBOL_TIMEOUT=5 or 8, but the reception is obviously unstable; automatic 2S is too long, can not be compatible with Lpreamble=8 a frame of different sf data, I want to solve This question: I have tested a different sf data, I designed a process
[image]

beegee-tokyo, 2019-11-08:
I am not sure if I understand your question correct.

What I found is that **```SX126xSetCadParams```** has a parameter **```cadTimeout```**. But the description of this parameter is confusing.
In the datasheet it is explained as
```
The parameter cadTimeout is only used when the CAD is performed with cadExitMode = CAD_RX.  
Here, the cadTimeout indicates the time the device will stay in Rx following a successful CAD.
Rx Timeout = cadTimeout * 15.625
```
But in the Semtech example code it is explained as
```
/*!
 * \brief Sets the Channel Activity Detection (CAD) parameters
 *
 * \param [in]  cadSymbolNum   The number of symbol to use for CAD operations
 *                             [LORA_CAD_01_SYMBOL, LORA_CAD_02_SYMBOL,
 *                              LORA_CAD_04_SYMBOL, LORA_CAD_08_SYMBOL,
 *                              LORA_CAD_16_SYMBOL]
 * \param [in]  cadDetPeak     Limit for detection of SNR peak used in the CAD
 * \param [in]  cadDetMin      Set the minimum symbol recognition for CAD
 * \param [in]  cadExitMode    Operation to be done at the end of CAD action
 *                             [LORA_CAD_ONLY, LORA_CAD_RX, LORA_CAD_LBT]
 * \param [in]  cadTimeout     Defines the timeout value to abort the CAD activity
 */
	void SX126xSetCadParams(RadioLoRaCadSymbols_t cadSymbolNum, uint8_t cadDetPeak, uint8_t cadDetMin, RadioCadExitModes_t cadExitMode, uint32_t cadTimeout);
```
Not sure which one is correct. But you could try to change the **```cadTimeout```** parameter and see how it works.

laughingboyjj, 2019-11-08:
Ok, I uploaded new content in my LoRa repositories: CAD polls the unknown sf signal in sf7 and sf8, the test of sf7 and sf8 alone is OK, plus sf7 and sf8 switch will not work! In theory, it is successful, which makes me very collapse! When Radio.Rx(0) is set, the receiving time is determined by LORA_SYMBOL_TIMEOUT; attach a relational table reference。
[image]

laughingboyjj, 2019-11-08:
Haha, every day has a different progress, I abandoned CAD, directly received data with Rx, set a certain time to constantly query the unknown sf, now it looks good! But I found another problem, when receiving and sending sf11, there will be receiving errors. Have you ever encountered it? If you are convenient, you can email me, we will discuss with LoRa, in short, thank you!

beegee-tokyo, 2019-11-08:
I have never tried SF11 and specially not in the frequency you use.

What is your email address?

laughingboyjj, 2019-11-08:
You can contact me in time via my email: 717859718@qq.com, the simple receiving mode will encounter this problem, sf11 is receiving data error

laughingboyjj, 2019-11-08:
By the way, my name is richard, is a small engineer

closed 2019-11-19 by laughingboyjj

piyushy6, 2021-02-25:
Hi, 
I am using SX1262 for LoRa to LoRa. I am a bit confused about CAD, can I use Rx.StartCad() on the receiver end instead of Radio.SetRxDutyCycle to listen to preamble sent by Transmitter LoRa device?

beegee-tokyo, 2021-02-26:
Yes, you can use StartCad() to start receiving, but it is not the "normal" way.
StartCad is usually used when you want to send a packet, but need to make sure that the frequency channel is not occupied by another LoRa node.
Rx() and SetRxDutyCycle() are the usual functions used to start listening to incoming messages.

If you want to use StartCad() you have to change the 4th parameter `cadExitMode` to `LORA_CAD_RX` which will start receiving if StartCad() detects a preample. However, if a packet from another node is already in sending process, StartCad() might not be able to receive the packet.

rego21, 2023-07-20:
Hi @beegee-tokyo,

I'm trying to implement a Single Channel Gateway (SX1261) and have some question that you may help me to clarify :) 
So basically :
- The CAD_DONE IRQ is always being triggered even if the transmitter is not sending anything. Is this the expected behavior ? 
- The CAD_DETECT_ACTIVITY is triggered when a package is sent with the same SF as the gateway or the adjacent SF. For example if gateway is using SF7 and transmitter SF7  CAD_DETECT_ACTIVITY is triggered and the data can be received. if gateway is using SF7 and transmitter SF8  CAD_DETECT_ACTIVITY is triggered but you cant read the data.

The must confusing thing to me seems to be the CAD_DONE and from the datasheet:

> The length of the search is configured via the command SetCadParams(...). At the end of
the search period, the device triggers the IRQ CADdone if it has been enabled

Which I understand that CAD_DONE is triggered at the end of the search period defined by SetCadParams, so it would trigger the IRQ in 20ms for example. And CAD_DETECT_ACTIVITY is the one responsible for "finding" activity on that frequency band, however I cant find activity in all the SFs...

Thanks and hope you can help me understand better this :)

beegee-tokyo, 2023-07-20:
Hi Rego21,

I "played" once with a single channel gateway application, but since it is kind of "not appreciated" by LoRaWAN servers, I gave up on it.

For CAD, it is usually used in LoRa P2P applications and not with LoRaWAN. 
The LoRaWAN stack has a LBT (Listen before talk) functionality that is required by some LoRaWAN regions. This doesn't use CAD, instead it is checking RSSI levels on the channel that will be used.

For your other questions (getting activity on all SF), I think your setup of CAD is most likely wrong.

Check the definition of the setup function:
```cpp
/*!
 * @brief Set Channel Activity Detection parameters
 * 
 * \param   cadSymbolNum   The number of symbol to use for CAD operations
 *                             [LORA_CAD_01_SYMBOL, LORA_CAD_02_SYMBOL,
 *                              LORA_CAD_04_SYMBOL, LORA_CAD_08_SYMBOL,
 *                              LORA_CAD_16_SYMBOL]
 * \param   cadDetPeak     Limit for detection of SNR peak used in the CAD
 * \param   cadDetMin      Set the minimum symbol recognition for CAD
 * \param   cadExitMode    Operation to be done at the end of CAD action
 *                             [LORA_CAD_ONLY, LORA_CAD_RX, LORA_CAD_LBT]
 * \param   cadTimeout     Defines the timeout value to abort the CAD activity
 */
void RadioSetCadParams(uint8_t cadSymbolNum, uint8_t cadDetPeak, uint8_t cadDetMin, uint8_t cadExitMode, uint32_t cadTimeout);
```

In my P2P example I use the parameters that were setup before:
```cpp
Radio.SetCadParams(LORA_CAD_08_SYMBOL, LORA_SPREADING_FACTOR + 13, 10, LORA_CAD_ONLY, 0);
```

For more details how to use CAD please check the Semtech documentation. I implemented the basic functions and use it in the LoRa P2P examples, but I never went into the details.

rego21, 2023-07-20:
Hi again @beegee-tokyo , thanks for the quick reply.

The only way I found to get activity in all SFs was with the following settings:
`SX126xSetCadParams(LORA_CAD_02_SYMBOL, 17, 10, LORA_CAD_RX, ((2000 * 1000) / 15.625));` 
Where the gateway SF is set to 7 and BW125

And activity is detected even when data is not transmitted, however I can filter this by setting a limit to the received RSSI. Basically the gateway triggers CAD_DETECT_ACTIVITY when no data is being sent with RSSI of -104 and when a real package is being sent the RSSI is usually below -90.
Sadly  the datasheet is not very clear regarding CAD...

beegee-tokyo, 2024-07-12:
The data sheet says about CAD RX exit mode:

> The chip performs a CAD operation and if an activity is detected, it stays in RX until
a packet is detected or the timer reaches the timeout defined by
cadTimeout * 15.625 us

What timeout do you set when calling `SX126xSetCadParams(RadioLoRaCadSymbols_t cadSymbolNum, uint8_t cadDetPeak, uint8_t cadDetMin, RadioCadExitModes_t cadExitMode, uint32_t cadTimeout)` ?

I never used CAD with RX mode before, not sure how it works. I only used it to detect whether I can send a packet.