beegee-tokyo/SX126x-Arduino #5: NRF52810 support?

brolly759, 2019-10-31:
Is the NRF52810 supported or only the nrf52832 ?

beegee-tokyo, 2019-11-01:
N4F52810 is like a 52832 with less memory?
I don't have one, so I cannot test. But it should work.

beegee-tokyo, 2019-11-03:
Please let me know if it works.

brolly759, 2019-11-04:
I meant the nRF51822. It runs the nRF51 software stack, not nRF52. I dont have one either to test

beegee-tokyo, 2019-11-04:
Nope, nRF51 is not supported atm. Not sure if it is possible to make the library compatible as I have no clue how different the nRF51 and nRF52 are from software side.     

Right now I have no time to check. Sorry.

brolly759, 2019-11-04:
I am working with another developer right now and he got your ping pong code to work as well on the NRF52 platform. He did mention 1 thing though:

"The Nordic SDK they used is 14.2.0.
That is a very old SDK.
The latest Nordic SDK is 16.0.
It should have some problems to DFU from 14.2 directly to 16."

beegee-tokyo, 2019-11-04:
As I am using the Adafruit nRF52 Arduino framework and cannot change the SDK version they are using.
But I am running the library without problems on a module that combines a nRF52832 and a SX1262 chip in one module => [Insight SIP ISP4520](https://www.insightsip.com/shop-online/product/listing/27-isp4520). OTA DFU is working without problems.

brolly759, 2019-11-04:
Insight SIP ISP4520 is more expensive than just using nRF52832 and SX1262 module separately but the library does work on other hardware that I tested so I am not forced into the ISP4520 module. 

Anyways, good work!

beegee-tokyo, 2019-11-04:
I mentioned the ISP4520 only because it is the nRF52832 module that I used to test the library.

closed 2020-01-18 by beegee-tokyo