beegee-tokyo/SX126x-Arduino #8: DRF1262T support?

brolly759, 2019-12-11:
DRF1262T module uses 1 pin for Antenna switching using SW pin. 

I had a skype conversation with them and they said:

1. we use the PE4259 as RF switch
SW pin is used as the power supply for the RF swith the DIO2 is internally used to swtich RX/TX
for other modules, they might use AS179 when you need the PE4259 to work, you can set SW to high logic when you want to turn off PE4259 in order to reduce the current leak of PE4259, you can set it to logic low for SW pn

Does your library have support for an SW pin or high / low pin for RF switch? (nRF52832)

Thanks

http://www.dorji.com/docs/data/DRF1262T.pdf

brolly759, 2019-12-12:
Also:

"hwConfig.USE_DIO3_TCXO = true;			  // Example uses an eByte E22 module which uses DIO3 to control oscillator voltage"

on the E22 module, DIO3 is connected to what pin on nRF52832?

beegee-tokyo, 2020-01-18:
Sorry for late reply. This issue got lost in my emails.

Right now there is no support for the SW pin the Dorji modules use. But I might add it, as I just got some Dorji DRF1262 modules.

buccaneer-jak, 2020-08-14:
> Sorry for late reply. This issue got lost in my emails.
> 
> Right now there is no support for the SW pin the Dorji modules use. But I might add it, as I just got some Dorji DRF1262 modules.

Just wanted to check if you are still planning (?) to add SW support for the DRF1262T or how your tests with that module went as I have just received some for testing.

Thanks

brolly759, 2020-08-14:
@buccaneer-jak 
The library should work as is. The SW Pin is the supply voltage for the RF Switch. My recommendation is, without editing beegee-tokyo's library, on battery powered device, Use any GPIO to toggle HIGH or LOW the SW pin when you need to use SX1262T (send or receive messages). If what you are doing is externally powered without batteries, connect SW to VCC always. Here is the datasheet for the DRF1262T module, you can look at the schematic yourself for the SW pin. http://www.dorji.com/docs/data/DRF1262T.pdf

beegee-tokyo, 2020-08-14:
@buccaneer-jak Please follow @brolly759 advice. I am not planning to add specific support for the DRF126x modules.

beegee-tokyo, 2020-09-29:
You are lucky, I needed the feature to control the power of the antenna switch in another project.

in **`hwConfig`** structure a new setting is added to enable power control of the antenna switch. The GPIO controlling the power is defined in the already existing RADIO_RXEN entry.
Example: 
```cpp
	// Define the HW configuration between MCU and SX126x
	hwConfig.CHIP_TYPE = SX1262_CHIP;		  // Example uses an eByte E22 module with an SX1262
	hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // LORA RESET
	hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;	 // LORA SPI CS
	hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;   // LORA SPI CLK
	hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;   // LORA SPI MISO
	hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // LORA DIO_1
	hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;   // LORA SPI BUSY
	hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;   // LORA SPI MOSI
	hwConfig.RADIO_TXEN = RADIO_TXEN;		  // LORA ANTENNA TX ENABLE
	hwConfig.RADIO_RXEN = RADIO_RXEN;		  // LORA ANTENNA RX ENABLE
	hwConfig.USE_DIO2_ANT_SWITCH = true;	  // Example uses an CircuitRocks Alora RFM1262 which uses DIO2 pins as antenna control
	hwConfig.USE_DIO3_TCXO = true;			  // Example uses an CircuitRocks Alora RFM1262 which uses DIO3 to control oscillator voltage
	hwConfig.USE_DIO3_ANT_SWITCH = false;	 // Only Insight ISP4520 module uses DIO3 as antenna control
	hwConfig.USE_RXEN_ANT_PWR = true;              // Enable power control of RF switch over GPIO defined by RADIO_RXEN
```
**hwConfig.RADIO_RXEN = RADIO_RXEN;** ==> RADIO_RXEN should be the GPIO used to control the RF switch power
**hwConfig.USE_RXEN_ANT_PWR = true;** ==> Enable the power control of the RF switch

closed 2020-10-23 by beegee-tokyo