beegee-tokyo/SX126x-Arduino #11: Class of Nodes (LoRaWAN)

josjimenezja, 2020-03-27:
Hi again @beegee-tokyo  I hope you are well.
I have a questions about the class B and C of nodes in lorawan, i try to switch betwen A,B and C, but just work well the class A, for my project is will possible need  the class C, do you know how can do that this class of node works correctly.

beegee-tokyo, 2020-03-28:
What exactly is your problem?

I just run a quick test with my LoRaWan test app and Class C works just fine.

In the callback lorawan_has_joined_handler I request switch to CLASS_C and I see in lorawan_confirm_class_handler callback I see that the switch was done.

Here are my two callbacks:
```cpp
/**@brief LoRa function for handling HasJoined event.
 */
static void lorawan_has_joined_handler(void)
{
#if (OVER_THE_AIR_ACTIVATION != 0)
	Serial.println("Network Joined");
#else
	Serial.println("OVER_THE_AIR_ACTIVATION != 0");

#endif
	// lmh_class_request(CLASS_A);
	lmh_class_request(CLASS_C);

	TimerSetValue(&appTimer, LORAWAN_APP_TX_DUTYCYCLE);
	TimerStart(&appTimer);
	// app_timer_start(lora_tx_timer_id, APP_TIMER_TICKS(LORAWAN_APP_TX_DUTYCYCLE), NULL);
	Serial.println("Sending frame");
	send_lora_frame();
}
```
```cpp
/**@brief Function to confirm LORaWan class switch.
 *
 * @param[in] Class  New device class
 */
static void lorawan_confirm_class_handler(DeviceClass_t Class)
{
	Serial.printf("switch to class %c done\n", "ABC"[Class]);

	// Informs the server that switch has occurred ASAP
	m_lora_app_data.buffsize = 0;
	m_lora_app_data.port = LORAWAN_APP_PORT;
	lmh_send(&m_lora_app_data, LMH_UNCONFIRMED_MSG);
}
```

josjimenezja, 2020-03-28:
I tested the class A sending a downlink from the gateway and response fine after the uplink. I try to test the class C that is the node that always listen but when i try to send a downlink from the gateway, the have to wait the uplink to send the message like the class A.

beegee-tokyo, 2020-03-28:
While the LoRaWan node Class A or Class B it is only listening for a short time after sending a package. So for the switch from A to C the device will receive the message containing the switch command only after it did a transmit.

Found [this](https://www.rfwireless-world.com/Tutorials/LoRaWAN-classes.html) helpful to understand the differences between the classes.

If you are using my example code, the code to switch the class must be sent using port #3, otherwise the downlink is treated as a data package.

On the device, do you receive the downlink message? 

My gateway has a problem with downlink messages, so I cannot really test right now.

beegee-tokyo, 2020-03-28:
Ok, got my downlink problem fixed. I can now send messages from TTN to the device.

Now there is something I do not understand. TTN sending is always scheduled, independent of the device class. So even if I set the device to Class C, TTN is sending the message only after a message was received from the device. No idea how to get around this.

josjimenezja, 2020-03-28:
> On the device, do you receive the downlink message?
Yes, i tryed firts with the node classe A and works like you say, each time send and the node open two listening windows.

> Now there is something I do not understand. TTN sending is always scheduled, independent of the device class. So even if I set the device to Class C, TTN is sending the message only after a message was received from the device. No idea how to get around this.

This is my problem too.

> Found the answer. TTN is not supporting immediate sending to Class C devices (yet)
> https://www.thethingsnetwork.org/forum/t/class-c-does-not-work-on-the-ttn-community/32492/7
Ah okay, y going to try to do a test with another server that we have.

> How are you trying to send data to the device? Through TTN or another server
My first tests was with TTN, i'm going to do another teset with a chrisptack server and then we tell you.

Thanks you very much for your help.

> Found the answer. TTN is not supporting immediate sending to Class C devices (yet)
> https://www.thethingsnetwork.org/forum/t/class-c-does-not-work-on-the-ttn-community/32492/7
Ah okay, y going to try to do a test with another server that i have.

> How are you trying to send data to the device? Through TTN or another server
My first tests was with TTN, i'm going to do another teset with a chrisptack server.

josjimenezja, 2020-03-28:
@beegee-tokyo  i have one more question, do you know how i can send a messege when a event happend, not for time interval ? Both at node A and node C.

beegee-tokyo, 2020-03-29:
Depends on the event, but for example if you have a sensor that sets an interrupt signal on one of your GPIO's you can do something like:
```cpp
...
#define SENSOR_INTERRUPT 16 // Sensor interrupt line connected to GPIO16
bool sensorTriggered = false;
...
void sensorTriggered(void)
{
  sensorTriggered = true;
}
...
void setup(void)
{
...
  pinMode (SENSOR_INTERRUPT, INPUT);
  attachInterrupt(SENSOR_INPUT, sensorTriggered, RISING); // Interrupt from sensor is active HIGH
...
}
void loop( void)
{
...
  // Send data if interrupt triggered AND device has already joined LoRaWan!
   if ((sensorTriggered) && (lmh_join_status_get() == LMH_SET))
  {
    // Read data from your sensor here
    float temperature =  sensor.getTemperature();
    float humidity = sensor.getHumidity();

    // Build the data package, here as a JSON object
    DynamicJsonDocument doc(512);
    JsonObject temp = doc.createNestedObject("Temp");
    humid = doc.createNestedObject("Humid");
    temp["value"] = (round(currTemp * 100) / 100);
    humid["value"] = (round(currHumid * 100) / 100);

    // Write JSON object into the data buffer
    i = serializeJson(doc, (char *)m_lora_app_data.buffer, 512);

    // Set the port number you want to use
    m_lora_app_data.port = LORAWAN_APP_PORT;
    // Set the data size
    m_lora_app_data.buffsize = i;

   // Send the package
   lmh_error_status error = lmh_send(&m_lora_app_data, LMH_UNCONFIRMED_MSG);
   Serial.printf("lmh_send result %d\n", error);
  
    // Reset flag
    sensorTriggered = false;
  }
...  
}
```
The procedure is the same for Class A, B or C

josjimenezja, 2020-03-29:
yes, i try to do something like your example and works, but a i have a problem, when i programa an event the work fine, but at the same time the timer to send periodically works too and in the gateway come 2 message, if i just want the the event message works how i can do that?

beegee-tokyo, 2020-03-29:
In your code, look for
```cpp
	// Initialize Scheduler and timer
	uint32_t err_code = timers_init();
	if (err_code != 0)
	{
		Serial.printf("timers_init failed - %d\n", err_code);
	}
```
# and
```cpp
	TimerSetValue(&appTimer, LORAWAN_APP_TX_DUTYCYCLE);
	TimerStart(&appTimer);
```

and comment it out. Then the timed sending of data will stop.

beegee-tokyo, 2020-03-29:
Additional info regarding Class C device.

I opened a free test account at [ResIOT](https://www.resiot.io/en/) and added my gateway and my lora device.
When I sent a downlink from there, the message is immediately sent and shown on the log output of my ESP32. 

So immediate sending to Node Class C works with my library, it is just not supported by TTN.

josjimenezja, 2020-03-31:
Thank you very much @beegee-tokyo , yesterday i try to do a test with the chirpstack server but i will have to do a lot of changes to the server. Your library works spectacularly

beegee-tokyo, 2020-03-31:
Glad to hear your satisfied. But I have to forward this to Semtech and Insight, because 90% of the code was done by them. I just made it Arduino framework compatible and fixed a few bugs.

closed 2020-05-20 by beegee-tokyo