beegee-tokyo/SX126x-Arduino #12: e22-900m30s ebyte

fandySondok, 2020-03-31:
Hy @beegee-tokyo, i really enjoy your discussions in radiolib issues about ebyte module.

currently, i try to develop e22-900m30s ebyte module, as far as i know, your library and radiolib can apply in the e22-900m22s module. I already implementation radiolib and your library too in my project for e22-900m30s, and its work and can transmit. The reason why i use e22-900m30s module is i want to reach 30dbm of my signal power transmit and I use this RF Power meter (like this one https://id.aliexpress.com/item/32798899747.html) as a monitor to see how high the e22-900m30s module can transmit with your library and radiolib, and the result of my measurement is just can reach 20 dBm and cannot reach little higher from that number. As far as I know, the e22-900m22s can reach 22dbm because it uses the sx1262 Lora module. do you know how to configure your library so my e22-900m30s module can reach a little higher from that number or maybe can reach 30 dBm?

i already setting the current limitation on sx1262 library and my output in my monitor just reach -3dbm, I try to make the parameter of TX_OUTPUT_POWER 30 dBm and comment some limitation power code in sx1262 library and the result keeps -5 dBm, become so much lower, did you know why this could be happening?

thanks before

fandySondok, 2020-03-31:
sorry, i forgot something, do you know what exactly function of the TXEN and RXEN pin in the ebyte module. i see in your code in issue in radiolib, you define high on TXEN when you enter transmit mode, but in your library right now, you didn't use that both pins. i read in the e22-900m30s datasheet, its something like a switch, I have no idea about it

beegee-tokyo, 2020-03-31:
@fandySondok 

Regarding the TX power:
When I made the library I was not aware of the e22-900m30s and I have no idea how they want to achieve the 30dbm. Jan from RadioLib assumes they have an additional amplifier on the antenna output to push the power up. The Semtech chips are only capable of 22dbm and this is the max value you can set with the library. Higher values are automatically adjusted to 22dbm as the SX1262 doesn't accept higher settings.

I cannot find anything in the datasheet of the e22-900m30s if any adjustment has to be done to activate the additional amplifier. I can only guess that it is always active. So 22dbm setting of the library would lead to 30dbm at the antenna.

For the RXEN and TXEN:
That is a special configuration of eByte to reduce power consumption when the chip is neither transmitting nor receiving. The RXEN and TXEN should be connected to two GPIO's to control the antenna switch of the module. The states of RXEN and TXEN are
|RXEN|TXEN|Function|
|--|--|--|
|0|0|RX and TX switched off for max power savings|
|1|0|Antenna switch is in receive mode|
|0|1|Antenna switch is in transmit mode|
|1|1|Invalid, not sure what happens when you do that|

I am handling of RXEN and TXEN inside the library automatically if you define the GPIOs in the hwConfig structure. If these two pins are set to valid GPIO's, the library will switch RXEN and TXEN automatically to the required combination.

For the eByte module, make as well sure to set DIO2 and DIO3 to the correct function.
For an ESP32 connected to an E22-900M22 module my hwConfig structure looks like:
```cpp
...
hw_Config hwConfig;
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 4;  // LORA RESET
int PIN_LORA_NSS = 5;	// LORA SPI CS
int PIN_LORA_SCLK = 18;  // LORA SPI CLK
int PIN_LORA_MISO = 19;  // LORA SPI MISO
int PIN_LORA_DIO_1 = 21; // LORA DIO_1
int PIN_LORA_BUSY = 22;  // LORA SPI BUSY
int PIN_LORA_MOSI = 23;  // LORA SPI MOSI
int RADIO_TXEN = 26;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = 27;	 // LORA ANTENNA RX ENABLE
...
void setup() {
	// Define the HW configuration between MCU and SX126x
	hwConfig.CHIP_TYPE = SX1262_CHIP;		  // eByte E22 module with an SX1262
	hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // LORA RESET
	hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;	 // LORA SPI CS
	hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;   // LORA SPI CLK
	hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;   // LORA SPI MISO
	hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // LORA DIO_1
	hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;   // LORA SPI BUSY
	hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;   // LORA SPI MOSI
	hwConfig.RADIO_TXEN = RADIO_TXEN;		  // LORA ANTENNA TX ENABLE
	hwConfig.RADIO_RXEN = RADIO_RXEN;		  // LORA ANTENNA RX ENABLE
	hwConfig.USE_DIO2_ANT_SWITCH = false;	 // Example uses an eByte E22 module which uses RXEN and TXEN pins as antenna control
	hwConfig.USE_DIO3_TCXO = true;			  // Example uses an eByte E22 module which uses DIO3 to control oscillator voltage
	hwConfig.USE_DIO3_ANT_SWITCH = false;	 // Only Insight ISP4520 module uses DIO3 as antenna control

	// Initialize the LoRa chip
	lora_hardware_init(hwConfig);
...
}
```
In this example RXEN is connected to GPIO26 and TXEN is connected to GPIO27.
**`hwConfig.RADIO_TXEN = RADIO_TXEN;`**    
**`hwConfig.RADIO_RXEN = RADIO_RXEN;`**    
And the eByte modules do not use DIO2 as antenna switch    
**`hwConfig.USE_DIO2_ANT_SWITCH = false;`**    
And the eByte modules use DIO3 to control the oscillator voltage    
**`hwConfig.USE_DIO3_TCXO = true;`**    
And last, DIO3 is NOT used to control the antenna switch either
**`hwConfig.USE_DIO3_ANT_SWITCH = false;`**

fandySondok, 2020-03-31:
thank you for answer @beegee-tokyo, i really appreciate it.

about the configuration RX and TX, I missed that setting in your library because I focus on the output power, thanks for the explanation.

 yeah same as you, I cannot find any clue in the e22-900m30s datasheet about this additional amplifier, there is one clue that I think could be setting this module so it can achieve 30dbm is the TX current on this module can set to 650 ma, for the sx1262 library, there is a current limitation on that, so its gonna limit the TX current only 140ma or 0x38 in hex.
`SX126xWriteRegister(REG_OCP, 0x38);`
but when I set that number a little higher, the output becomes lower than the 0x38. i don't know how this current limitation works, maybe there is another parameter that must be set to increase this current limitation?

there one more question, when I read the sx1262 datasheet, if we look in the range of the power output, the range of the Lora sx126x is about 31 step, that's why the sx1261 is -17 to +14 and the sx1262 is -9 to +22. do you have an idea to shift this range becomes 0 to +31dbm, or it cannot be set in the code?

beegee-tokyo, 2020-03-31:
For both questions I will have to check both the library code and the datasheet of of the SX126x chips. If the chip accepts the values I can change the library.

The original code of the library is from Semtech, so they might have put in some protection to stay within the legal limits (14dbm in Europe and 22dbm in US). I will look into it the next days.

fandySondok, 2020-04-06:
sorry for distracting your time @beegee-tokyo, i just really want to know your response about that shifting that dBm range, do you already have an idea about that problem?

beegee-tokyo, 2020-04-06:
Sorry, I have no solution for your problem. And my time to look into the e22-900m30s is quite limited right now.

### Did you try to contact eByte about the problem?

### What I know and what you can try:

The data sheet says the power can be set as:

>  13.4.4 SetTxParams
> This command sets the TX output power by using the parameter power and the TX ramping time by using the parameter
> RampTime. This command is available for all protocols selected.
> The output power is defined as power in dBm in a range of
> • - 17 (0xEF) to +14 (0x0E) dBm by step of 1 dB if low power PA is selected
> • - 9 (0xF7) to +22 (0x16) dBm by step of 1 dB if high power PA is selected
> Selection between high power PA and low power PA is done with the command SetPaConfig and the parameter deviceSel.
> By default low power PA and +14 dBm are set.

And in the library the values that can be given are limited between -9 and 22 dBm. 
I had no chance to check what happens if you send a set TXparams command to the SX1262 with a value higher than 22dBm.

You can try by changing the limitation in the file **`sx126x.cpp`** in the library folder **`/src/radio/sx126x`**. Look for **`void SX126xSetTxParams(int8_t power, RadioRampTimes_t rampTime)`** (around line 478) and change the limitation from 22 to 30:
```cpp
	void SX126xSetTxParams(int8_t power, RadioRampTimes_t rampTime)
	{
		uint8_t buf[2];

		if (SX126xGetPaSelect(0) == SX1261)
		{
			if (power == 15)
			{
				SX126xSetPaConfig(0x06, 0x00, 0x01, 0x01);
			}
			else
			{
				SX126xSetPaConfig(0x04, 0x00, 0x01, 0x01);
			}
			if (power >= 14)
			{
				power = 14;
			}
			else if (power < -17)
			{
				power = -17;
			}
			SX126xWriteRegister(REG_OCP, 0x18); // current max is 80 mA for the whole device
		}
		else // sx1262
		{
			// WORKAROUND - Better Resistance of the SX1262 Tx to Antenna Mismatch, see DS_SX1261-2_V1.2 datasheet chapter 15.2
			// RegTxClampConfig = @address 0x08D8
			SX126xWriteRegister(0x08D8, SX126xReadRegister(0x08D8) | (0x0F << 1));
			// WORKAROUND END

			SX126xSetPaConfig(0x04, 0x07, 0x00, 0x01);
			if (power > 30) // was 22
			{
				power = 30; // was 22
			}
			else if (power < -9)
			{
				power = -9;
			}
			SX126xWriteRegister(REG_OCP, 0x38); // current max 160mA for the whole device
		}
		buf[0] = power;
		buf[1] = (uint8_t)rampTime;
		SX126xWriteCommand(RADIO_SET_TXPARAMS, buf, 2);
	}
```

fandySondok, 2020-04-06:
no problem @beegee-tokyo, i really appreciate your response about this issue.

i didn't think to contact ebyte for this problem before, because i read your comment in radiolib issue, you say they slowly answer the question. But i think i will try it.

oh i see, i gonna try that setting for my e22-900m30s module, thanks.

beegee-tokyo, 2020-05-20:
@fandySondok 

Any success with the eByte module?

fandySondok, 2020-05-21:
Yes, i got 28 dbm using your code, just a little bit different configuration. But i'm stuck when i try your lorawan code, and my eByte module is broken and i'm busy with another project too. Thanks again anyway.

beegee-tokyo, 2020-05-22:
For the 28dbm, is there any change I should apply in the library?

The LoRaWan part had some issues. I fixed it with the last two releases. Try library version 1.1.2

fandySondok, 2020-05-22:
No, your library is fine. Thanks for your info

closed 2020-05-22 by fandySondok