beegee-tokyo/SX126x-Arduino #14: Over-the-air-activation (OTAA) for LoraWAN (TTN) doesn't seem to work

mroehrich, 2020-05-03:
Hello,
i am trying to set up an OTAA activation with LoraWAN on The-Things-Network. but with no success.
I managed to have a successfull running configuration with ABP activation. Therefore hardware configuation an code should be fine.
I tried to switch from ABP to OTAA by setting the application key and set the following definition in "Commissioning.h" of the library.
`#define OVER_THE_AIR_ACTIVATION 1`
Does anyone managed to run OTAA?
Thanks for your support!

beegee-tokyo, 2020-05-20:
OTAA requires that the download link from your LoRaWan gateway works. 
I never tried OTAA to be honest.

beegee-tokyo, 2020-05-21:
OK, I finally gave it a test and got it to work.    

But I need to make some changes in the library. Will publish it asap.

One thing to be aware. At the moment a DevEUI is automatically generated if you enable OTAA with **`#define OVER_THE_AIR_ACTIVATION 1`** in **`Commissioning.h`**. To avoid this set    
**`#define STATIC_DEVICE_EUI 1`** instead of **`0`**    
in **`Commissioning.h`**. 

Result of successful OTAA registration:

Log output on ESP32:
```log
=====================================
SX126x LoRaWan test
Node 260211CE
=====================================
BoardId: 00-00-0C-66-6C-BF-71-3C
[E][LoRaMacHelper.cpp:696] lmh_init(): OTAA
DevEui=3C-71-BF-6C-66-0C-00-00
AppEui=70-B3-D5-7E-D0-02-01-E1
AppKey=2B-84-E0-B0-9B-68-E5-CB-42-17-6F-E7-53-DC-EE-79
Network Joined
switch to class C done
Data: {"i":1,"n":000}
Size: 15
Port: 2
lmh_send result -1
```
Device status on TTN:
[image]

beegee-tokyo, 2020-05-22:
Fixed in Version 1.1.2

closed 2020-05-22 by beegee-tokyo