beegee-tokyo/SX126x-Arduino #15: Does it support nRF52840?

RAKWireless, 2020-05-20:
Hi @beegee-tokyo ,

According to your example and description, this repo can work on nRF52832 + SX126X, right?
If it is, i want to ask two questions please:
1. I think this repo can support nRF52840 + SX1262 surely, right?
2. What should i do if i want to run it on nRF52840 + SX1262?
Looking forward for your reply, thank you!

beegee-tokyo, 2020-05-20:
Hi @RAKWireless , nice to see you here, I tried some stuff from you.

Yes, the library should support the nRF52840, but I did not test it. The only thing that might be required is a global **`#define NRF52`** to get the library compiled for nRF52xxx chips.

Are you using ArduinoIDE or PlatformIO?

RAKWireless, 2020-05-20:
Hi @beegee-tokyo ,
Thank you for your reply and nice to meet you here too.:)
Yes, we will use Arduino IDE, and program through nRF52840 USB.

beegee-tokyo, 2020-05-20:
Should work. 
If it throws compilation errors, let me know. I might need to change the **`#ifdef NRF52`** in the library with something that covers both the nRF52832 and the nRF52840.

Which nRF52840 are you using? The Sparkfun or Adafruit boards or something you made in house?
If you are working on a RAKWireless board/module with the nRF52840 and the SX126x combined, I am very interested in that for a customers project.

RAKWireless, 2020-05-20:
Hi @beegee-tokyo ,
OK, let me have a try.

Yes, it is RAK module, and we're using NRF52840-QIAA.
We want to run an Arduino solution on this module, but need a powerful LoRaWAN library which can work with nRF52840 and SX1262. That's why i'm very interested on your repo. :)

beegee-tokyo, 2020-05-20:
Good!
Is the module already available?

Just an update, the library fails to compile for nRF52840 on ArduinoIDE.

I need to replace all **`#ifdef NRF52`** with **`#ifdef NRF52_SERIES`** in the library sources to get it to compile.
Can you manually change the library after installation? It might take me some days before I can push a new version of the library.

RAKWireless, 2020-05-20:
Hi @beegee-tokyo ,

It will be best If you can help to update a new version library which can be compiled for nRF52840 on Arduino IDE.
The module is debugging in house finally, and it will be released soon once this Arduino software works OK on it.
Maybe we can add a test case of RAK module using nRF52840 + SX1262 in your repo after our testing, and i can send a module to you for checking. :)

beegee-tokyo, 2020-05-20:
Ok, I pushed V1.1.1 which should work with nRF52840. I had as well to fix some stuff for ArduinoIDE.

Not sure how long it takes until the version is published. But you can download the library as ZIP and install manually. Link: https://github.com/beegee-tokyo/SX126x-Arduino/archive/1.1.1.zip

RAKWireless, 2020-05-20:
Thank you very much! @beegee-tokyo

beegee-tokyo, 2020-05-20:
Let me know if you need my address for a test sample of the module :smiley:

RAKWireless, 2020-05-21:
OK 🤝

RAKWireless, 2020-05-21:
Hi @beegee-tokyo ,

It seems an issue when i try to run V1.1.1 on our module, please help:

17:27:50.302 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:27:51.284 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:27:52.271 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:27:53.253 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:27:54.201 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:27:55.182 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:27:56.162 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:27:57.138 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:27:58.121 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:27:59.100 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:00.077 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:01.059 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:02.043 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:03.025 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:04.006 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:04.982 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:05.963 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:06.913 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:07.895 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:08.872 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:09.851 -> [LORA  ] [SX126xWaitOnBusy] Timeout waiting for BUSY low
17:28:09.851 -> OVER_THE_AIR_ACTIVATION != 0
17:28:09.851 -> 
17:28:09.851 -> BSP Library : 0.20.1
17:28:09.851 -> Bootloader  : s140 6.1.1
17:28:09.851 -> Serial No   : 5CD8E8442939F1D5
17:28:09.851 -> 
17:28:19.870 -> Sending frame

beegee-tokyo, 2020-05-21:
This is either a connection problem or a wrong pin assignment. I had this before as well.

1) Check the SPI lines, DIO1, BUSY and NSS lines if possible. (Maybe not possible because it is inside the module). If you drive the RST pin of the SX1262 manually, check that one as well.

2) Make sure the correct GPIO's are assigned.

My setup for a Adafruit nRF52 Feather connected to a SX1262 chip:
```cpp
// nRF52832 - SX126x pin configuration
int PIN_LORA_RESET = 4;  // LORA RESET
int PIN_LORA_NSS = 28;   // LORA SPI CS
int PIN_LORA_SCLK = 12;  // LORA SPI CLK
int PIN_LORA_MISO = 14;  // LORA SPI MISO
int PIN_LORA_DIO_1 = 11; // LORA DIO_1
int PIN_LORA_BUSY = 29;  // LORA SPI BUSY
int PIN_LORA_MOSI = 13;  // LORA SPI MOSI
int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;	 // LORA ANTENNA RX ENABLE
// Replace PIN_SPI_MISO, PIN_SPI_SCK, PIN_SPI_MOSI with your
SPIClass SPI_LORA(NRF_SPIM2, PIN_LORA_MISO , PIN_LORA_SCLK , PIN_LORA_MOSI );
```

As you can see, I do not use the default SPI settings in above example.

RAKWireless, 2020-05-21:
OK, we'll try to check it, and give the feedback here soon.
Thank you!

beegee-tokyo, 2020-05-21:
I know it is not a released product yet, but if you feel comfortable with it, I can check your schematics and code to find out where the problem might be.

RAKWireless, 2020-05-22:
Hi @beegee-tokyo ,
It is a prototype now, and there are some hardware issues which are found in our testing, so that the schematics is not final, and it may take some confuse. I can send you the schematics surely once it is done, and send you a module. :)
Please wait for my good news. 😄

beegee-tokyo, 2020-05-22:
Perfect. If you need any more help with the code let me know.

Regarding LoRaWan, I found another problem if you want to use OTAA instead of APB. Will release a new version today.

RAKWireless, 2020-05-26:
Hi @beegee-tokyo ,
I think i can send you the shematics now, and maybe send you the module soon, please contact with me in email fomi@rakwireless.com.
Thank you!

beegee-tokyo, 2020-06-27:
Should work now (Tested with Sparkfun nRF52840 board and Circuitrocks Alora SX1262).

closed 2020-06-27 by beegee-tokyo