beegee-tokyo/SX126x-Arduino #17: questions

tcpipchip, 2020-07-24:
Good morning @beegee-tokyo  @ivankravets 

First of all, congratulations!

I have some questions

-have you plan to port to STM32 ? I ask you because i have a lot of LoRa modules here and they use the STM32. Today we program it using the Semtech libraries, but with KEIL. I would like to use the ARDUINO and make a lot of blogs.
-will your lib work with 1276 ?

Have a good day!

beegee-tokyo, 2020-07-24:
Hi,

Answer #1 - No this library does not work with SX1276 transceivers. It supports only SX1261/SX1262 and SX1268 transceivers.
Answer #2 - I was thinking about porting it to STM32 architecture, but right now I do not have much time to do it. It should be not too difficult, but the problems of porting are
- inconsistent Arduino implementation of timers. On nRF52 it is SoftwareTimer, on ESPxxx it is Ticker, I have to find out what it is in the STM32 Arduino framework
- inconsistent Arduino implementation of the SPI interfaces.
- I do not have any board with the STM32 at hand

tcpipchip, 2020-07-24:
hi

 #1 :(
 #2 We will fork and try to solve this problems late!
Just now we are tryng to port SPI interfaces to STM32 work with WIFININA.

tcpipchip, 2020-07-24:
btw, congratulations about your Job, we hope help you soon! I use the NRF52832/840

beegee-tokyo, 2020-07-24:
About #1, I was looking into it several times. Unfortunately it's not that easy to integrate the SX127x chips. Semtech's sources are not easy to combine. If there would be an mbed lib that supports both chip families I would have based this lib on it.
About #2, if you solve it, I am happy to merge it!!!

beegee-tokyo, 2020-08-24:
@tcpipchip  How is the port for STM32 going?

tcpipchip, 2020-08-24:
i aborted because the company that produce the STM32+SX127X (module) didnt give the GPIO connection

beegee-tokyo, 2020-08-24:
May I ask which company it is?

tcpipchip, 2020-08-24:
wisol

closed 2020-08-25 by beegee-tokyo