beegee-tokyo/SX126x-Arduino #18: EBYTE E22-900T30S using UART, can it be plugged into the library?

avillacis, 2020-07-25:
Hello. I have this project using an Espressif ESP32 with an EBYTE E22-900T30S. I have been skimming over this library, and so far I can see that, unlike the E22-900M22S that has an SPI interface and some pins that directly enable or disable transmission and such, the E22-900T30S is controlled mainly through an UART interface to read and write data over the radio link, and two additional pins (M0 and M1) to switch between data and register-programming modes. My project involves speaking with a LoRaWAN gateway, and since I already know that the E22 does not have any internal LoRaWAN stack, I was considering the possibility of tweaking the abstraction layer to speak through the UART instead of using SPI. However, there exists the possibility that the UART interface does not expose enough of the underlying chip capabilities to actually implement LoRaWAN on top of the UART, or that it implements its own abstraction over the radio link that gets in the way of speaking to the LoRa chipset.

Can LoRaWAN, in principle, be implemented on top of the UART interface? Is it known one way or the other? If not, what should I know about capabilities exposed through the interface in order to make an informed decision?

beegee-tokyo, 2020-07-26:
I have one Ebyte E22 with UART, the one which has an USB connector on it to use with a computer. To be honest, after 1 day of trying to find out how to set it up I put it away and never took it out again.
Ebyte didn't want to tell me how the interface works and how I could access the SX126x chip registers. So it turned out useless.

If you want to try, there are four source files which handle the low level of the communication and need to be changed to use UART. This would be in **_/src/boards/mcu/espressif/_** the files **`spi_board.cpp`** and **`spi_board.h`**. And in **_/src/boards/sx126x/_**  the files **`sx126x-board.cpp`** and **`sx126x-board.h`**.

But unless you know how to access the SX126x registers directly over the UART interface I don't think it can be done.

closed 2020-08-17 by beegee-tokyo

Anurag1999127, 2022-05-20:
Hi
I have worked with Melange's LORA module that work with UART and now I have to shift to Ebyte E22 due to some reason. May someone help me on how to get started. I am using ESP32 as the micro-controller.

beegee-tokyo, 2022-05-20:
You could start with the [LoRaWAN.ino](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/examples/LoRaWan/LoRaWan.ino) example.
You see there how the pin assignments between the module and the ESP32 are done
[Required pins](https://github.com/beegee-tokyo/SX126x-Arduino/blob/dc15fab55f3975171959c2255153d14427d989d4/examples/LoRaWan/LoRaWan.ino#L29) !! can be different for your module !!
[Setup the HW configuration](https://github.com/beegee-tokyo/SX126x-Arduino/blob/dc15fab55f3975171959c2255153d14427d989d4/examples/LoRaWan/LoRaWan.ino#L129)  !! can be different for your module !!
[Initialize the SX1262](https://github.com/beegee-tokyo/SX126x-Arduino/blob/dc15fab55f3975171959c2255153d14427d989d4/examples/LoRaWan/LoRaWan.ino#L168)

Anurag1999127, 2022-05-21:
Hi Bernd Giesecke
Thank You for replying back to me.
The code that you have shared is of LORAWAN, and I got to know about how to
declare the pins and connect them to ESP32. What if After declaration, I
only want to send a "HELLO" string through LORA. I donot want to use
LORAWAN. Is there any help.

beegee-tokyo, 2022-05-21:
LoRa P2P is here: [PingPong](https://github.com/beegee-tokyo/SX126x-Arduino/tree/master/examples/PingPong)

Anurag1999127, 2022-05-21:
Oh I see.

Ok
1)So if I am using an ESP32 micro-controller then I can delete the portions
of ESP8266 and NRF52.
2)Since I want to communicate between two LORA node devices with different
ESP32 as micro-controllers. Do I upload the same firmware in both the
devices?

beegee-tokyo, 2022-05-21:
Yes, you can remove the ESp8266 and nRF52 parts.

PingPong example should work if you upload the same code to two ESP32+SX1262. I didn't run it for a long time and the code is more than 1 year old. If you run into problems, let me know.

Anurag1999127, 2022-05-23:
I compiled the code, there wasn't any error but when I uploaded the code, It got stuck. I am giving the serial monitor output:

11:54:15.958 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:54:15.958 -> configsip: 0, SPIWP:0xee
11:54:15.958 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:54:15.958 -> mode:DIO, clock div:1
11:54:15.958 -> load:0x3fff0018,len:4
11:54:15.958 -> load:0x3fff001c,len:1216
11:54:15.958 -> ho 0 tail 12 room 4
11:54:15.958 -> load:0x40078000,len:10944
11:54:15.958 -> load:0x40080400,len:6388
11:54:15.958 -> entry 0x400806b4
11:54:16.052 -> =====================================
11:54:16.052 -> SX126x PingPong test
11:54:16.052 -> =====================================
11:54:16.098 -> MCU Espressif ESP32
11:54:16.098 -> BoardId: 00-00-CC-99-42-14-4B-8C
11:54:16.098 -> Starting lora_hardware_init
11:54:16.144 -> Starting Radio.Rx
11:54:46.530 -> OnRxTimeout

And it gets stuck here as both of the devices are listening.

beegee-tokyo, 2022-05-23:
Both modules are declared as `master` [static bool isMaster = true;](https://github.com/beegee-tokyo/SX126x-Arduino/blob/dc15fab55f3975171959c2255153d14427d989d4/examples/PingPong/PingPong.ino#L92) so after the timeout they should start to send. Whichever module reach the timeout first, should stay master and the other module become slave.

If nothing happens after the `OnRxTimeout`, then I think there is something else wrong.
Here you can see how it looks like:

[image]

If that does not happen, then your connection between the MCU and the LoRa transceiver are wrong.

beegee-tokyo, 2022-05-23:
Are you using Arduino IDE or Platform IO?
You could enable library debug output to see more details.

**Arduino IDE**
You have to edit [sx126x-debug.h](https://github.com/beegee-tokyo/SX126x-Arduino/blob/dc15fab55f3975171959c2255153d14427d989d4/src/sx126x-debug.h#L18) and change `#define LIB_DEBUG 0` to `#define LIB_DEBUG 1`

**PlatformIO**
Add
```
build_flags = 
	-DLIB_DEBUG=0 
```
in platformio.ini of your project.

Anurag1999127, 2022-05-23:
Hi.
Thanks a lot for your time and help. Actually there some some antenna issue. 
It is working fine. Communication has been established between two devices.

Should I still change the above mentioned changes.

Also, is there any way to decrease the RxTimeOut Period. It is taking almost 4-5 mins.

beegee-tokyo, 2022-05-23:
No need to do the changes if it works now.

The timeout is defined as [#define RX_TIMEOUT_VALUE 3000](https://github.com/beegee-tokyo/SX126x-Arduino/blob/dc15fab55f3975171959c2255153d14427d989d4/examples/PingPong/PingPong.ino#L83)

4-5 minutes is strange. the value of 3000 means that the timeout is 3 seconds.

I tried it and the timeout is indeed very long. Not sure why. 
Didn't use that example for a long time. It was meant as a simple example for LoRa P2P.

Try to change the last parameter of [Radio.SetRxConfig()](https://github.com/beegee-tokyo/SX126x-Arduino/blob/dc15fab55f3975171959c2255153d14427d989d4/examples/PingPong/PingPong.ino#L182) to false.

Seems to be a bug in the example. This parameter selects continuous RX which causes the long timeout.
After changing it fro m true to false, the timeout is 3 seconds.
```cpp
Radio.SetRxConfig(MODEM_LORA, LORA_BANDWIDTH, LORA_SPREADING_FACTOR,
					  LORA_CODINGRATE, 0, LORA_PREAMBLE_LENGTH,
					  LORA_SYMBOL_TIMEOUT, LORA_FIX_LENGTH_PAYLOAD_ON,
					  0, true, 0, 0, LORA_IQ_INVERSION_ON, false);
```

This changes the time to 4 seconds, which is normal because 
CAD time
timeout
delay of 500ms before next sending is adding up to ~4 seconds

[image]

Anurag1999127, 2022-05-23:
Oh Okay. Got it.
I am still going through the code and the library files to understand more of it.

Anurag1999127, 2022-05-24:
Hi Bernd
Just wanted to know how much range have you tested using this firmware ?.

Anurag1999127, 2022-05-24:
Also is there any help which I can get if I had to do the same project in ESP IDF.

Anurag1999127, 2022-05-24:
Oh I see.

Anurag1999127, 2022-05-27:
Hi Bernd

I had a question while going through the library. Can you mention where these functions are getting called. I can see there definations in pingpong.ino but i cant find where these are getting called  :

OnTxDone;
OnRxDone;
OnTxTimeout;
 OnRxTimeout;
 OnRxError;
OnCadDone;

Anurag1999127, 2022-05-30:
Hi Bernd
I have done couple of changes in the code and I came across a problem.
In the uploaded png file, you can see that I am sending a Device ID instead of PING/PONG.
After Radio.send(data, buffersize); , it takes exactly 20 seconds and then the data is actually sent.

Every time there is a delay of 20 seconds. Can you help me decrease this delay.
Just to let you know, I have not added any delay myself.
[image]

beegee-tokyo, 2022-05-30:
Did you remove that part from `OnCadDone()`
[image]

It fills the payload up to the buffer size of 64 bytes.
So you are not sending only the 4 bytes of your ID but always send 64 bytes, which can take some time.

Other than that I have no explanations for the 20 seconds.

Anurag1999127, 2022-05-31:
Yes I have removed this and I have also declared my buffer size as 4. So that less data is transferred.

Anurag1999127, 2022-05-31:
> Yes I have removed this and I have also declared my buffer size as 4. So that less data is transferred.

Can that cause a 20 second delay?

Anurag1999127, 2022-05-31:
Ok.