beegee-tokyo/SX126x-Arduino #21: sx1262 on nucleo-l452re-p

fandySondok, 2020-10-22:
Hi @beegee-tokyo, have you ever try the library from MBED for sx1262 in SX126xMB2CAS board for stm32 https://os.mbed.com/components/SX126xMB2xAS/, I already try that library in my Nucleo board, and it is success to send and receive data from ttn.

My problem is when I try that library for sx1262 from ebyte using my Nucleo board, it didn't work. I try to find that problem, and I found the busy pin didn't off or the ebyte didn't wake up when the NSS pin goes low.

I try your code on my esp32, and it behaves like the datasheet says, its wake up when the NSS pin goes low the same when I try library from MBED for sx1262 in SX126xMB2CAS board for stm32.

Did you know anything about this problem? I know they're so much difference between your library and the MBED library but i think the main problem in LoRaMacInitialization.

beegee-tokyo, 2020-10-22:
@fandySondok 
Sorry, I never tried the MBED library. 
My experience with the eByte modules are mixed. I have a few modules that show problems while most of them (all E22-900M22S) works fine.

A critical point seems to be the time between Reset going off and first access to the eByte module. I usually have the reset pin connected to the rest of the MCU.

fandySondok, 2020-10-22:
Thanks for your response, much appreciate it.
Sorry for didn't really catch up on your idea about "the time between Reset". You mean in RadioInit there is sx126xInit function and there is SX1262Reset function there right. I already monitor the output from that function, it's gonna wake the E22-900M22S right. 

The MBED library using reset pin too, same as your library. Sorry if my issues out of topic from your lib.

beegee-tokyo, 2020-10-22:
Forget what I wrote before. Long time since I fought with the reset problem. I just looked into my code and there is basically no extra delay between SX1262Reset() and accessing the module over SPI.

What you can check is the timing for the reset signal in the MBED library. I do 10 ms LOW and wait 20ms after pulling the RESET back to high.
```cpp
	void SX126xReset(void)
	{
		pinMode(_hwConfig.PIN_LORA_RESET, OUTPUT);
		digitalWrite(_hwConfig.PIN_LORA_RESET, LOW);
		delay(10);
		digitalWrite(_hwConfig.PIN_LORA_RESET, HIGH);
		delay(20);
		dio3IsOutput = false;
	}
```

That works for me on several modules (ISP4520, RAK4631 and the Alora board with the eByte E22)

fandySondok, 2020-10-23:
I already did like you suggest, but it's still didn't wake the ebyte e22. The SPI working because the MCU can set the ebyte e22 to enter sleep mode, but when the mcu wants to set the ebyte again, mcu must wake the ebyte right and finally get stuck in waitOnBusy() because the NSS low state cannot wake ebyte e22 module.

But thank you for your answer, even this issue didn't directly related to your library.

this is the reset func in MBED lib for stm32, if just in case you curious to see it 

`void SX126xReset( void )
{
  GPIO_InitTypeDef initStruct = { 0 };

  initStruct.Mode =GPIO_MODE_OUTPUT_PP;
  initStruct.Pull = GPIO_NOPULL;
  initStruct.Speed = GPIO_SPEED_HIGH;

  // Wait 10 ms
  DelayMs( 10 );

  // Set RESET pin to 0
  HW_GPIO_Init( RADIO_RESET_PORT, RADIO_RESET_PIN, &initStruct );
  HW_GPIO_Write( RADIO_RESET_PORT, RADIO_RESET_PIN, 0 );

  // Wait 20 ms
 DelayMs( 20 );

  // Configure RESET as input
//  initStruct.Mode = GPIO_MODE_ANALOG;
  HW_GPIO_Init( RADIO_RESET_PORT, RADIO_RESET_PIN, &initStruct );
  HW_GPIO_Write( RADIO_RESET_PORT, RADIO_RESET_PIN, 1); // internal pull-up

  // Wait 10 ms
  DelayMs( 10 )
}`

beegee-tokyo, 2020-10-23:
@fandySondok 
It is a strange behavior and it is weird that it seems to work with my library and ESP32 but not with the MBED stuff.
Maybe you can raise the question on MBED. Not sure if they have a similar issue function.

I guess it is ok if I close this issue now.

closed 2020-10-23 by beegee-tokyo