beegee-tokyo/SX126x-Arduino #22: Control Downlink

Iotcreator-low-power-wireless, 2020-11-27:
Hello, just start to play with RAK5005-0 v1.0 and  RAK 1920 ver. A , successfully upload .ino file, and transmit messages to gateway. Now i wonder what payload i need to send to the node and what port i need to use . In the repo i found this: 
static void lorawan_rx_handler(lmh_app_data_t *app_data)
{
	Serial.printf("LoRa Packet received on port %d, size:%d, rssi:%d, snr:%d\n",
				  app_data->port, app_data->buffsize, app_data->rssi, app_data->snr);

	switch (app_data->port)
	{
	case 3:
		// Port 3 switches the class
		if (app_data->buffsize == 1)
		{
			switch (app_data->buffer[0])
			{
			case 0:
				lmh_class_request(CLASS_A);
				break;

			case 1:
				lmh_class_request(CLASS_B);
				break;

			case 2:
				lmh_class_request(CLASS_C);
				break;

			default:
				break;
			}
		}
		break;

	case LORAWAN_APP_PORT:
		// YOUR_JOB: Take action on received data
		break;

	default:
		break;
	}
} 
And my other question is , how i can compare the payload , with some value and manage some IO pin of board. 
Thanks!!!

beegee-tokyo, 2020-11-30:
Hi,
the data port is defined in your code, the default is `#define LORAWAN_APP_PORT 2` so you need to use data port 2. But you can change it in the code to any port you want to use.

I do not know `what payload i need to send to the node`. If you need to send data to the node, you should know what you want to send and in what format you want to send it.

On the node side, the data is arriving as a byte[] array and you have to convert it to same format you used when you sent it.

Iotcreator-low-power-wireless, 2020-11-30:
Ok i get it . I'm using for server ChirpStack , and sending payload in encode base64 format. In the Serial Monitor of Arduino IDE software i received the payload which is in
 void lorawan_rx_handler(lmh_app_data_t *app_data)
{
	Serial.printf("LoRa Packet received on port %d, size:%d, rssi:%d, snr:%d, data:%s\n",
				  app_data->port, app_data->buffsize, app_data->rssi, app_data->snr, app_data->buffer);
}
The basic question is how to compare the received payload, and then (for example: some of IO go to HIGH ). I Can't figure out how to write a code to compare the received payload. If you can help me will be great.

beegee-tokyo, 2020-11-30:
If you send it as base64 encoded data then you need to decode it on once you received it. There are libraries available to do this for example https://github.com/adamvr/arduino-base64.
Then you have to save the values like GPIO1 is high, GPIO2 is low, ....
Then you check the current status of the GPIO's with the new value and switch it on/off if different.

nikolaieniware, 2020-12-02:
Hello again . Can you help me with some example  to compare the received downlink . In my case i send base64 MDE= , which is 01 in text . This i received in the WiSblock Serial Monitor  ==  data:01  ==. 
First thing where i compare is the fport because i need to send diferenet data to diferent port.. 
with command  if (app_data->port == 1) {
} 
and this i works . But i dont know what is the syntax of compare the received data. I try with       if (app_data->buffer[1] == 01){
}If you can help me will be great.

beegee-tokyo, 2020-12-03:
I am sorry, but I do not have the time to teach.

closed 2020-12-03 by beegee-tokyo