beegee-tokyo/SX126x-Arduino #23: Sx1262-Arduino ESP32 node not connected to Gateway(RPI+RHF0M301)

iamcalledusman, 2020-11-29:
Hi All , 

First of All thank you for SX1262-Arduino library . I am facing an issue while using this library , i need your help in this regard .
i have followed the following steps to test the SX1262-Arduino library .

1)i have connected the Gateway(RPI+RHF0M301) with loriot server .
2)Node which i use to connect with gateway was "Semtech SX1262MB2xAs" and controller Stm32Nucleo board.
3)You can see the data sent from the node to loriot server .
  
[image]

4)After that, i changed the node device controller  to "ESP32"  and kept the Lora device same "Semtech SX1262MB2xAs" as mentioned above in point #2 .

5)Connections i have made are according to the mentioned in the library. These are total 7 connections and 2 more connections are 3.3v and Gnd.

 int PIN_LORA_RESET = 4;  // LORA RESET
int PIN_LORA_NSS = 5;	// LORA SPI CS
int PIN_LORA_SCLK = 18;  // LORA SPI CLK
int PIN_LORA_MISO = 19;  // LORA SPI MISO
int PIN_LORA_DIO_1 = 21; // LORA DIO_1
int PIN_LORA_BUSY = 22;  // LORA SPI BUSY
int PIN_LORA_MOSI = 23;  // LORA SPI MOSI

6)Compiled the library in Arduino IDE and as well as in Visual Code Studio  (PlatformIO extension).
7)Got two different results with both the IDE's 

Results are as Follows 

A)When compiled with PlatformIO . Following output i got on docklight terminal 

[image]

B)When compiled with the Arduino IDE then 

[image]

8)To solve the above issues , i have followed the instructions mentioned in this link 
  https://github.com/beegee-tokyo/SX126x-Arduino/issues/1
regarding the commissioning.h file . But still the same issue 

9) i also changed the LORAWAN_APPLICATION_EUI ,LORAWAN_APPLICATION_KEY and LORAWAN_DEVICE_EUI  same when i was able to connect the node to gateway and then to loriot server as mentioned above from Point # 1-3 .

But Still the same issue . Can you please guide me where is the issue ? I will appreciate the help and thank you very much in advance !

Best Regards ,

beegee-tokyo, 2020-11-30:
Hi

Problem (A) `LORA Busy timeout waiting for BUSY low`
Causes:
- Wrong pin connections => can be ruled out because in second test it could connect
- ESP32 problem. I get the same error (but only on ESP32) sometimes after programming the ESP32 and automatic restart of the ESP32. What solves the problem is a power down / power up cycle and after that it works.

Problme (B) `lmh_send result -1`
Causes:
- lmh_send result -1 is thrown when the payload is too large for the selected data rate does not allow the payload size.
Did you select the same DR that you used with the `Stm32Nucleo` or did you change the payload size?

iamcalledusman, 2020-11-30:
Hi 
First of all , thank you ! for the response .

Problem (A): LORA Busy timeout waiting for BUSY low

- Wrong pin connections => can be ruled out because in second test it could connect

i have doubt on this , can you please explain little more on it , '-1' means no pin selected am i right ? we no need connect 
these pins right ? in SX1262 i didn't see Tx & Rx Enable pins.

int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;	 // LORA ANTENNA RX ENABLE

- ESP32 problem

When i used the PlatformIO i got different behavior and with arduino IDE i got the different one ,Can you please explain about automatic restart of esp32 ?How it solve the issue ? i did restart it via serial port software but still the same issue .

Yes , i have set the same date rate which was DR_2 in "stm32_Nucleo" board case . As for payload size is concerned i am sending "Hello World!" which is a default payload in this library example .

Once again thank you for the concern !

beegee-tokyo, 2020-11-30:
RADIO_TXEN and RADIO_RXEN are used for some modules (eByte E22 e.g.) to control the antenna switch. If you control the antenna with DIO2 of the SX1262 then setting them to -1 is correct.

What I wanted to say is that `LORA Busy timeout waiting for BUSY low` is usually thrown when 
- any GPIO connected to the SX1262 is wrong
- the SX1262 is hanging

`Automatic  restart`
When finished flashing the ESP32 is automatically restarting. In some cases I see that the SX1262 is not responding after that `automatic restart`. Then I need to do the power down - power up sequence. After that it works. A reset via serial port is not sufficient.

I made the examples for both ArduinoIDE and PlatformIO and I never saw a different behavior between the code generated by any of them.

For the send problem I have no explanation what could cause it. As I said lmh_send result -1 is usually thrown by a payload that is too large.

iamcalledusman, 2020-11-30:
Thank you so much ! for the expert opinion . I am looking into it again and observe the things ! Will share my results as soon as i get positive results .

Again Thank You!

iamcalledusman, 2020-12-03:
Hi 
Thanks for the previous help ! 
There is a little progress , now able to connect the end node with the gateway using your SX1262 library .Let me share which lines i have commented after that it get connected . 

[image]

[image]

[image]

I have few questions regarding above images . In last image if you see i am sending only  "HELLO".From your suggestion i reduced the size of payload .When i increase it i get -1 error .Although i have checked it DR_2 and as well with DR_3 .But still the payload size issue . When i was using the stm32 +sx1262 then i was sending 18 characters with above mentioned setting . 

One more question about loriot server app data.Let me share the data on gateway .First paragraph is the STM32+SX1262 Data and second one is the ESP32+SX1262 data .

[image]

All parameters are same but i am unable to see data on loriot server .But in case of Stm32+sx1262 able to see the data .Can you please guide me in this regard ?

Thanks in Advance

iamcalledusman, 2020-12-03:
Sorry , one more question to ask .When my gateway is not powered on then still i get this message as shown in the image below .

I pasted the following line in commissioning.h

#define OVER_THE_AIR_ACTIVATION         1

[image]

Can you please explain , why its showing Network Joined message .Although , Gateway is powered off . But when i power up the gateway and configure it with the commands then gateway shows the message via packet forwarder as i sent in my previous post .

Again Thanks in Advance .

iamcalledusman, 2020-12-05:
Hi Every One 
Issue has been resolved , i mis configured one line in main.c file .Let me share the  of code .

[image]

After that , after a few second it get connected and i am able to see the messages on loriot server .

[image]

and at gateway messages are like that 

[image]

Thanks for the Sx1262 +ESP 32 library . And for the startup help !

Again Thanks !

closed 2020-12-05 by iamcalledusman