beegee-tokyo/SX126x-Arduino #24: Compatible with Arduino Mbed Core?

sabas1080, 2020-12-14:
Hi @beegee-tokyo 

I am trying to test this library with the [Arduino Core Mbed and Arduino Nano 33 BLE](https://github.com/arduino/ArduinoCore-mbed) but I have the next errors

Undefined functions
```
Arduino/libraries/SX126x-Arduino/src/boards/sx126x/sx126x-board.cpp:165: undefined reference to `BoardDisableIrq'
/Arduino/libraries/SX126x-Arduino/src/boards/sx126x/sx126x-board.cpp:178: undefined reference to `BoardEnableIrq'
```
and SPI undefined

can you help me?

Thanks

beegee-tokyo, 2020-12-14:
Hi @sabas1080 
The world is small, we can discuss this as well on WeChat `BastWAN Group` 😸 
I never tried the library with [Arduino Core Mbed and Arduino Nano 33 BLE](https://github.com/arduino/ArduinoCore-mbed) so I am not sure if it will work with Mbed at all.

**For the missing SPI declaration.**
For nRF52 MCU's you have to set the SPI in your code, see for example in [LoRaWan.ino](https://github.com/beegee-tokyo/SX126x-Arduino/blob/266a81975f333f21e0445c7a9bd6444520b87353/examples/LoRaWan/LoRaWan.ino#L52) like
```cpp
SPIClass SPI_LORA(NRF_SPIM2, 14, 12, 13);
```
**For the missing `BoardDisableIrq` and `BoardEnableIrq`**
They should be defined for nRF52 in the file [nrf52832/board.cpp](https://github.com/beegee-tokyo/SX126x-Arduino/blob/0e0025ab2b12ff6065bc3641c9c7fb66e9767a3e/src/boards/mcu/nrf52832/board.cpp)
For this file to be included, a `#define NRF52_SERIES 1` is needed. But this should be done by the Arduino BSP already. Not sure about Mbed.
But for nRF52 MCU's these functions are empty anyway. They were only needed for the ESP32 to disable core switching during critical phases.

sabas1080, 2020-12-14:
With the core Arduino Core  `#define NRF52_SERIES` is 1, I have set the SPI

`SPIClass SPI_LORA(NRF_SPIM2, PIN_LORA_MISO, PIN_LORA_SCLK, PIN_LORA_MOSI);`

but I have the next error:
```
error: no matching function for call to 'arduino::HardwareSPI::HardwareSPI(NRF_SPIM_Type*, int&, int&, int&)'
 SPIClass SPI_LORA(NRF_SPIM2, PIN_LORA_MISO, PIN_LORA_SCLK, PIN_LORA_MOSI);
                                                                         ^
In file included from /Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/cores/arduino/api/ArduinoAPI.h:31:0,
                 from /Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/cores/arduino/Arduino.h:51,
                 from /Users/sabas/Library/Arduino15/packages/raknrf/hardware/nrf52/0.21.1/libraries/RAK_examples/examples/RAK4631/Communications/LoRa/LoRaP2P/LoRaP2P_RX/LoRaP2P_RX.ino:19:
/Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/cores/arduino/api/HardwareSPI.h:104:7: note: candidate: constexpr arduino::HardwareSPI::HardwareSPI()
 class HardwareSPI
       ^~~~~~~~~~~
/Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/cores/arduino/api/HardwareSPI.h:104:7: note:   candidate expects 0 arguments, 4 provided
/Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/cores/arduino/api/HardwareSPI.h:104:7: note: candidate: constexpr arduino::HardwareSPI::HardwareSPI(const arduino::HardwareSPI&)
/Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/cores/arduino/api/HardwareSPI.h:104:7: note:   candidate expects 1 argument, 4 provided
/Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/cores/arduino/api/HardwareSPI.h:104:7: note: candidate: constexpr arduino::HardwareSPI::HardwareSPI(arduino::HardwareSPI&&)
/Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/cores/arduino/api/HardwareSPI.h:104:7: note:   candidate expects 1 argument, 4 provided
```

other errors

```
folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_cache_708271/core/core_electroniccats_mbed_bast_frontier_02345bd81a1c423dfe4f1366d72767e1.a
Linking everything together...
/Users/sabas/Library/Arduino15/packages/electroniccats/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -L/var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240 -Wl,--gc-sections -w -Wl,--as-needed @/Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/variants/BAST_FRONTIER/ldflags.txt -T/Users/sabas/Library/Arduino15/packages/electroniccats/hardware/mbed/1.0.4/variants/BAST_FRONTIER/linker_script.ld -Wl,-Map,/var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/LoRaP2P_RX.ino.map --specs=nosys.specs -o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/LoRaP2P_RX.ino.elf /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/sketch/LoRaP2P_RX.ino.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/ISP4520_MOD.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/RAK4630_MOD.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/boards/mcu/board.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/boards/mcu/espressif/board.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/boards/mcu/espressif/spi_board.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/boards/mcu/espressif/timer.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/boards/mcu/nrf52832/board.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/boards/mcu/nrf52832/spi_board.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/boards/mcu/nrf52832/timer.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/boards/sx126x/sx126x-board.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/LoRaMac.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/LoRaMacCrypto.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/LoRaMacHelper.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/region/Region.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/region/RegionAS923.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/region/RegionAU915.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/region/RegionCN470.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/region/RegionCN779.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/region/RegionCommon.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/mac/region/RegionEU433.cpp.o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX12 [... cut ...]
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/radio/sx126x/radio.cpp:1052: undefined reference to `TimerStart'
/var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_327240/libraries/SX126x-Arduino/radio/sx126x/radio.cpp.o: In function `RadioIsChannelFree':
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/radio/sx126x/radio.cpp:638: undefined reference to `TimerGetCurrentTime'
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/radio/sx126x/radio.cpp:641: undefined reference to `TimerGetElapsedTime
```

beegee-tokyo, 2020-12-14:
Seems Mbed doesn't like the declaration of the SPI with    
```cpp
SPIClass SPI_LORA(NRF_SPIM2, PIN_LORA_MISO, PIN_LORA_SCLK, PIN_LORA_MOSI);
```
No idea how to assign MISO/MOSI/SCLK pins in Mbed.

sabas1080, 2020-12-14:
I have declared
`MbedSPI SPI_LORA(PIN_LORA_MISO, PIN_LORA_SCLK, PIN_LORA_MOSI);`

I now have an error with timers
```
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:44:2: error: 'SoftwareTimer' does not name a type
  SoftwareTimer timerTickers[10];
  ^~~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp: In function 'void TimerInit(TimerEvent_t*, void (*)())':
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:67:6: error: 'timerTickers' was not declared in this scope
      timerTickers[idx].begin(10000, (TimerCallbackFunction_t)obj->Callback, NULL, false);
      ^~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:67:6: note: suggested alternative: 'timerTimes'
      timerTickers[idx].begin(10000, (TimerCallbackFunction_t)obj->Callback, NULL, false);
      ^~~~~~~~~~~~
      timerTimes
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:67:38: error: 'TimerCallbackFunction_t' was not declared in this scope
      timerTickers[idx].begin(10000, (TimerCallbackFunction_t)obj->Callback, NULL, false);
                                      ^~~~~~~~~~~~~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:71:6: error: 'timerTickers' was not declared in this scope
      timerTickers[idx].begin(10000, (TimerCallbackFunction_t)obj->Callback, NULL, true);
      ^~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:71:6: note: suggested alternative: 'timerTimes'
      timerTickers[idx].begin(10000, (TimerCallbackFunction_t)obj->Callback, NULL, true);
      ^~~~~~~~~~~~
      timerTimes
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:71:38: error: 'TimerCallbackFunction_t' was not declared in this scope
      timerTickers[idx].begin(10000, (TimerCallbackFunction_t)obj->Callback, NULL, true);
                                      ^~~~~~~~~~~~~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp: In function 'void TimerStart(TimerEvent_t*)':
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:88:3: error: 'timerTickers' was not declared in this scope
   timerTickers[idx].stop();
   ^~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:88:3: note: suggested alternative: 'timerTimes'
   timerTickers[idx].stop();
   ^~~~~~~~~~~~
   timerTimes
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp: In function 'void TimerStop(TimerEvent_t*)':
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:95:3: error: 'timerTickers' was not declared in this scope
   timerTickers[idx].stop();
   ^~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:95:3: note: suggested alternative: 'timerTimes'
   timerTickers[idx].stop();
   ^~~~~~~~~~~~
   timerTimes
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp: In function 'void TimerReset(TimerEvent_t*)':
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:102:3: error: 'timerTickers' was not declared in this scope
   timerTickers[idx].stop();
   ^~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:102:3: note: suggested alternative: 'timerTimes'
   timerTickers[idx].stop();
   ^~~~~~~~~~~~
   timerTimes
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp: In function 'void TimerSetValue(TimerEvent_t*, uint32_t)':
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:110:3: error: 'timerTickers' was not declared in this scope
   timerTickers[idx].setPeriod(value);
   ^~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126x-Arduino/src/boards/mcu/nrf52832/timer.cpp:110:3: note: suggested alternative: 'timerTimes'
   timerTickers[idx].setPeriod(value);
   ^~~~~~~~~~~~
   timerTimes
```

beegee-tokyo, 2020-12-14:
That is a problem. 
I am using the Arduino nRF52 BSP SoftwareTimer library to time the receive windows RX1 and RX2 and for timeouts.
Seems MBed doesn't support that part of the Arduino BSP as well.
Maybe you need to include it as separate library. The Arduino SoftwareTimer lib is here https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/cc69ecc95d71f8b0d42ccff69e316cda38edf41e/cores/nRF5/utility

beegee-tokyo, 2021-02-03:
@sabas1080 As we are discussing this as well on another channel, I will close it here.

closed 2021-02-03 by beegee-tokyo

EricAndrechek, 2025-04-24:
How did this ever get solved? Having similar issues myself.

beegee-tokyo, 2025-04-25:
@EricAndrechek 
No, the library does not support mbed for nRF52.
The only MCU that is supported with mbed is the RP2040.