beegee-tokyo/SX126x-Arduino #25: Sx1262-Arduino ESP32 Payload Size

iamcalledusman, 2020-12-15:
Hi Beegee_tokyo!

 Hope you are doing well!. I am having issue in understanding that how to increase the payload size .Recently , able to send the "HELLO WORLD!" only when i increase the payload size it gives me error "-1" . In another post you replied me that this error is because of increase in payload size . According to my understanding i changed the default Data rate from DR_0 to DR_1 , DR_2 , DR_3 and DR_4 but same error with no data receive at server side .

Thanks in advance .

Best Regards,

beegee-tokyo, 2020-12-16:
@iamcalledusman 
There seems to be a problem in the library when ADR is set to on.
Try to disable ADR and try with DR3 .
Unfortunately I have no time at the moment to debug the problem.

iamcalledusman, 2020-12-16:
@beegee-tokyo 

Thanks for the response ! Tried it by disabling the ADR , and changing it to DR_3 . But same issue !
Sorry to ask ! When you will be able to see this issue ?

[image]

[image]

Thank You !

beegee-tokyo, 2020-12-17:
@iamcalledusman 
I run a test and can send up to 232 bytes. RAK7258 Gateway and Chirpstack running on RPI3.

```log
Search sensors on I2C
=====================================
RAK4631 LoRaWan test
=====================================
=====================================
SX126x initialization
=====================================
=====================================
Syncword = 2414
=====================================
Start network join request
OTAA joined and got dev address 01DDA933
switch to class C done
Sending frame
Send success 2 bytes
Sending frame
Send success 12 bytes
Sending frame
Send success 22 bytes
Sending frame
Send success 32 bytes
Sending frame
Send success 42 bytes
Sending frame
Send success 52 bytes
Sending frame
Send success 62 bytes
Sending frame
Send success 72 bytes
Sending frame
Send success 82 bytes
Sending frame
Send success 92 bytes
Sending frame
Send success 102 bytes
Sending frame
Send success 112 bytes
Sending frame
Send success 122 bytes
Sending frame
Send success 132 bytes
Sending frame
Send success 142 bytes
Sending frame
Send success 152 bytes
Sending frame
Send success 162 bytes
Sending frame
Send success 172 bytes
Sending frame
Send success 182 bytes
Sending frame
Send success 192 bytes
Sending frame
Send success 202 bytes
Sending frame
Send success 212 bytes
Sending frame
Send success 222 bytes
Sending frame
Send success 232 bytes
Sending frame
Send success 242 bytes
Sending frame
UP failed -1
```
I attached the sources. But you have to change the **`nodeDeviceEUI`**, **`nodeAppEUI`**, **`nodeAppKey`**
The log from Chirpstack is in the ZIP file as well.

[LoRaWan-Payload-Size-Test.zip](https://github.com/beegee-tokyo/SX126x-Arduino/files/5706863/LoRaWan-Payload-Size-Test.zip)

iamcalledusman, 2020-12-17:
@beegee-tokyo 

Thank you so much for the help ! I will test and update you soon on this .

Again Thank You

beegee-tokyo, 2020-12-17:
I forgot to mention
Region US915
Subband channel 1
DR set to DR3
ADR is off

iamcalledusman, 2020-12-17:
Hi @beegee-tokyo !

Thank you the help!

I have tested and attached the main.c file and also  image files . I am still facing the same issue :( 

First time i used first set of nodeDeviceEUI, nodeAppEUI, nodeAppKey compiled them and load into the ESP32 and 
in my second test i have used the second set of nodeDeviceEUI, nodeAppEUI, nodeAppKey as mentioned in below image.
But still the same results.

[image]

Results on serial 

[image]

[image]

[image]

[image]

[main.txt](https://github.com/beegee-tokyo/SX126x-Arduino/files/5712299/main.txt)

Can you please help me in this regard ! Thanks for the help and your library .

Best Regards

beegee-tokyo, 2020-12-18:
I adapted my code to run on ESP32 and it sends without problems up to 242 bytes package size.

Please adapt the SX1262 Pin assignment (main.cpp lines 23-41) to your configuration and hwConfig setting (main.cpp line 214 to 226) to your configuration and **try to run my code.**

I see as well that you are using BLE and WiFi together. That can lead to a memory problem on a none-Wrover ESP32. You might try to switch to [NimBLE](https://github.com/h2zero/NimBLE-Arduino) library which uses less heap than the original BLE lib from Espressif.

I am sorry, but there is no problem with the library, I tried on both nRF52 and ESP32 and it works.
[LoRaWan-Payload-Size-Test-ESP32.zip](https://github.com/beegee-tokyo/SX126x-Arduino/files/5713168/LoRaWan-Payload-Size-Test-ESP32.zip)

iamcalledusman, 2020-12-18:
@beegee-tokyo 

Thank you for the response and guidelines .i will follow and update you soon on this .

Thanks 

Best Regards

beegee-tokyo, 2020-12-18:
@iamcalledusman 
Just for reference, my test environment:
- Custom boards with [Insight ISP4520](https://www.insightsip.com/products/combo-smart-modules/isp4520) modules.
- [RAKwireless WisBlock with RAK4631](https://docs.rakwireless.com/Product-Categories/WisBlock/)
- [Adafruit ESP32 Feather](https://www.adafruit.com/product/3405) with [Circuitrocks Alora board](https://circuit.rocks/product:2685).

iamcalledusman, 2020-12-18:
@beegee-tokyo 

I have tried with another file which was the default library main.c  with default project . Now there is no BLE and Wifi .

i have attached the main file 
[main.txt](https://github.com/beegee-tokyo/SX126x-Arduino/files/5717880/main.txt)

Can you please go through it and check where is the issue .Because now using the default project . 

Facing the same issue with payload size . I know at your end its working with your gateway and with end device . At my end i did a lot of effort to find the issue but fail to do so . 

Thanks for the help you did for me !

beegee-tokyo, 2020-12-19:
Line 190:    
`lmh_class_request(CLASS_A);`     
try `Class_C` instead.

No idea why it works with `Class_C`. The low-level parts of the library are Semtech code. I just build the wrapper around it.

iamcalledusman, 2020-12-19:
@beegee-tokyo 

  Thank You So Much for your Support ! Yes now it work for Class C

[image]

[image]

Again Thank You ! One more question . Have you tested this library with Class B ? 

Best Regards

Muhammad Usman

beegee-tokyo, 2020-12-19:
Just tried it and Class B seems to work somehow. But there are send failures showing.
Sorry, I am not an expert in LoRaWAN. I originally created the library for LoRa P2P usage.
```
=====================================
RAK4631 LoRaWan test
=====================================
=====================================
SX126x initialization
=====================================
=====================================
Syncword = 2414
=====================================
Start network join request
OTAA joined and got dev address 01363726
switch to class B done
Sending frame
Send success 2 bytes
Sending frame
Send success 12 bytes
Sending frame
Send success 22 bytes
Sending frame
UP failed -1
Sending frame
Send success 42 bytes
Sending frame
Send success 52 bytes
Sending frame
Send success 62 bytes
Sending frame
UP failed -1
Sending frame
Send success 82 bytes
Sending frame
UP failed -1
Sending frame
Send success 102 bytes
Sending frame
UP failed -1
Sending frame
Send success 122 bytes
Sending frame
UP failed -1
Sending frame
Send success 142 bytes
Sending frame
UP failed -1
Sending frame
Send success 162 bytes
Sending frame
UP failed -1
Sending frame
Send success 182 bytes
Sending frame
UP failed -1
Sending frame
Send success 202 bytes
Sending frame
UP failed -1
Sending frame
Send success 222 bytes
Sending frame
UP failed -1
Sending frame
Send success 242 bytes
Sending frame
UP failed -1
```

iamcalledusman, 2020-12-19:
@beegee-tokyo 

Thank you for your great response ! As for Class B is concerned . What  do you recommend me that can i use Class B in my project ? Can it be stable enough ? or can i make changes in it ? Because i don't have experience of converting the libraries . Before that i was using the I-CUBE-LRWAN 1.3.1 for stm32 nucleo .And last question is which lorawan server are you using for class B testing ? TTN / Chirpstack

iamcalledusman, 2020-12-19:
@beegee-tokyo 

i have attached the excel file having comparison between class A (10 Bytes) , B(25+ Bytes)  and C(25+Bytes) successful and lost Packets when time interval was 2 Seconds . And When time increased by 10 seconds in Class B it goes to 100% successful Transmission rate .
[ClassABandC packet comparision.xlsx](https://github.com/beegee-tokyo/SX126x-Arduino/files/5720112/ClassABandC.packet.comparision.xlsx)

For Quick view i have attached the images of first few packets .

[image]

Class B only , 100% success rate when time interval between packets sent is 10 Seconds .

[image]

closed 2020-12-20 by iamcalledusman