beegee-tokyo/SX126x-Arduino #26: SX1262-ESP32 Class B testing

iamcalledusman, 2020-12-20:
@beegee-tokyo 

Hi Again !

Thank you for your amazing support for my previous quires. I have a question , i am able to switch to class B as you told me in previous question . Have you tested class B lorawan , is it fulfilling the Class B TX and RX criteria as per the following link 

https://www.rfwireless-world.com/Tutorials/LoRaWAN-classes.html

?  Can you please guide me how can i test Class B ?  Using which server is the best choice to start chirpstack or any other alternative ? 

Thanks In advance

beegee-tokyo, 2020-12-21:
The original Semtech code I used for the library didn't support Class B at all. It did not allow to switch to Class B.
I made a minor change to allow to switch to Class B, but I did not do any tests, because Class B features (beacon, ...) are not supported.

iamcalledusman, 2020-12-22:
@beegee-tokyo 

Thank you for your answer ! yes i have tested the class B and shared the results issue # 25 , It get connected with class B

https://github.com/beegee-tokyo/SX126x-Arduino/issues/25

Can you please help me to solve Class A payload size issue ? if you have limited time to solve the issue can you guide me
so that i try to solve it .

Thanks And Regards,

beegee-tokyo, 2020-12-22:
@iamcalledusman 
I am sorry, the problem is somewhere inside the Semtech code, I do not even know where to start.
I think it would be easier to update the library to the latest version of the Semtech code. But I have no time to do it.

iamcalledusman, 2020-12-22:
@beegee-tokyo 

Thank you for the update ! You mean to say that to replace the  sx126x.h and sx126x.cpp files with latest ones ?

beegee-tokyo, 2020-12-23:
The LoRaWan stuff is all in the **`mac`** folder. That is where the problem with the Class A most likely happens. **`boards`**, **`system`** and **`radio`** folders are the MCU and SX126x support files. There are no problems in LoRa communication itself, so I don't think the problem is there.
But I know that Semtech made changes in the **`radio`** files as well.

iamcalledusman, 2020-12-23:
@beegee-tokyo 

Thank you ! i will update you on this after debugging the issue .

beegee-tokyo, 2021-02-03:
@iamcalledusman 
New version V1.3.0 should fix the payload problem. 
I added a table that shows [Max packet size per region / data rate](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/MAX_PACKET_SIZE.md)

closed 2021-02-03 by beegee-tokyo

iamcalledusman, 2021-02-05:
@beegee-tokyo 

Thank you so much ! i will test and update you soon on this !