beegee-tokyo/SX126x-Arduino #28: Class C Support

christianpaetz, 2021-03-18:
Hi, I like to use Class C support. In Class A the application works well but at the moment when I switch to Class C , the guru kicks in with the following stack. Any suggestion or any working sample code for Class I can use to debug?

PC: 0x400914f9: uxListRemove at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c line 218
EXCVADDR: 0x00000004

Decoding stack results
0x400914f9: uxListRemove at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c line 218
0x4009082e: xTaskRemoveFromEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3092
0x4008e509: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 767
0x400ecf85: spiEndTransaction at /Users/christian/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-spi.c line 765
0x400eada6: SPIClass::endTransaction() at /Users/christian/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI/src/SPI.cpp line 143
0x400dfecc: SX126xWakeup() at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/boards/sx126x/sx126x-board.cpp line 151
0x400e4bac: SX126xCheckDeviceReady() at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/radio/sx126x/sx126x.cpp line 128
0x400dfee3: SX126xWriteCommand(RadioCommands_t, uint8_t*, uint16_t) at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/boards/sx126x/sx126x-board.cpp line 162
0x400e4ef9: SX126xSetDioIrqParams(uint16_t, uint16_t, uint16_t, uint16_t) at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/radio/sx126x/sx126x.cpp line 403
0x400e4a78: RadioRx(uint32_t) at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/radio/sx126x/radio.cpp line 1021
0x400e0ee6: RxWindowSetup(bool, uint32_t) at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/mac/LoRaMac.cpp line 1503
0x400e0fba: OnRxWindow2TimerEvent() at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/mac/LoRaMac.cpp line 1475
0x400e1f08: LoRaMacMibSetRequestConfirm(MibRequestConfirm_t*) at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/mac/LoRaMac.cpp line 2723
0x400e2f5b: lmh_class_request(DeviceClass_t) at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/mac/LoRaMacHelper.cpp line 1076
0x400d35a9: lorawan_has_joined_handler() at /Users/christian/Documents/Arduino/MC5/lorawan.ino line 213

beegee-tokyo, 2021-03-18:
@christianpaetz 
First of all:
_**Thank you very much for the best documented bug report I have seen so far.**_

I am running an ESP32 with eByte E22-M900S (SX1262 module) as Class C device without any problem. I am using the LoRaWAN example, only changes I made is to use CLASS_C instead of CLASS_A. My Region setup is AS923.

A few questions to narrow down your problem:
- Are you using the LoRaWAN example?
   - If yes, is it the unchanged code (beside of using CLASS_C of course)?
- Are you compiling with Arduino IDE or PlatformIO?
- What region did you setup the LoRaWAN parameters?

christianpaetz, 2021-03-18:
Honest Answer: No, I dont use the example but have some other activities around that use SPI. The SPI seems to be the problem. What I wonder that on crash point one core is in SPI read while the other one is in SPI write. When I turn off all SPI operations I can prevent crashing but real-world operations require such things. 

CORE 0:
0x4008fafe: vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/include/freertos/portmacro.h line 285
0x4008e4c9: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 734
0x400ecfe1: spiEndTransaction at /Users/christian/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-spi.c line 765
0x400eae02: SPIClass::endTransaction() at /Users/christian/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI/src/SPI.cpp line 143
0x400dffe0: SX126xReadCommand(RadioCommands_t, uint8_t*, uint16_t) at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/boards/sx126x/sx126x-board.cpp line 197

CORE 1:
0x400ecf11: spiTransaction at /Users/christian/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-spi.c line 722
0x400eade5: SPIClass::beginTransaction(SPISettings) at /Users/christian/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI/src/SPI.cpp line 135
0x400dff14: SX126xWakeup() at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/boards/sx126x/sx126x-board.cpp line 148
0x400e4c08: SX126xCheckDeviceReady() at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/radio/sx126x/sx126x.cpp line 128
0x400dff3f: SX126xWriteCommand(RadioCommands_t, uint8_t*, uint16_t) at /Users/christian/Documents/Arduino/libraries/SX126x-Arduino/src/boards/sx126x/sx126x-board.cpp line 162

christianpaetz, 2021-03-28:
Sorry to get back on this. Maybe I just need a small hint to understand the code. In CLASS_C the RX2 Window is active and triggered by HW timer resulting in OnMacStateCheckTimerEvent() (in LoRaMAC.cpp) running on CPU0. This function accesses the SX chip using the usual SPI functions without any further protection.  The main loop running on CPU 1 needs to call RadioIrqProcess() (in radio.cpp) frequently which is also accessing the SPI interface to SX1261. Sooner or later the two processes will try to access the chip at the very same time and the Guru will hit. In Class_A this is quite unlikely but in CLASS_C it happens more often. I would assume that the main loop needs to be blocked from accessing SX chip when timer routine is running but there is no code for this :-o

beegee-tokyo, 2021-03-29:
RadioIRQProcess is only doing SPI activities if an Interrupt occurred. While RX2 Window is active, an interrupt means that the SX126x received something during RX2 Window and it will be handled by the MAC. I don't see any conflict here and I never saw a problem.

closed 2021-03-29 by beegee-tokyo

GuillaumeHupi, 2023-03-16:
Hello @christianpaetz 

We are having a similar issue with an heltec wireless stick lite V2 using lorawam class C. We are using some SPI sensors, and when I see the logs, i also think it is related to we spi communication.

To help me solving this problem, I'd like to know if you managed to solve your problem and how, or if at least you managed to earn more informations ? 

Here are my logs after a crash: 

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008d9ba  PS      : 0x00060f34  A0      : 0x8008cbaf  A1      : 0x3ffbfff0
A2      : 0x3ffb8a8c  A3      : 0x3ffd87a4  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060f23  A7      : 0x00000000  A8      : 0x3ffd87a4  A9      : 0x3ffd87a4
A10     : 0x0000000e  A11     : 0x0000000e  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00060f21  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x4008beea  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008d9ba

ELF file SHA256: 0000000000000000

Backtrace: 0x4008d9ba:0x3ffbfff0 0x4008cbac:0x3ffc0010 0x4008acbf:0x3ffc0030 0x400f0c95:0x3ffc0070 0x400e67d9:0x3ffc0090 0x400e99b4:0x3ffc00c0 0x400e99ec:0x3ffc00e0 0x400e9ac1:0x3ffc0100 0x400ec63d:0x3ffc0120 0x400eaf71:0x3ffc0170 0x400eaa1e:0x3ffc01c0 0x400e769c:0x3ffc01e0 0x400e76b7:0x3ffc0200 0x400e7749:0x3ffc0220 0x400ecd55:0x3ffc0240 0x40081281:0x3ffc0260 0x40088239:0x3ffc0280 0x400e67e7:0x3ffd8540 0x400eda3f:0x3ffd8560 0x400eda7a:0x3ffd8580 0x400edb33:0x3ffd85a0 0x400d2e1f:0x3ffd85c0 0x400d2e4d:0x3ffd85f0 0x400d3030:0x3ffd8620 0x400d3088:0x3ffd8640 0x400d30e0:0x3ffd8660 0x400d4e90:0x3ffd8690 0x400d710b:0x3ffd86c0 0x4008ae86:0x3ffd86f0
  #0  0x4008d9ba:0x3ffbfff0 in vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:188
  #1  0x4008cbac:0x3ffc0010 in vTaskPlaceOnEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #2  0x4008acbf:0x3ffc0030 in xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2038
  #3  0x400f0c95:0x3ffc0070 in spiTransaction at C:/Users/mehdi/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-spi.c:283
  #4  0x400e67d9:0x3ffc0090 in SPIClass::beginTransaction(SPISettings) at C:/Users/mehdi/.platformio/packages/framework-arduinoespressif32@3.10006.210326/libraries/SPI/src/SPI.cpp:269
  #5  0x400e99b4:0x3ffc00c0 in McuClass::singleTransfer(unsigned short, unsigned char) at D:\\61722\\Documents\\Arduino\\libraries\\ESP32_LoRaWAN_VASR\\src\/Mcu.cpp:381
  #6  0x400e99ec:0x3ffc00e0 in McuClass::readRegister(unsigned short) at D:\\61722\\Documents\\Arduino\\libraries\\ESP32_LoRaWAN_VASR\\src\/Mcu.cpp:381
  #7  0x400e9ac1:0x3ffc0100 in read0 at D:\\61722\\Documents\\Arduino\\libraries\\ESP32_LoRaWAN_VASR\\src\/Mcu.cpp:381
  #8  0x400ec63d:0x3ffc0120 in SX1276Read at lib/ESP32_LoRaWAN/src/sx1276.c:1263
      (inlined by) SX1276SetRxConfig at lib/ESP32_LoRaWAN/src/sx1276.c:550
  #9  0x400eaf71:0x3ffc0170 in RegionEU868RxConfig at lib/ESP32_LoRaWAN/src/region/RegionEU868.c:624
  #10 0x400eaa1e:0x3ffc01c0 in RegionRxConfig at lib/ESP32_LoRaWAN/src/region/Region.c:1020
  #11 0x400e769c:0x3ffc01e0 in OnRxWindow2TimerEvent at lib/ESP32_LoRaWAN/src/LoRaMac.c:3069 (discriminator 1)
  #12 0x400e76b7:0x3ffc0200 in OpenContinuousRx2Window at lib/ESP32_LoRaWAN/src/LoRaMac.c:3069 (discriminator 1)
  #13 0x400e7749:0x3ffc0220 in OnRadioRxTimeout at lib/ESP32_LoRaWAN/src/LoRaMac.c:3069 (discriminator 1)
  #14 0x400ecd55:0x3ffc0240 in SX1276OnDio1Irq at lib/ESP32_LoRaWAN/src/sx1276.c:1649
  #15 0x40081281:0x3ffc0260 in __onPinInterrupt at C:/Users/mehdi/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-gpio.c:274
  #16 0x40088239:0x3ffc0280 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #17 0x400e67e7:0x3ffd8540 in SPIClass::endTransaction() at C:/Users/mehdi/.platformio/packages/framework-arduinoespressif32@3.10006.210326/libraries/SPI/src/SPI.cpp:269
  #18 0x400eda3f:0x3ffd8560 in Adafruit_SPIDevice::endTransaction() at lib/Adafruit_BusIO-master/Adafruit_SPIDevice.cpp:491
  #19 0x400eda7a:0x3ffd8580 in Adafruit_SPIDevice::endTransactionWithDeassertingCS() at lib/Adafruit_BusIO-master/Adafruit_SPIDevice.cpp:491 
  #20 0x400edb33:0x3ffd85a0 in Adafruit_SPIDevice::write_then_read(unsigned char const*, unsigned int, unsigned char*, unsigned int, unsigned char) at lib/Adafruit_BusIO-master/Adafruit_SPIDevice.cpp:491
  #21 0x400d2e1f:0x3ffd85c0 in MAX31865::readRegisterN(unsigned char, unsigned char*, unsigned char, unsigned char) at src/MAX31865.cpp:100  
  #22 0x400d2e4d:0x3ffd85f0 in MAX31865::readRegister8(unsigned char, unsigned char) at src/MAX31865.cpp:100
  #23 0x400d3030:0x3ffd8620 in MAX31865::clearFault(unsigned char) at src/MAX31865.cpp:100
  #24 0x400d3088:0x3ffd8640 in MAX31865::readRTD(unsigned char) at src/MAX31865.cpp:475
  #25 0x400d30e0:0x3ffd8660 in MAX31865::temperature(float, float, unsigned char) at src/MAX31865.cpp:383
  #26 0x400d4e90:0x3ffd8690 in readTemp(unsigned char) at src/io.cpp:1313
[... cut ...]
  #14 0x400e6cc9:0x3ffecf70 in SendFrame() at lib/ESP32_LoRaWAN/src/ESP32_LoRaWAN.cpp:201
  #15 0x400e6e93:0x3ffecfb0 in LoRaWanClass::send(eDeviceClass) at lib/ESP32_LoRaWAN/src/ESP32_LoRaWAN.cpp:495
  #16 0x400d2b54:0x3ffecfd0 in sendLora() at src/LoraWan.cpp:298
  #17 0x400d7390:0x3ffed000 in vTaskLora(void*) at src/main.cpp:1080
  #18 0x4008ae86:0x3ffed130 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

christianpaetz, 2023-03-16:
In 2021 I fixed the problem by adding some IRQ blocking but later versions solved the problem and my version works fine since then.

GuillaumeHupi, 2023-03-20:
OK. I'll check which version we are using and if we can try to upgrade. 
Thank you very much !

christianpaetz, 2023-03-20:
Actually I dont have the code anymore. Please try to install most-recent version of the lib. There are so many fixes included. it does not make sense to work with old code and try to add things that were fixed long time ago.