beegee-tokyo/SX126x-Arduino #32: OTAA join request fails forever until WiFi is turned on

avillacis, 2021-03-29:
I am doing an exercise on a possible use scenario of LoRaWAN: my sketch uses a library that conditionally initializes WiFi so that a configuration interface can be shown to the user. After entering the values for appEUI, deviceEUI and appKey, the LoRaWAN connection should be established, and the WiFi can then be turned off to save energy. The LoRaWAN configuration values are stored in NVRAM so that a reboot can use them immediately.

The problem I am having (using ChirpStack as networkserver and RAK7258 as gateway devices) is that, when starting from a cold boot, with the WiFi turned OFF, and correct LoRaWAN values already stored in NVRAM, the join request keeps failing over and over again, even with the gateway less than one meter from the device:

`
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: net/1/identity NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: net/1/password NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: net/2/identity NOT_FOUND
[E][Preferences.cpp:472] getString(): nvs_get_str len fail: net/2/password NOT_FOUND
[V][Preferences.cpp:341] getUChar(): nvs_get_u8 fail: net/2/lastsel NOT_FOUND
[E][Preferences.cpp:49] begin(): nvs_open failed: NOT_FOUND
[E][Preferences.cpp:49] begin(): nvs_open failed: NOT_FOUND
INFO: WiFi inicialmente APAGADO
[I][LoRaMacHelper.cpp:734] lmh_init(): OTAA
DevEui=C8-2B-96-9D-3E-9C-00-00
DevAdd=00000000
AppEui=00-00-00-00-00-00-00-00
AppKey=E4-9C-8A-C1-CC-D0-34-9A-BF-75-0C-DA-CA-FB-2E-75
[E][sx126x-board.cpp:149] SX126xWaitOnBusy(): LORA Busy timeout waiting for BUSY low
[E][LoRaMacHelper.cpp:309] lmh_setSubBandChannels(): [LMH] Selected subband 2
[E][LoRaMacHelper.cpp:309] lmh_setSubBandChannels(): [LMH] Selected subband 2
[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 4991 #2 6002
[D][LoRaMac.cpp:662] OnRadioTxDone(): OnRadioTxDone => TX was Join Request
[D][radio.cpp:1340] RadioIrqProcess(): RadioIrqProcess => IRQ_RX_TX_TIMEOUT
[D][LoRaMac.cpp:1225] OnRadioRxTimeout(): OnRadioRxTimeout
[D][LoRaMac.cpp:1315] OnMacStateCheckTimerEvent(): Join network failed 1 time(s)

[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 4991 #2 6002
[D][LoRaMac.cpp:662] OnRadioTxDone(): OnRadioTxDone => TX was Join Request
[D][radio.cpp:1340] RadioIrqProcess(): RadioIrqProcess => IRQ_RX_TX_TIMEOUT
[D][LoRaMac.cpp:1225] OnRadioRxTimeout(): OnRadioRxTimeout
[D][LoRaMac.cpp:1315] OnMacStateCheckTimerEvent(): Join network failed 2 time(s)

[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 4991 #2 6002
[D][LoRaMac.cpp:662] OnRadioTxDone(): OnRadioTxDone => TX was Join Request
[D][radio.cpp:1340] RadioIrqProcess(): RadioIrqProcess => IRQ_RX_TX_TIMEOUT
[D][LoRaMac.cpp:1225] OnRadioRxTimeout(): OnRadioRxTimeout
[D][LoRaMac.cpp:1315] OnMacStateCheckTimerEvent(): Join network failed 3 time(s)

ERR: OVER_THE_AIR_ACTIVATION failed! Retrying...
[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 4991 #2 6002
[D][LoRaMac.cpp:662] OnRadioTxDone(): OnRadioTxDone => TX was Join Request
[D][radio.cpp:1340] RadioIrqProcess(): RadioIrqProcess => IRQ_RX_TX_TIMEOUT
[D][LoRaMac.cpp:1225] OnRadioRxTimeout(): OnRadioRxTimeout
[D][LoRaMac.cpp:1315] OnMacStateCheckTimerEvent(): Join network failed 1 time(s)

[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 4991 #2 6002
[D][LoRaMac.cpp:662] OnRadioTxDone(): OnRadioTxDone => TX was Join Request
[D][radio.cpp:1340] RadioIrqProcess(): RadioIrqProcess => IRQ_RX_TX_TIMEOUT
[D][LoRaMac.cpp:1225] OnRadioRxTimeout(): OnRadioRxTimeout
[D][LoRaMac.cpp:1315] OnMacStateCheckTimerEvent(): Join network failed 2 time(s)

[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 4991 #2 6002
[D][LoRaMac.cpp:662] OnRadioTxDone(): OnRadioTxDone => TX was Join Request
[D][radio.cpp:1340] RadioIrqProcess(): RadioIrqProcess => IRQ_RX_TX_TIMEOUT
[D][LoRaMac.cpp:1225] OnRadioRxTimeout(): OnRadioRxTimeout
[D][LoRaMac.cpp:1315] OnMacStateCheckTimerEvent(): Join network failed 3 time(s)

ERR: OVER_THE_AIR_ACTIVATION failed! Retrying...

`

However, if I then turn the WiFi ON (with an external button), the join request succeeds after a few retries:

`
[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 4991 #2 6002
[D][LoRaMac.cpp:662] OnRadioTxDone(): OnRadioTxDone => TX was Join Request
[D][radio.cpp:1340] RadioIrqProcess(): RadioIrqProcess => IRQ_RX_TX_TIMEOUT
[D][LoRaMac.cpp:1225] OnRadioRxTimeout(): OnRadioRxTimeout
[D][LoRaMac.cpp:1315] OnMacStateCheckTimerEvent(): Join network failed 2 time(s)

[294487]INFO: YUBOX Framework toma control de WiFi...
[295528][D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][yubox-lorawan-SX126X.ino:92] WiFiEvent_ledStatus(): [WiFi-event] event: 0

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][yubox-lorawan-SX126X.ino:92] WiFiEvent_ledStatus(): [WiFi-event] event: 14

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][yubox-lorawan-SX126X.ino:92] WiFiEvent_ledStatus(): [WiFi-event] event: 14

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[D][yubox-lorawan-SX126X.ino:92] WiFiEvent_ledStatus(): [WiFi-event] event: 1

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][yubox-lorawan-SX126X.ino:92] WiFiEvent_ledStatus(): [WiFi-event] event: 2

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][yubox-lorawan-SX126X.ino:92] WiFiEvent_ledStatus(): [WiFi-event] event: 2

[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 4991 #2 6002
[D][LoRaMac.cpp:662] OnRadioTxDone(): OnRadioTxDone => TX was Join Request
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][yubox-lorawan-SX126X.ino:92] WiFiEvent_ledStatus(): [WiFi-event] event: 4

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.0.17, MASK: 255.255.255.0, GW: 192.168.0.1
[D][yubox-lorawan-SX126X.ino:92] WiFiEvent_ledStatus(): [WiFi-event] event: 7

[D][radio.cpp:1340] RadioIrqProcess(): RadioIrqProcess => IRQ_RX_TX_TIMEOUT
[D][LoRaMac.cpp:1225] OnRadioRxTimeout(): OnRadioRxTimeout
[D][LoRaMac.cpp:1315] OnMacStateCheckTimerEvent(): Join network failed 3 time(s)

ERR: OVER_THE_AIR_ACTIVATION failed! Retrying...
[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 4991 #2 6002
[D][LoRaMac.cpp:662] OnRadioTxDone(): OnRadioTxDone => TX was Join Request
[D][LoRaMac.cpp:712] OnRadioRxDone(): OnRadioRxDone
[D][LoRaMac.cpp:768] OnRadioRxDone(): OnRadioRxDone => FRAME_TYPE_JOIN_ACCEPT
OVER_THE_AIR_ACTIVATION != 0
INFO: enviando payload (10 bytes)... OK
[D][LoRaMac.cpp:610] OnRadioTxDone(): OnRadioTxDone
[D][LoRaMac.cpp:629] OnRadioTxDone(): OnRadioTxDone => RX Windows #1 991 #2 2002
[D][LoRaMac.cpp:712] OnRadioRxDone(): OnRadioRxDone
[D][LoRaMac.cpp:837] OnRadioRxDone(): OnRadioRxDone => FRAME_TYPE_DATA_(UN)CONFIRMED_DOWN
INFO: enviando payload (10 bytes)... OK

`

If I then turn off the WiFi while the device is joined to the network, the device keeps sending and receiving packets as normal.

What can I do to debug this issue?

beegee-tokyo, 2021-03-30:
This will be difficult to debug, there should be no dependencies between WiFi and LoRa transceiver.

I have one setup where I use ESP32 with WiFi and LoRa. The LoRa (not LoRaWAN, just P2P communcication) is initialized before WiFi and I didn't see any problems.

beegee-tokyo, 2021-03-31:
Just guessing here, but it seems that WiFi is delaying something in the SX126x library. That could explain why you have to extend the RX windows.

I remember there is `WiFi.persistent(bool persistent)` which is by default true. It stores the last WiFi credentials in NV-RAM and on restart automatically tries to connect automatically (not sure but maybe whether you start WiFi or not).

avillacis, 2021-03-31:
In my library, I run WiFi.persistent(false) as one of the very first steps in my WiFi manager constructor. I then install http route managers (but these should be passive until WiFi is available and a request comes in), and an event manager for WiFi (ditto). Only if a flag loaded from NVRAM says so, the WiFi is actually started. So WiFi is largely out of the way in the non-working case.

Is there some way to see the timestamp at which the LoRaWAN library appears to receive radio activity, AND recognizes the LoRaWAN join request as valid, without hacking further into the library? I am not sure whether the reception window delay (the 5-second delay I had to shorten) is supposed to be enforced by the networkserver (ChirpStack) or the LoRa gateway (RAK7258).

avillacis, 2021-03-31:
I have finally found the root cause of the issue. The example sketch takes 62 milliseconds between TX-Done IRQ reception and actually processing the IRQ callback. However, my sketch takes 650 milliseconds or more. This caused my sketch to miss the RX windows. However, the library should account for the lag when opening the RX windows. Pull request #34 does just that.

beegee-tokyo, 2021-04-04:
If you application blocks the **`Radio.IrqProcess`** for 650ms then there is something wrong in your code.
Your Pull Request would break backward compatibility, so I will not merge this.

I am working on a solution to take out **`Radio.IrqProcess()`** from the main application (your code) and handle it in the background without the users code being even able to block it. Testing will be done this coming week and I hope I can release a new version soon.

closed 2021-04-04 by beegee-tokyo

avillacis, 2021-04-06:
On closer examination, there indeed was an unintended delay loop in my code that ate up to one second if there was no WiFi connection. By fixing this, I was able to run the LoRaWAN join without any patches. However other projects might legitimately require lengthy processing before having the opportunity to process radio interrupts. So the background-task solution by @beegee-tokyo is still necessary in some form.

beegee-tokyo, 2021-04-07:
I have first test results and it looks good. I will keep you informed.

If you know Git, how to clone a library and how to manually install a library, you can try the [bg_handling](https://github.com/beegee-tokyo/SX126x-Arduino/tree/bg_handling) branch of this library.