beegee-tokyo/SX126x-Arduino #37: pingpong example Taking initial time!

Ahsan-Averos, 2021-04-14:
Hello Sir, why in pingpong example it is almost take more than 5 minutes to setup tx and rx.
where to reduce this initial time.
12:08:23.947 -> MCU Espressif ESP32
12:08:23.947 -> BoardId: 00-00-2C-A7-58-C4-0A-24
12:08:23.947 -> Starting lora_hardware_init
12:08:23.994 -> <BRD> SyncWord = 2414
12:08:23.994 -> <BRD> LoRa Task started
**12:08:24.040 -**> Starting Radio.Rx
Then it is start on :

**12:13:11.248** -> OnRxTimeout
12:13:11.761 -> CAD returned channel free after 13ms
12:13:11.761 -> Sending a PING in OnCadDone as Master
12:13:11.898 -> OnTxDone
12:13:12.457 -> OnRxDone
12:13:12.457 -> RssiValue=-15 dBm, SnrValue=12
12:13:12.457 -> 50 4F 4E 47 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 
12:13:12.457 -> Received a PONG in OnRxDone as Master
12:13:12.967 -> CAD returned channel free after 13ms
12:13:12.967 -> Sending a PING in OnCadDone as Master

Ahsan-Averos, 2021-04-14:
Please see above logging.
it setup from 12.08.24.040-> Starting Radio.RX
and start on 12.13.11.248-> OnRxTimeout
how to reduce this time, I did not find in code.
Although you mention RXTIMEOUT value is 3000 but it does not impact on this.
Thanks
I will wait for our reply!

beegee-tokyo, 2021-04-23:
Don't know, but it does not happen on nRF52 MCU. Could be ESP32 WiFi initialization running in the background.
[image]

closed 2021-04-23 by beegee-tokyo