beegee-tokyo/SX126x-Arduino #38: LORAMAC_STATUS_BUSY

Ahsan-Averos, 2021-04-21:
Hello Guys,
I am using NRF52832 DK with E22900M22S module, I used LORAWAN example, its work fine for initial 20 packet, but after that serial monitor show this below message:
_12:15:37.870 -> <LM> LoRaMacMcpsRequest LORAMAC_STATUS_BUSY
12:15:37.917 -> <LMH> lmh_send -> LoRaMacMcpsRequest failed
12:15:37.917 -> lmh_send result -1_
Although multiple time I restart my node but still I received this message, someone can helping out why it happening.
I used my own GATEWAY.
Thanks

beegee-tokyo, 2021-04-21:
Without additional information there is no answer.

What Region are you using?
What is your packet size?
Do you have ADR enabled?
What is the time between sending packets?
Are you using confirmed or unconfirmed packets?
Is there a downlink packet sent to the node from the gateway and if yes, how often?

Averosiation, 2021-04-21:
Dear beegee-tokyo
Thanks for your response, I am same person from other email ID. Please have a look your question.
By the way I am directly use your _Lorawan.ino code_ from example folder.

_What Region are you using?_
I am using EU868 Europe Region.
(I made some changes in RX window 800ms from 3000ms but still have a problem)

_What is your packet size?_
As per given example (Lorawan.ino) which is simple HELLOWORLD packet size is 20 byte.
(I tried to played with  packet size as well as Header which is fixed/variable I changed to variable but does not happen)

_Do you have ADR enabled?_
Yes ADR is enabled, but I also check with disabled but same issue.

_What is the time between sending packets?_
It was 30 sec _LORAWAN_APP_TX_DUTYCYCLE 30000_ 

_Are you using confirmed or unconfirmed packets?_
Unconfirmed Messages.

_Is there a downlink packet sent to the node from the gateway and if yes, how often?_
No.

I will waiting for your reply. Thanks
I am using same code first for testing. Below link is your GitHub link of code.
_**https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/examples/LoRaWan/LoRaWan.ino**_

beegee-tokyo, 2021-04-23:
I did not much testing on EU868, because this band is not legal in my country.
But I did a quick test this morning, and can confirm your problem.

Problem can be solved by changing line 76 in the example code.
Instead of **`LORAWAN_ADR_ON`** use **`LORAWAN_ADR_OFF`** and the problem is gone.

```cpp
/**@brief Structure containing LoRaWan parameters, needed for lmh_init()
*/
static lmh_param_t lora_param_init = {LORAWAN_ADR_OFF, LORAWAN_DEFAULT_DATARATE, LORAWAN_PUBLIC_NETWORK, JOINREQ_NBTRIALS, LORAWAN_DEFAULT_TX_POWER, LORAWAN_DUTYCYCLE_OFF};
```

It is not happening on AS923 which is the legal band for my country.

closed 2021-04-23 by beegee-tokyo

Ahsan-Averos, 2021-04-25:
Dear Sir,
As I told you that  I already changed **LORAWAN_ADR_ON** to **LORAWAN_ADR_OFF** but did not do anything for me.

_Do you have ADR enabled?
Yes ADR is enabled, but I also check with disabled but same issue._

Yes I understand due to ADR_ON server taken first 20 message and then command to Node to manage _dynamically optimizes data rate, airtime, and energy consumption._ Although I was assume when I restart both NRF52 and E22 module so  it restart it cycle with server but does not and stuck on this:
_12:15:37.870 -> LoRaMacMcpsRequest LORAMAC_STATUS_BUSY
12:15:37.917 -> lmh_send -> LoRaMacMcpsRequest failed
12:15:37.917 -> lmh_send result -1_
Do you have any idea why it stuck at this point, why it still in busy state? may be I am doing something wrong. By the why I am using my own gateway!
Thanks