beegee-tokyo/SX126x-Arduino #41: Battery level in WisBlock RAK4630

Ganesh-Shanbhag-face, 2021-06-14:
Hi why the BoardGetBatteryLevel in the src/boards/mcu/nrf52832 is not implemented yet?

What is the significance of BoardGetRandomSeed function?

beegee-tokyo, 2021-06-15:
@Ganesh-Shanbhag-face 
The BoardGetBatteryLevel is not implemented because the library is written for many boards that use nRF52 or ESP32 MCU's. Some boards don't have an option to read the battery level, and different boards use different analog inputs to read the battery level.

If using LoRaWAN communication, a custom battery level function should be created and added the LoRaWAN callbacks in the structure lmh_callback_t. You need to replace BoardGetBatteryLevel with your own function.
```c++
static lmh_callback_t lora_callbacks = {BoardGetBatteryLevel, BoardGetUniqueId, BoardGetRandomSeed,
			lorawan_rx_handler, lorawan_has_joined_handler, 
			lorawan_confirm_class_handler, lorawan_join_failed_handler};
```
----

BoardGetRandomSeed  is only used if a random device address should be assigned to the device instead of setting the device address with `lmh_setDevAddr()`. The option to use a random device address can enabled by changing `#define STATIC_DEVICE_ADDRESS 1` to `#define STATIC_DEVICE_ADDRESS 0` in Commissioning.h

closed 2021-06-15 by beegee-tokyo

Ganesh-Shanbhag-face, 2021-06-15:
@beegee-tokyo Thanks for the inputs and clarification.