beegee-tokyo/SX126x-Arduino #43: Disable debug logs

jpmeijers, 2021-06-21:
Using the DeepSleep example I am seeing a Serial print:
```
16:18:30.751 -> <LM> OnRadioRxDone
16:19:25.634 -> <LM> OnRadioRxDone => FRAME_TYPE_JOIN_ACCEPT
16:19:26.197 -> <LM> OnRadioTxDone
16:19:31.178 -> <LM> OnRadioTxDone => RX Windows #1 4989 #2 5995
16:19:35.163 -> <RADIO> RadioIrqProcess => IRQ_RX_TX_TIMEOUT
16:19:35.163 -> <LM> OnRadioRxTimeout
```

Looking at the source for this library I see logging is enabled by default, and one can disable it by setting `LIB_DEBUG` to 0.

In the DeepSleep example, in main.h, I have added the following code after the MAX_SAVE define.

```
// Comment the next line if you want DEBUG output. But the power savings are not as good then!!!!!!!
#define MAX_SAVE

#ifdef MAX_SAVE 
#define LIB_DEBUG 0
#endif
```

This however does not work. 

Also the compiler does not like it:
```
In file included from /home/jpmeijers/Izinto/Arduino Code/RAK4631-DeepSleep-LoRaWan-GX701P-new/RAK4631-DeepSleep-LoRaWan-GX701P-new.ino:24:
/tmp/arduino_build_880825/sketch/main.h:20: warning: "LIB_DEBUG" redefined
   20 | #define LIB_DEBUG 0
      | 
In file included from /home/jpmeijers/Arduino/libraries/SX126x-Arduino/src/boards/sx126x/sx126x-board.h:42,
                 from /home/jpmeijers/Arduino/libraries/SX126x-Arduino/src/boards/mcu/board.h:50,
                 from /home/jpmeijers/Arduino/libraries/SX126x-Arduino/src/LoRaWan-Arduino.h:15,
                 from /home/jpmeijers/Arduino/libraries/SX126x-Arduino/src/LoRaWan-RAK4630.h:4,
                 from /tmp/arduino_build_880825/sketch/main.h:14,
                 from /home/jpmeijers/Izinto/Arduino Code/RAK4631-DeepSleep-LoRaWan-GX701P-new/RAK4631-DeepSleep-LoRaWan-GX701P-new.ino:24:
/home/jpmeijers/Arduino/libraries/SX126x-Arduino/src/sx126x-debug.h:18: note: this is the location of the previous definition
   18 | #define LIB_DEBUG 1
```

Changing the code in main.h to read as follows makes the compiler warning go away, but still lets the library print to Serial:
```
#define MAX_SAVE

#ifdef MAX_SAVE 
#undef LIB_DEBUG
#define LIB_DEBUG 0
#endif
```

I guess we'll need to have a similar check and define rather in the library than in the user code. I'll test and report back.

jpmeijers, 2021-06-21:
Moving `#include <LoRaWan-RAK4630.h>` to after `#define LIB_DEBUG 0` also does not seem to solve it for me.

For now I'm changing the value of LIB_DEBUG in sx126x-debug.h, as the comments at the top of that file suggests one do when using Arduino IDE.

beegee-tokyo, 2021-06-21:
Having the debug enabled by default was a bad mistake. In the next release, the debug will be disabled.

closed 2021-06-21 by beegee-tokyo