beegee-tokyo/SX126x-Arduino #48: Wish list item: STM32WL support

jpmeijers, 2021-09-28:
According to https://www.st.com/en/microcontrollers-microprocessors/stm32wl-series.html#overview the 
> STM32WL microcontrollers feature a sub-GHz radio based-on Semtech SX126x

The STM32 Arduino core is adding support for this series too, and I have been able to program a blink example to my Seeed Studio LORA E5 module.
https://github.com/stm32duino/Arduino_Core_STM32

I am wondering if it would be possible, and if it is, how much effort it would be to get this library to work on the STM32WL series of microcontrollers?

beegee-tokyo, 2021-10-03:
Hello @jpmeijers 
Sorry for the late answer. 
I do not have much experience with the STM32 MCU's and right now I have no time to look into the STM32WL5xx. It should be possible as the STM32 Arduino Core is Mbed and two months ago I adapted the library for the RP2040 where the Arduino Core is Mbed as well. 

We have a big product launch at RAK in October, maybe I find time after that.

jpmeijers, 2021-10-19:
Thanks for the update @beegee-tokyo. If you happen to find a way to use the STM32WLE5 that is not as complicated as STM32Cube, please let us know. MBed seems ok, but support for other peripherals are limited. Ideally one would want Arduino support.