beegee-tokyo/SX126x-Arduino #50: Eror "lmh_send result 1"

Muuhoooo, 2021-11-25:
I am utilizing ESP32_Wrover Module and E22_900M_22S (sx126x) Module. I  have tried OTAA with Helium Console. I  " #define OVER_THE_AIR_ACTIVATION 1"  and " #define STATIC_DEVICE_EUI 1" instead of 0 in "Commissioning.h" . Also I reduced the size of payload.I am sending only "HELLO". And I have eror. I see "lmh_send_result 1" in serial
[image]
.
[image]

beegee-tokyo, 2021-11-25:
Hello @Muuhoooo 

What are you LoRaWAN settings?
ADR enabled/disabled?
Datarate?
Duty cycles enabled or disabled?

How often do you try to send the packet?

Muuhoooo, 2021-11-25:
Hello, I'm trying to send the word "hello" to the helium console for now.

I used  default setting for DUTYCYCLE as enable. 
#define LORAWAN_DUTYCYCLE_ON true	/**< LoRaWAN duty cycle enabled */
#define LORAWAN_DUTYCYCLE_OFF false /**< LoRaWAN duty cycle disabled */    in  "Commissioning.h".

#define LORAWAN_APP_TX_DUTYCYCLE 10000 
#define APP_TX_DUTYCYCLE_RND 1000  in "LoRaWAN.ino"   

I change ADR as disable.
 #define LORAWAN_ADR_ON false	  
#define LORAWAN_ADR_OFF true     in  "Commissioning.h". 
and I change Data Rate 
#define LORAMAC_DEFAULT_DATARATE DR_2 
#define LORAMAC_DEFAULT_TX_POWER TX_POWER_20_DBM in "LoRaMac-definitions.h" and LoRaMacRegion_t region = LORAMAC_REGION_EU868.   But I have error" lmh_send result 1 " on serial. Also I saw Network Joined  but my Helium Console I did not see anything. What is the promblem. What am I doing wrong? What  shoud I change?

beegee-tokyo, 2021-11-25:
First of all, you should not touch any file inside the SX126x-Arduino library. All settings that can be set by the application are available through the API.

Which of the example codes did you use?
Did you use unique DevEUI, AppEUI and AppKey and created a device in your Helium Console with the matching EUI's and Keys?

**`lmh_send`** returning **1** means the LoRa transceiver is still busy with some tasks. Try to change the send frequency to 30000ms to something higher. A LoRaWAN packet transmission can take up to 7 seconds. You might be sending packets too often.

Muuhoooo, 2021-11-25:
First of all, thank you for your comments. 

I change inside SX126x-Arduino library because ı did not found change this parameter  example code. 
I use  " https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/examples/LoRaWan/LoRaWan.ino " 
and I created new device  my Helium Console and checked a lot of time this is same.   

I didnot know what was lmh_send returning 1 thank you so much. How can ı change my frequency I utilized EU868MHz.But I  can change dutycycle.

Muuhoooo, 2021-11-25:
[LoRaWan.zip](https://github.com/beegee-tokyo/SX126x-Arduino/files/7602928/LoRaWan.zip)
 If you wonder this is my code.

beegee-tokyo, 2021-11-25:
The examples are a little bit outdated, but I didn't find time to update them.
The region is now set with `lmh-init()` check here the syntax: https://github.com/beegee-tokyo/SX126x-Arduino#initialize.

Muuhoooo, 2021-11-25:
I change dutycycle time LORAWAN_APP_TX_DUTYCYCLE 60000 and APP_TX_DUTYCYCLE_RND 10000 on LoRaWAN.ino code not Sx1262-library and I set lmh-init() yours link but I didnot change anything. And I just have created new device on my helium console and use new  unique DevEUI, AppEUI and AppKey. I joined network as you can see in the picture below. But I didnot active my  device on helium console.     
[image]
[image]
.

beegee-tokyo, 2021-11-26:
This is wrong:
[image]
It makes no sense what you type here and throws a compilation error.    
Correct would be
```c++
err_code = lmh_init(&lora_callbacks, lora_param_init, true, CLASS_A, LORAMAC_REGION_EU868);
```

Works fine for me with the same example code, but I do not have ESP32 and the eByte module available, so I tested with some WisBlock modules.
Serial Log:
[image]

Log from Helium Console:
[image]

Muuhoooo, 2021-11-26:
I changed as "err_code = lmh_init(&lora_callbacks, lora_param_init, true, CLASS_A, LORAMAC_REGION_EU868);" but this time i got error "OVER_THE_AIR_ACTIVATION failed!","Check your EUI's and Keys's!","Check if a Gateway is in range!" .I know there is  no problem with gateway. I used DeviceEUI (lsb)-AppEUI (lsb) and AppKey(msb) . lsb-lsb-msb is it correct? or should i change?

beegee-tokyo, 2021-11-27:
All EUI and Keys are MSB!

Muuhoooo, 2021-11-27:
I tried all eui and keys are msb but it didnot change anything. I saw my serial "OVER_THE_AIR_ACTIVATION failed!","Check your EUI's and Keys's!","Check if a Gateway is in range!" but I know eui and keys true and no problem gateway range because I tried with sx1262 and arduino MO processor and I saw data on Helium Network but I change proccessor ESP32 Wrover B  I saw this error. I did not understand what is wrong. I tried TTN it is same. İs there any suggestion?

Muuhoooo, 2021-11-27:
thank you for your help so far 🙏

Muuhoooo, 2021-11-27:
#include <Arduino.h>
#include <LoRaWan-Arduino.h>
#include <SPI.h>

#define SCHED_MAX_EVENT_DATA_SIZE  APP_TIMER_SCHED_EVENT_DATA_SIZE /**< Maximum size of scheduler events. */
#define SCHED_QUEUE_SIZE 60                      /**< Maximum number of events in the scheduler queue. */

#define LORAWAN_APP_DATA_BUFF_SIZE 64  /**< Size of the data to be transmitted. */
#define LORAWAN_APP_INTERVAL 20000   /**< Defines the application data transmission duty cycle. 10s, value in [ms]. */

#define LORAWAN_DEFAULT_DATARATE DR_2                   /*LoRaMac datarates definition, from DR_0 to DR_5*/
#define LORAWAN_DEFAULT_TX_POWER TX_POWER_5  
#define JOINREQ_NBTRIALS 3 
DeviceClass_t CurrentClass = CLASS_A;          /* class definition*/
LoRaMacRegion_t CurrentRegion = LORAMAC_REGION_EU868;    /* Region:EU868*/
lmh_confirm CurrentConfirm = LMH_UNCONFIRMED_MSG;   

hw_config hwConfig;

//#ifdef ESP32
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 32;   // LORA RESET
int PIN_LORA_NSS = 15;   // LORA SPI CS
int PIN_LORA_SCLK = 14;  // LORA SPI CLK
int PIN_LORA_MISO = 12;  // LORA SPI MISO
int PIN_LORA_DIO_1 = 4; // LORA DIO_1
int PIN_LORA_BUSY = 33;  // LORA SPI BUSY
int PIN_LORA_MOSI = 13;  // LORA SPI MOSI
int RADIO_TXEN = -1;   // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;   // LORA ANTENNA RX ENABLE

// Foward declaration
static void lorawan_has_joined_handler(void);
static void lorawan_rx_handler(lmh_app_data_t *app_data);
static void lorawan_confirm_class_handler(DeviceClass_t Class);
static void lorawan_join_failed_handler(void);
static void send_lora_frame(void);
static uint32_t timers_init(void);
void lmh_setSingleChannelGateway(uint8_t userSingleChannel, int8_t userDatarate);

// APP_TIMER_DEF(lora_tx_timer_id);                                              ///< LoRa tranfer timer instance.
TimerEvent_t appTimer;                              ///< LoRa tranfer timer instance.
static uint8_t m_lora_app_data_buffer[LORAWAN_APP_DATA_BUFF_SIZE];        ///< Lora user application data buffer.
static lmh_app_data_t m_lora_app_data = {m_lora_app_data_buffer, 0, 0, 0, 0}; ///< Lora user application data structure.

/**@brief Structure containing LoRaWan parameters, needed for lmh_init()
*/
static lmh_param_t lora_param_init = {LORAWAN_ADR_OFF,LORAWAN_DEFAULT_DATARATE, LORAWAN_PUBLIC_NETWORK,JOINREQ_NBTRIALS, LORAWAN_DEFAULT_TX_POWER , LORAWAN_DUTYCYCLE_OFF};

/**@brief Structure containing LoRaWan callback functions, needed for lmh_init()
*/

static lmh_callback_t lora_callbacks = {BoardGetBatteryLevel, BoardGetUniqueId, BoardGetRandomSeed,
                    lorawan_rx_handler, lorawan_has_joined_handler, lorawan_confirm_class_handler, lorawan_join_failed_handler};

#define LED_BUILTIN 0

uint8_t nodeDeviceEUI[8] = {0x80, 0x8F, 0x04, 0xD0, 0x7E, 0xD5, 0xB3, 0x70};        // {0x60, 0x81, 0xF9, 0x3B, 0x7A, 0x46, 0xEA, 0xFF};

uint8_t nodeAppEUI[8] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};         // {0x60, 0x81, 0xF9, 0x86, 0x04, 0x39, 0x17, 0x32};

uint8_t nodeAppKey[16] = {0x05, 0x90, 0xF4, 0x3F, 0x5F, 0xB0, 0xE4, 0x85, 0x77, 0xCB, 0x97, 0x7E, 0xB1, 0x45, 0x96, 0x99};         // {0x30, 0x97, 0xD4, 0xAC, 0x4C, 0x8A, 0x8B, 0xE9, 0x6B, 0x69, 0x56, 0x29, 0x85, 0x3F, 0xFE, 0xBE};

void setup()
{ delay(5000);
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  // Define the HW configuration between MCU and SX126x
  hwConfig.CHIP_TYPE = SX1262_CHIP;     // Example uses an eByte E22 module with an SX1262
[... 161 lines cut ...]
  lmh_error_status error = lmh_send(&m_lora_app_data, CurrentConfirm);
  Serial.printf("lmh_send result %d\n", error);
}

/**@brief Function for handling a LoRa tx timer timeout event.
*/
static void tx_lora_periodic_handler(void)
{
  TimerSetValue(&appTimer, LORAWAN_APP_INTERVAL);
  TimerStart(&appTimer);
  Serial.println("Sending frame");
  send_lora_frame();
}

/**@brief Function for the Timer initialization.
   @details Initializes the timer module. This creates and starts application timers.
*/
static uint32_t timers_init(void)
{
  appTimer.timerNum = 3;
  TimerInit(&appTimer, tx_lora_periodic_handler);
  return 0;
}

This is my code.Esp32 pins configuration are true.I checked. Please let me know if you see anything wrong. I've been getting the same error for days. "OVER_THE_AIR_ACTIVATION failed!","Check your EUI's and Keys's!","Check if a Gateway is in range!"

beegee-tokyo, 2021-11-28:
How did you connect the RXEN and TXEN pins of the E22-900M22?
[image]

In you code you show
```c++
int RADIO_TXEN = -1; // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1; // LORA ANTENNA RX ENABLE
```
That will not work. The antenna switch inside the module needs to controlled with this two pins. The library can take care of it, but with above settings, it assumes you are doing it (bad idea).

Without controlling the antenna switch, you might be able to send, but you will never receive the Join Accept packet.

Muuhoooo, 2021-11-28:
I used Arduino M0 and
![Ekran Alıntısı](https://user-images.githubusercontent.com/48700516/143731533-439481d0-c220-
[image]
498e-a914-63877ff062c1.PNG)
 SX1262 a did not use rxen and I did not connected rxen, I just connected txen ıo2 and tried but same result. And I did not rxen and txen but same result.

Muuhoooo, 2021-11-28:
I did not connected rxen, I just connected txen ıo2 and tried but same result. And I did not rxen and txen but same result.
 
[image]

[image]

beegee-tokyo, 2021-11-28:
I designed this [Alora board](https://circuit.rocks/product:2685) which uses the same eByte module.

I can tell you, you have to control both RXen and TXen pins, or the antenna switch does not work,
Unless you do not do so, you will never succeed.

valioiv, 2021-11-28:
@beegee-tokyo , just a general comment because recently I observe kind if excuses with the examples here and there under some of the issues.... You should always find time to constantly keep examples up to date just because they are the very essential part for the end user. No one is digging deeply in the library's code to understand how mechanics work. They just want to have clear API and clear example test apps which work at 100%. If you don't have bandwidth to maintain an example or two just delete them in order to not mislead a developer. Will save tons of hours of debugging and rising new "issues" (which are probably not real issues).
I apologize if the situation is different but put yourself of the other side. Don't understand me wrongly - the library is great, may be the best one for SX126x let's keep it like that! Your efforts are great! May be I can help with the LLCC68, will see how it goes. Thank you!

beegee-tokyo, 2021-11-28:
@valioiv 
I am working for a company in customer support and I know how difficult it is.
But this is _**FREE PROVIDED OPEN SOURCE LIBRARY**_ so it is as it is.
Your inputs, improvements and pull requests are very welcome.

closed 2021-11-28 by beegee-tokyo

Muuhoooo, 2021-11-28:
I know you are so busy and I really thank you so much. 
I soldered the pin. And I checked with a multimeter. No problem.
int RADIO_TXEN = 2;   // LORA ANTENNA TX ENABLE
int RADIO_RXEN = 0;   // LORA ANTENNA RX ENABLE

 hwConfig.RADIO_TXEN = RADIO_TXEN;     // LORA ANTENNA TX ENABLE
 hwConfig.RADIO_RXEN = RADIO_RXEN;     // LORA ANTENNA RX ENABLE
 
 But noting changed very interesting. Do you have any suggestions, I applied everything you said.

beegee-tokyo, 2021-11-28:
Running out of ideas.

RXen/TXen was the last thing to check I can think of. 

I started this library for nRF52 and added ESP32 later. Had no problems so far with it. 

Obviously the general communication between the ESP32 and the SX1262 (eByte module) works. So it can only be the control of the antenna switch or the antenna itself.

Do you have a second eByte module for testing?

Muuhoooo, 2021-11-28:
Actually this eByte have tested lora p2p communication no problem. But I  have second eByte module for testing. I will try during a day and will provide feed back.

Muuhoooo, 2021-11-29:
Hi Birnd Giesecke, I tried new eByte and new ESP32 Wrover B module but same result. Thank you for all your help.

DanielBustillos, 2022-12-28:
Hi @beegee-tokyo first of all this is a great piece of work.

I'm struggling make it work an esp32 with the helium network. I've tried to solve it by my own following several issues with no luck as I get an "OVER_THE_AIR_ACTIVATION failed!"

 I know that you are lack on time, but could you share a diagram or picture of how to connect the module to the mcu? Specifically the VCC, GND and ANT, I've connected all the pins (vc to 3v3 and gig to gng) except ANT but I don't know if that is correct.

This is my wiring at the moment:
int PIN_LORA_RESET = 4;	 // LORA RESET
int PIN_LORA_NSS = 5;	 // LORA SPI CS
int PIN_LORA_SCLK = 18;	 // LORA SPI CLK
int PIN_LORA_MISO = 19;	 // LORA SPI MISO
int PIN_LORA_DIO_1 = 21; // LORA DIO_1
int PIN_LORA_BUSY = 22;	 // LORA SPI BUSY
int PIN_LORA_MOSI = 23;	 // LORA SPI MOSI
int RADIO_TXEN = 26;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = 27;	 // LORA ANTENNA RX ENABLE

Thanks 😀

beegee-tokyo, 2022-12-28:
Is this for an eByte E22 module?

If you control RX and TX with GPIO's you need to set
hwConfig.USE_DIO2_ANT_SWITCH = false;
For the E22 you need to enable DIO3 to control the TCXO power
hwConfig.USE_DIO3_TCXO = true;

For the pin connections, I cannot tell you if it is correct, but I do not see any problems in the GPIO selection.

DanielBustillos, 2022-12-29:
Thank you for your response, I've been trying to figure it out why I can't join successfully. I've seen many issues and soldered a new ebyte module( e22 900m30s). I don't understand the RX, and TX configuration nor antenna switches. May be the issue us coming from it?

I'm using thins pinout and hw configuration (I've tried many configurations of the last 5 parameters on hwconfig) 

`int PIN_LORA_RESET = 4;	 // LORA RESET
int PIN_LORA_NSS = 5;	 // LORA SPI CS
int PIN_LORA_SCLK = 18;	 // LORA SPI CLK
int PIN_LORA_MISO = 19;	 // LORA SPI MISO
int PIN_LORA_DIO_1 = 21; // LORA DIO_1
int PIN_LORA_BUSY = 22;	 // LORA SPI BUSY
int PIN_LORA_MOSI = 23;	 // LORA SPI MOSI
int RADIO_TXEN = 26;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = 27;	 // LORA ANTENNA RX ENABLE`

`	hwConfig.CHIP_TYPE = SX1262_CHIP;		  // Example uses an eByte E22 module with an SX1262
	hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // LORA RESET
	hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;	  // LORA SPI CS
	hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;	  // LORA SPI CLK
	hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;	  // LORA SPI MISO
	hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // LORA DIO_1
	hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;	  // LORA SPI BUSY
	hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;	  // LORA SPI MOSI
	hwConfig.RADIO_TXEN = RADIO_TXEN;              // LORA ANTENNA TX ENABLE (e.g. eByte E22 module)
	hwConfig.RADIO_RXEN = RADIO_RXEN;              // LORA ANTENNA RX ENABLE (e.g. eByte E22 module)
	hwConfig.USE_DIO2_ANT_SWITCH = true							;                         // LORA DIO2 does not control antenna
	hwConfig.USE_DIO3_TCXO = hwConfig.USE_DIO3_TCXO;  // LORA DIO3 controls oscillator voltage (e.g. eByte E22 module)
	hwConfig.USE_DIO3_ANT_SWITCH = false;                         // LORA DIO3 does not control antenna`

Could you help me verifying if they are ok? I can share you mi code if you wish. Thank you.

If I make this work, I can make a PR with an updated example for esp32 with e22 900m30s and Helium network 😃

beegee-tokyo, 2022-12-29:
It is a long time since I worked with the eByte modules and I never understood why they needed two pins for the antenna switch control.
What I remember:

| RX Pin level | TX Pin level | Antenna direction |
| --- | --- | --- |
| 0 | 0 | Antenna switch off |
| 1 | 0 | Antenna switch in RX mode |
| 0 | 1 | Antenna switch in TX mode |
| 1 | 1 | not allowed |

For Helium and US915, you need as well to select subband 2 after you called lmh_init() like this:

```cpp
lmh_setSubBandChannels(2);
```

DanielBustillos, 2022-12-29:
Is working now, thank you!

I can make an PR with an example for using with the library with esp32 + e22 900m30s, this may facilitate the usage of this awesome library. Would you be interested on this?

On the other side, how did you know that I had to select subband 2? I would like to understand better this, do you have a reference that you would recommend?

For now I share my working code here, in case anyone is interested: [esp32+e22.zip](https://github.com/beegee-tokyo/SX126x-Arduino/files/10317769/esp32%2Be22.zip)

beegee-tokyo, 2022-12-29:
Yes, please issue an PR, I am happy to give others more options.