beegee-tokyo/SX126x-Arduino #51: Timeout Waiting For BUSY Now

Muuhoooo, 2021-12-01:
Hi Bernd, I hope you're fine. I try send "Helllo" over LoRaWAN wşth ESP32. But I saw serial "timeout waiting for BUSY now". I defined txen and rxen. And I send picture and LoraWAN settings. What should i change, this is the first time i saw such an error. 
ADR  : Disabled
Datarate : DR_2
Duty cycles : Enabled 
Region : LORAMAC_REGION_EU868
LORAWAN_APP_INTERVAL 20000 
LMH_UNCONFIRMED_MSG;

[image]

beegee-tokyo, 2021-12-02:
@Muuhoooo 
You closed it, so I guess you solved it.

`Timeout waiting for busy` is caused by the SPI communication not working or Busy pin of SX1262 not connected or defined incorrect.

Muuhoooo, 2021-12-02:
Hi Bernd, I saw another issue and I tried but did not solved. I checked the spi and all the other pins connected to the sx1262 with a multimeter.All of them are all connected and defined in my code. Again, an interesting error could not make sense.

//#ifdef ESP32
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 32;   // LORA RESET
int PIN_LORA_NSS = 15;   // LORA SPI CS
int PIN_LORA_SCLK = 14;  // LORA SPI CLK
int PIN_LORA_MISO = 12;  // LORA SPI MISO
int PIN_LORA_DIO_1 = 4; // LORA DIO_1
int PIN_LORA_BUSY = 33;  // LORA SPI BUSY
int PIN_LORA_MOSI = 13;  // LORA SPI MOSI
int RADIO_TXEN = 2; 
int RADIO_RXEN = 25;

  // Define the HW configuration between MCU and SX126x
  hwConfig.CHIP_TYPE = SX1262_CHIP;     // Example uses an eByte E22 module with an SX1262
  hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // LORA RESET
  hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;   // LORA SPI CS
  hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;   // LORA SPI CLK
  hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;   // LORA SPI MISO
  hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // LORA DIO_1
  hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;   // LORA SPI BUSY
  hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;   // LORA SPI MOSI
  hwConfig.RADIO_TXEN = RADIO_TXEN;     // LORA ANTENNA TX ENABLE
  hwConfig.RADIO_RXEN = RADIO_RXEN;     // LORA ANTENNA RX ENABLE

Muuhoooo, 2021-12-02:
I tried new SX1262 and ESP32 and I checked all pins and same result. "Timeout Waiting For BUSY now"

beegee-tokyo, 2021-12-02:
Can you retry your LoRa P2P test that you mentioned earlier.

I have an ESP32 working with an SX1262 (not eByte) without problems using LoRaWAN and sending packets. It's up since weeks. And another ESP32 connected to the same eByte module working on LoRa P2P working since months.

I just cannot take them out for tests, because they are part of my home control system.

So I am not sure how to solve your problem, but I am quite sure it is not the library.

Muuhoooo, 2021-12-02:
Okey understand. Thank you very much for your help.

Muuhoooo, 2021-12-02:
And LoRa P2P test is successful. I have seen serial screen "Hello World".

beegee-tokyo, 2021-12-03:
Digged in my "old devices cabinet" and found my Adafruit ESP32 Feather and the Alora1262 Feather (which has the eByte module on it):
[image]
[image]

Log output on the USB:
```
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
=====================================
SX126x LoRaWan test
=====================================
Start Join request
OVER_THE_AIR_ACTIVATION != 0
Sending frame
Data: {"i":1,"n":000}
Size: 15
Port: 2
lmh_send result 0
Uncomfirmed TX finished
Sending frame
Data: {"i":1,"n":001}
Size: 15
Port: 2
lmh_send result 0
Uncomfirmed TX finished
Sending frame
Data: {"i":1,"n":002}
Size: 15
Port: 2
lmh_send result 0
```
Log from Chirpstack LNS:
[image]

My code:
```cpp
#include <Arduino.h>

#include <LoRaWan-Arduino.h>
#include <SPI.h>
#include <Ticker.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
WiFiMulti wifiMulti;

// #define SCHED_MAX_EVENT_DATA_SIZE APP_TIMER_SCHED_EVENT_DATA_SIZE /**< Maximum size of scheduler events. */
// #define SCHED_QUEUE_SIZE 60										  /**< Maximum number of events in the scheduler queue. */

#define LORAWAN_APP_DATA_BUFF_SIZE 256  /**< Size of the data to be transmitted. */
#define LORAWAN_APP_TX_DUTYCYCLE 30000 /**< Defines the application data transmission duty cycle. 30s, value in [ms]. */
#define APP_TX_DUTYCYCLE_RND 1000	  /**< Defines a random delay for application data transmission duty cycle. 1s, value in [ms]. */
#define JOINREQ_NBTRIALS 3			   /**< Number of trials for the join request. */

hw_config hwConfig;

// ESP32 Feather - SX126x pin configuration
int PIN_LORA_RESET = 32; //16; // LORA RESET
int PIN_LORA_DIO_1 = 14; //14; // LORA DIO_1
int PIN_LORA_BUSY = 27;  //33; // LORA SPI BUSY
int PIN_LORA_NSS = 33;   //27; // LORA SPI CS
int PIN_LORA_SCLK = 5;   // LORA SPI CLK
int PIN_LORA_MISO = 19;  // LORA SPI MISO
int PIN_LORA_MOSI = 18;  // LORA SPI MOSI
[... 330 lines cut ...]
	ledTicker.once(1, ledOff);
}

/**@brief Function for handling a LoRa tx timer timeout event.
 */
static void tx_lora_periodic_handler(void)
{
	TimerSetValue(&appTimer, LORAWAN_APP_TX_DUTYCYCLE);
	TimerStart(&appTimer);
	Serial.println("Sending frame");
	send_lora_frame();
}

/**@brief Function for the Timer initialization.
 *
 * @details Initializes the timer module. This creates and starts application timers.
 */
static uint32_t timers_init(void)
{
	appTimer.timerNum = 3;
	TimerInit(&appTimer, tx_lora_periodic_handler);
	return 0;
}

```

Muuhoooo, 2021-12-03:
Hi Bernd, I tried updating  your code. I finally made it.ESP32 Wrover B module +EBYTE 22 +LoRaWAN  OTAA with Helium Console.Thank you for all your help.
[image]

Muuhoooo, 2021-12-03:
#include <Arduino.h>
#include <SX126x-Arduino.h>
#include <LoRaWan-Arduino.h>
#include <SPI.h>

#define LORAWAN_APP_DATA_BUFF_SIZE 256  /**< Size of the data to be transmitted. */
#define LORAWAN_APP_TX_DUTYCYCLE 30000 /**< Defines the application data transmission duty cycle. 30s, value in [ms]. */
#define APP_TX_DUTYCYCLE_RND 1000   /**< Defines a random delay for application data transmission duty cycle. 1s, value in [ms]. */
#define JOINREQ_NBTRIALS 3         /**< Number of trials for the join request. */

hw_config hwConfig;

// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 32;   // LORA RESET
int PIN_LORA_NSS = 15;   // LORA SPI CS
int PIN_LORA_SCLK = 14;  // LORA SPI CLK
int PIN_LORA_MISO = 12;  // LORA SPI MISO
int PIN_LORA_DIO_1 = 4; // LORA DIO_1
int PIN_LORA_BUSY = 33;  // LORA SPI BUSY
int PIN_LORA_MOSI = 13;  // LORA SPI MOSI
int RADIO_TXEN = 2;   //2 LORA ANTENNA TX ENABLE ,issues kısmında bu txen ve rxen kesinlikle tanımlı olması gerektiğini söylemişler yoksa asla başarılı olamayacağını bildirmiişler. 
int RADIO_RXEN = 25;   //25 LORA ANTENNA RX ENABLE ,Anten anahtarını kontrol etmeden gönderebilirsiniz,Join Accept  paketini asla alamazsınız.

uint8_t nodeDeviceEUI[8] = {0x60, 0x81, 0xF9, 0x3B, 0x7A, 0x46, 0xEA, 0xFF}; // msb 

uint8_t nodeAppEUI[8] = {0x60, 0x81, 0xF9, 0x86, 0x04, 0x39, 0x17, 0x32}; // msb

uint8_t nodeAppKey[16] = {0x30, 0x97, 0xD4, 0xAC, 0x4C, 0x8A, 0x8B, 0xE9, 0x6B, 0x69, 0x56, 0x29, 0x85, 0x3F, 0xFE, 0xBE}; //msb

#define LED_ON HIGH
#define LED_OFF LOW

// Foward declaration
/** LoRaWAN callback when join network finished */
static void lorawan_has_joined_handler(void);
/** LoRaWAN callback when join network failed */
static void lorawan_join_fail_handler(void);
/** LoRaWAN callback when data arrived */
static void lorawan_rx_handler(lmh_app_data_t *app_data);
/** LoRaWAN callback after class change request finished */
static void lorawan_confirm_class_handler(DeviceClass_t Class);
/** LoRaWAN callback after class change request finished */
static void lorawan_unconfirm_tx_finished(void);
/** LoRaWAN callback after class change request finished */
static void lorawan_confirm_tx_finished(bool result);
/** LoRaWAN Function to send a package */
static void send_lora_frame(void);
static uint32_t timers_init(void);

// APP_TIMER_DEF(lora_tx_timer_id);                                              ///< LoRa tranfer timer instance.
TimerEvent_t appTimer;                              ///< LoRa tranfer timer instance.
static uint8_t m_lora_app_data_buffer[LORAWAN_APP_DATA_BUFF_SIZE];        ///< Lora user application data buffer.
static lmh_app_data_t m_lora_app_data = {m_lora_app_data_buffer, 0, 0, 0, 0}; ///< Lora user application data structure.

/**@brief Structure containing LoRaWan parameters, needed for lmh_init()
 */
static lmh_param_t lora_param_init = {LORAWAN_ADR_OFF, DR_3, LORAWAN_PUBLIC_NETWORK, JOINREQ_NBTRIALS, LORAWAN_DEFAULT_TX_POWER};
// DR_8
// LORAWAN_DEFAULT_DATARATE
/**@brief Structure containing LoRaWan callback functions, needed for lmh_init()
*/
static lmh_callback_t lora_callbacks = {BoardGetBatteryLevel, BoardGetUniqueId, BoardGetRandomSeed,
                    lorawan_rx_handler, lorawan_has_joined_handler, lorawan_confirm_class_handler, lorawan_join_fail_handler,
                    lorawan_unconfirm_tx_finished, lorawan_confirm_tx_finished};

#define LED_BUILTIN 0

int32_t chipTemp = 0;

/**@brief Setup function
[... 260 lines cut ...]
  Serial.printf("lmh_send result %d\n", error);
  digitalWrite(LED_BUILTIN, LED_ON);

}

/**@brief Function for handling a LoRa tx timer timeout event.
 */
static void tx_lora_periodic_handler(void)
{
  TimerSetValue(&appTimer, LORAWAN_APP_TX_DUTYCYCLE);
  TimerStart(&appTimer);
  Serial.println("Sending frame");
  send_lora_frame();
}

/**@brief Function for the Timer initialization.
 *
 * @details Initializes the timer module. This creates and starts application timers.
 */
static uint32_t timers_init(void)
{
  appTimer.timerNum = 3;
  TimerInit(&appTimer, tx_lora_periodic_handler);
  return 0;
}

beegee-tokyo, 2021-12-03:
@Muuhoooo 
I am happy you could solve it. I will close this now.

closed 2021-12-03 by beegee-tokyo