beegee-tokyo/SX126x-Arduino #52: Radio.SetRxDutyCycle power usage not lower than Radio.Rx(0)

beele, 2021-12-05:
I have a [TTGO TBeamV1.1](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/LilyGo_TBeam_V1.1.pdf) which I have successfully programmed for low power usage. 0.82mA in deep sleep without power to or using the LoRa radio.

Once I power the LoRa radio via the AXP192 and set in into receive mode, power usage increases, as expected.
I tested the deep sleep of the ESP and it successfully woke up when receiving a message.
Next I tried using the SetRxDutyCycle method configured like the example code that can be found here, this also seems to work and the esp is woken up by the incoming message.

I have however noticed that the power usage between using Radio.Rx(0) and Radio.SetRxDutyCycle is exactly the same, 6.9mA.
I measured the current with a multimeter placed in between (series) the battery and the board.
According to the [datasheet](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/datasheets/SX1261_AN1200.36_SX1261-2_RxDutyCycle_V1.0.pdf) in this repo the power saving should be significant.

Am I missing something obvious? Could there be an issue or bug in the library?

`#include <Arduino.h>
#include <SPI.h>
#include <rom/rtc.h>
#include <driver/rtc_io.h>
#include <SX126x-Arduino.h>
#include <WiFi.h>
#include <esp_wifi.h>
#include "axp20x.h"
#include "esp_bt.h"
#include "driver/adc.h"

AXP20X_Class PMU;

int PIN_LORA_RESET = 23;
int PIN_LORA_DIO_1 = 33;
int PIN_LORA_BUSY = 32;
int PIN_LORA_NSS = 18;
int PIN_LORA_SCLK = 5;
int PIN_LORA_MISO = 19;
int PIN_LORA_MOSI = 27;

#define RF_FREQUENCY 868000000  // Hz
#define TX_OUTPUT_POWER 17      // dBm
#define LORA_BANDWIDTH 0        // [0: 125 kHz, 1: 250 kHz, 2: 500 kHz, 3: Reserved]
#define LORA_SPREADING_FACTOR 7 // [SF7..SF12]
#define LORA_CODINGRATE 1       // [1: 4/5, 2: 4/6,  3: 4/7,  4: 4/8]
#define LORA_PREAMBLE_LENGTH 8  // Same for Tx and Rx
#define LORA_SYMBOL_TIMEOUT 0   // Symbols
#define LORA_FIX_LENGTH_PAYLOAD_ON false
#define LORA_IQ_INVERSION_ON false
#define RX_TIMEOUT_VALUE 3000
#define TX_TIMEOUT_VALUE 3000

static RadioEvents_t RadioEvents;
hw_config hwConfig;
uint8_t cadRepeat;

void OnTxDone(void);
void OnRxDone(uint8_t *payload, uint16_t size, int16_t rssi, int8_t snr);
void OnTxTimeout(void);
void OnRxTimeout(void);
void OnRxError(void);
void OnCadDone(bool cadResult);

uint8_t deviceId[8];

const uint8_t i2c_sda = 21;
const uint8_t i2c_scl = 22;

void goToSleep(void) {
  Serial.println("Preparing to sleep!");

  Radio.Standby();
  SX126xSetDioIrqParams(IRQ_RX_DONE | IRQ_RX_TX_TIMEOUT,
                        IRQ_RX_DONE | IRQ_RX_TX_TIMEOUT,
                        IRQ_RADIO_NONE, IRQ_RADIO_NONE);
  // To get maximum power savings we use Radio.SetRxDutyCycle instead of Radio.Rx(0)
  // This function keeps the SX1261/2 chip most of the time in sleep and only wakes up short times
  // to catch incoming data packages
  Radio.SetRxDutyCycle(2 * 1024 * 1000 * 15.625, 10 * 1024 * 15.625);
  //Radio.Rx(0);
  Serial.println("Radio in low power receiver mode!");

  // Make sure the DIO1, RESET and NSS GPIOs are hold on required levels during deep sleep
  rtc_gpio_pullup_en((gpio_num_t)PIN_LORA_RESET);
  rtc_gpio_pullup_en((gpio_num_t)PIN_LORA_NSS);
  rtc_gpio_pulldown_en((gpio_num_t)PIN_LORA_DIO_1);

  esp_sleep_enable_ext1_wakeup(0x200000000, ESP_EXT1_WAKEUP_ANY_HIGH);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  Serial.println("IO configured for sleep!");

  Serial.println("Going to sleep!");
  Serial.flush();
  esp_deep_sleep_start();
}

void setup() {
  setCpuFrequencyMhz(20);
  Serial.begin(115200);

  RESET_REASON cpu0WakeupReason = rtc_get_reset_reason(0);
  RESET_REASON cpu1WakeupReason = rtc_get_reset_reason(1);

  // Define the HW configuration between MCU and SX126x
  hwConfig.CHIP_TYPE = SX1262_CHIP;         // SX1261_CHIP for Semtech SX1261 SX1262_CHIP for Semtech SX1262/1268
  hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // GPIO pin connected to NRESET of the SX126x
  hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;     // GPIO pin connected to NSS of the SX126x
  hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;   // GPIO pin connected to SCK of the SX126x
  hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;   // GPIO pin connected to MISO of the SX126x
  hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // GPIO pin connected to DIO 1 of the SX126x
  hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;   // GPIO pin connected to BUSY of the SX126x
  hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;   // GPIO pin connected to MOSI of the SX126x
  hwConfig.RADIO_TXEN = -1;                 // GPIO pin used to enable the TX antenna of the SX126x
  hwConfig.RADIO_RXEN = -1;                 // GPIO pin used to enable the RX antenna of the SX126x
  hwConfig.USE_DIO2_ANT_SWITCH = false;     // True if DIO2 is used to switch the antenna from RX to TX
  hwConfig.USE_DIO3_TCXO = true;           // True if DIO3 is used to control the voltage of the TXCO oscillator
  hwConfig.USE_DIO3_ANT_SWITCH = false;     // True if DIO3 is used to enable/disable the antenna
  hwConfig.USE_LDO = false;                  // False if SX126x DCDC converter is used, true if SX126x LDO is used
  hwConfig.USE_RXEN_ANT_PWR = false;        // If set to true RADIO_RXEN pin is used to control power of antenna switch

  RadioEvents.TxDone = OnTxDone;
  RadioEvents.RxDone = OnRxDone;
  RadioEvents.TxTimeout = OnTxTimeout;
  RadioEvents.RxTimeout = OnRxTimeout;
  RadioEvents.RxError = OnRxError;
  RadioEvents.CadDone = OnCadDone;

  if ((cpu0WakeupReason == DEEPSLEEP_RESET) || (cpu1WakeupReason == DEEPSLEEP_RESET)) {
    lora_hardware_re_init(hwConfig);
    Radio.ReInit(&RadioEvents);
    Radio.IrqProcessAfterDeepSleep();
  } else {
    //Red onboard LED
    pinMode(4, OUTPUT);
    digitalWrite(4, LOW);

    int ret = 0;
    Wire.begin(i2c_sda, i2c_scl);
    PMU.begin(Wire, AXP192_SLAVE_ADDRESS);

    do {
      // In order to ensure that it is set correctly,
      // the loop waits for it to return the correct return value
      Serial.println("Set PMU in sleep mode");
      ret = PMU.setSleep();
      delay(500);
    } while (ret != AXP_PASS);

    //PMU.setTSmode(AXP_TS_PIN_MODE_DISABLE);
    //PMU.enableChargeing(false);
    PMU.setChgLEDMode(AXP20X_LED_OFF);
    PMU.setPowerOutPut(AXP192_DCDC1, AXP202_OFF);
    PMU.setPowerOutPut(AXP192_DCDC2, AXP202_OFF);
    PMU.setPowerOutPut(AXP192_LDO2, AXP202_OFF);
    PMU.setPowerOutPut(AXP192_LDO3, AXP202_OFF);
    PMU.setPowerOutPut(AXP192_EXTEN, AXP202_OFF);

    PMU.setLDO2Voltage(3300);   //LoRa VDD
    PMU.setPowerOutPut(AXP192_LDO2, AXP202_ON);

    WiFi.disconnect(true);
    WiFi.mode(WIFI_OFF);
    esp_wifi_stop();
    btStop();
    esp_bt_controller_disable();
    adc_power_off();

    delay(3000);
    Serial.println("PMU init done!");

    lora_hardware_init(hwConfig);
    Radio.Init(&RadioEvents);
    Radio.SetChannel(RF_FREQUENCY);
    Radio.SetTxConfig(MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
                      LORA_SPREADING_FACTOR, LORA_CODINGRATE,
                      LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
                      false, 0, 0, LORA_IQ_INVERSION_ON, TX_TIMEOUT_VALUE);
    Radio.SetRxConfig(MODEM_LORA, LORA_BANDWIDTH, LORA_SPREADING_FACTOR,
                      LORA_CODINGRATE, 0, LORA_PREAMBLE_LENGTH,
                      LORA_SYMBOL_TIMEOUT, LORA_FIX_LENGTH_PAYLOAD_ON,
                      0, false, 0, 0, LORA_IQ_INVERSION_ON, true);
  }
  goToSleep();
}

void loop() {
}

void OnTxDone(void) {
  goToSleep();
}

void OnRxDone(uint8_t *payload, uint16_t size, int16_t rssi, int8_t snr) {
  char debugOutput[1024];
  for (int idx = 0; idx < size; idx++) {
    sprintf(&debugOutput[(idx * 2)], "%02X", payload[idx]);
  }
  Serial.print("Data: ");
  Serial.println(debugOutput);

  // Now its up to YOU to do something with the data

  goToSleep();
}

void OnTxTimeout(void) {
  goToSleep();
}

void OnRxTimeout(void) {
  goToSleep();
}

void OnRxError(void) {
  goToSleep();
}

void OnCadDone(bool cadResult) {
  cadRepeat++;
  Radio.Standby();
  if (cadResult) {
    if (cadRepeat < 6) {
      Radio.Standby();
      SX126xSetCadParams(LORA_CAD_08_SYMBOL, LORA_SPREADING_FACTOR + 13, 10, LORA_CAD_ONLY, 0);
      SX126xSetDioIrqParams(IRQ_CAD_DONE | IRQ_CAD_ACTIVITY_DETECTED,
                            IRQ_CAD_DONE | IRQ_CAD_ACTIVITY_DETECTED,
                            IRQ_RADIO_NONE, IRQ_RADIO_NONE);
      Radio.StartCad();
    } else {
      goToSleep();
    }
  }
}`

beegee-tokyo, 2021-12-06:
I thought I fixed this in the examples.

`Radio.SetRxDutyCycle(2 * 1024 * 1000 * 15.625, 10 * 1024 * 15.625);`
should be
`Radio.SetRxDutyCycle(2 * 1024 * 1000 * 15.625, 10 * 1024 * 1000 * 15.625);`

Can you retry with the correct sleep time?

beele, 2021-12-06:
Thx! Can't believe I missed that.

However it does not make any difference, tried multiple times, even flashing the sketch multiple times, but the current stays at 6.9mA same as with the regular receive mode :/

beegee-tokyo, 2021-12-06:
I do not have an EP32 at hand for testing, so I tried on my nRF52 modules.

Same like you, I cannot see any difference between Rx(0) or SetRxDutyCycle().

But it could be that I got the values wrong. Maybe reading the Semtech Application Note [RxDutyCycle](https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R0000001O3w/zsdHpRveb0_jlgJEedwalzsBaBnALfRq_MnJ25M_wtI) can help.

Sleep and Listen time need to calculated by several LoRa parameters.

beele, 2021-12-06:
Through some further [searching](https://blog.csdn.net/xqw19891201/article/details/120336076) and trial and error I now have something that works!

```cpp
Radio.SetRxDutyCycle((4 * 1.024 * 1000) / 15.625, (4 * 1.024 * 1000) / 15.625);
```

This lowers the power usage to about 4.88mA. If I lower the rxTime any lower than 4 preamble symbols, I do not receive any messages.
I will try lengthening the preamble to improve the rxTime vs sleepTime ratio which should further lower power usage.

beegee-tokyo, 2021-12-07:
Interesting. Please let me know the results.

beegee-tokyo, 2022-02-28:
No more feedback. Closed.

closed 2022-02-28 by beegee-tokyo