beegee-tokyo/SX126x-Arduino #53: LoRaWAN Change Region

Muuhoooo, 2021-12-09:
Hi Bernd, I hope you're fine.

I am working on ESP32 Sx1262 LoRaWAN on Arduino Plartform and first of all I has tried to connect EU868 and I have successfully sent the package. Then I changed the region inside the void loop and called the lmh_init and lmh_join functions again. The region has changed, I saw on the serial screen, but normally when I called the lmh_join function, it would enter the lorawan_has_joined_handler function, but in the code, it went into the tx_lora_periodic_handler function and "Did not join network, skip sending frame" 

is this normal?  What should we do if we want to join another region after joining a region?

beegee-tokyo, 2021-12-09:
The library does not support changing the region after it has joined. You have to set the region before starting lmh_join() in lmh_init().

The only option is to reset the device and do a fresh initialization.

IMHO it makes no sense to change the region during runtime in a real application (unless your ESP32 can teleport from continent to continent).

Muuhoooo, 2021-12-09:
I understand 😂, thank you very much.

Muuhoooo, 2021-12-09:
In fact, we fly the card with a balloon and transfer data to the gateway, so a region change was required. Thank you.

beegee-tokyo, 2021-12-09:
Ah, I heard about that balloon thing.

A possible workaround:
What you can try is to use the ESP32 Preferences class.
Save the new required region in the preferences, force reboot of the ESP32, read new region from the preferences , do initialization with new region.

Muuhoooo, 2021-12-09:
Thank you so much for the most logical solution.

beegee-tokyo, 2021-12-09:
I did something like this in my _**MHC-Sunnyboy**_ application: [prefs.cpp](https://github.com/beegee-tokyo/MHC-Sunnyboy-RAK13300/blob/main/src/prefs.cpp) 
I setup the complete LoRaWAN credentials and settings over BLE instead of having them hard-coded. Then I save them in the Preferences and read them back on boot.

beegee-tokyo, 2021-12-09:
@Muuhoooo 
Now you sparked my curiosity.

How do you handle the fact that LoRaWAN nodes are registered with a specific Region setting in the LoRaWAN server? You might need to change the DevEUI as well I guess.

Muuhoooo, 2021-12-09:
I tried to change it while in the same region(EU868) for now. First of all  I defined it as follows:
  LoRaMacRegion_t CurrentRegion = LORAMAC_REGION_US915;  
   Serial.println(CurrentRegion);
   And I saw CurrentRegion=0 and  OTAA joined failed,Check LPWAN credentials and if a gateway is in range,Restart network join request and if i enter the function i get this output i updated the value and I defined it as follows:
    LoRaMacRegion_t CurrentRegion = LORAMAC_REGION_EU868;  
   Serial.println(CurrentRegion);
   And I saw CurrentRegion=5 and I changed my region and transmit packet.

beegee-tokyo, 2021-12-09:
Never tried something like this. 
Just gave away my EU868 gateway unfortunately, now I have only AS923-3 available.

Muuhoooo, 2021-12-20:
Hi Bernd,  I have tried run time to change region even I started with EU868 and  I send 3 packet and I tried again EU868 and code started and send data no problem then I started as EU868 and send 3 packet and change region AS915 I saw did not join  and  when it came to the function I returned a value again than tried EU868 successfully joined at runtime. I change in "void loop" function. It was a surprise to me, but when the region was changed, it was successfully rejoined. I wanted to share with you.

beegee-tokyo, 2021-12-21:
@Muuhoooo 
👏 
Just to confirm I understand correct, you were able to change the Region "on the fly" by calling `lmh_init` again with the new region?

If that works, it is great.

Muuhoooo, 2021-12-21:
We've just tested it on the ground. There's time for us to take off. While I was in the European region, I started the Asian region with lmh_init, then I returned to the European frequency again, I called the lmh_init function, I joined again.

Muuhoooo, 2021-12-21:
but I'll let you know when we do our air fly tests.

beegee-tokyo, 2021-12-21:
I should get my EU868 gateway back in 1 or 2 weeks.
Then I can test if it joins both regions (EU868 and AS923-3) when I call lmh_init with each of the regions without restart.

Muuhoooo, 2021-12-21:
Hmm, I'm looking forward to the results

beegee-tokyo, 2022-02-28:
No more feedback. Closed.

closed 2022-02-28 by beegee-tokyo