beegee-tokyo/SX126x-Arduino #55: Stack size for the LORA task

robsonos, 2021-12-11:
Hi Bernd, I hope you are doing well.

I was wondering if there a reason I am not seeing for stack size of the LORA task to be 4096?:
https://github.com/beegee-tokyo/SX126x-Arduino/blob/910234742bd70b127c49c24ac4a8e0ab6b391e44/src/boards/mcu/board.cpp#L375-L379

From my tests it looks like 200 would be enough for the task to run:
[image]

Cheers,
Robson.

beegee-tokyo, 2021-12-11:
Hi @robsonos 
Yes, I guess it could be set smaller, I didn't really check the requirements. nRF52 and ESP32 have enough RAM.

robsonos, 2021-12-12:
Are you happy for me to push a PR with those changes?

beegee-tokyo, 2021-12-13:
This would start a complete re-test of the library for all MCU's and all supported LoRaWAN regions, max and min payloads, and in addition for LoRa P2P.
I cannot do this right now, I simply do not have the time.

beegee-tokyo, 2022-02-28:
Closed for now.

closed 2022-02-28 by beegee-tokyo