beegee-tokyo/SX126x-Arduino #58: SX1262 LoRa Node Module for Raspberry Pi Pico

stuartdd, 2022-01-11:
I was wondering if this library would work (assuming a new SX126x pin configuration) with the Waveshare Pico-LoRa-SX1262-868M.

Waveshare code examples are for the Pico C/C++ build process which I have no experience of. I have been using Arduino for all RP2040 coding as the learning curve is lower.

For details:
[https://www.waveshare.com/wiki/Pico-LoRa-SX1262-868M](https://www.waveshare.com/wiki/Pico-LoRa-SX1262-868M).

I wanted your view on the possibility before investing loads of time trying to get it to work. Your library looks like the best starting point for this. I have no requirement for LoRaWAN functionality.

I need basic communications. N nodes to 1 base. The base talks to a raspberry PI via USB. Sensors need to be able to take commands from the base so I can switch on lights etc, check battery status. I would use my own command set. The RP2040 and SX1262 has a battery so low power is critical.

Any advice would be really appreciated.

Regards

Stuart

beegee-tokyo, 2022-01-12:
Hi @stuartdd 
Should be no big problem, I tested the library with a Pico board before it was used with the WisBlock RAK11310 (RP2040+ SX1262 in one module).

You have to setup the **`hw_config`** structure as shown here: [Basic LoRa communication](https://github.com/beegee-tokyo/SX126x-Arduino#basic-lora-communication) to match with your connections. 

Then you can initialize the communication between the Pico and the SX1262 with **`lora_hardware_init(hwConfig);`**

stuartdd, 2022-01-12:
Ok. Thanks for the quick reply.

I will let you know the outcome.

Should I leave this open until I am done or close it now?

Regards

Stuart

beegee-tokyo, 2022-01-12:
Leave it open for now.

There will be some problems coming up. e.g. I do not know how the antenna switch is controlled on that specific LoRa board.

beegee-tokyo, 2022-02-28:
Seems to work. Closing this issue.

closed 2022-02-28 by beegee-tokyo