beegee-tokyo/SX126x-Arduino #59: Odd behavior of ADR on AS923

yourapiexpert, 2022-02-27:
I seem to have come across some odd behavior of ADR on AS923 not present on EU868 (these are the only two I have tested so far).

### ENVIRONMENT

AS923:
- RAK7268 with firmware 1.3.1_RAK b71 w/ gateway config for TTN Australia (AS923)
- WisBlocks API (1.1.11) w/ SX126x-Arduino (2.0.9)
- TTN Community Edition (EU1) w/ device config for TTN Australia (AS923). **DEVICE IS IN CLASS C**
- RAK4631 set to TX_POWER_5, MODE_ADR_ON, REGION 0, LMH_UNCONFIRMED_MSG

EU868:
- RAK2287 w/ 7244C enclosure
- BalenaOS + Basic Station Container
- WisBlocks API (1.1.11) w/ SX126x-Arduino (2.0.9)
- TTN Community Edition (EU1) w/ device config for TTN Australia (AS923). **DEVICE IS IN CLASS C**
- RAK4631 set to TX_POWER_5, MODE_ADR_ON, REGION 5, LMH_UNCONFIRMED_MSG

### OBSERVATIONS

With ADR enabled:

- Looking at the overview page of the RAK7268 I can see that only the two primary channels are being used.  Moreover, when looking at the packet logger I only seem to be receiving one out of every 2 packets in a fairly consistent manner.  Occasionally I will receive a subsequent transmission successfully. Thus at an interval of 5 minutes, every second transmission is almost consistently missed.  The experience and observation are mirrored when sending CLASS C acknowledgments from within the application where such packets are often missed.

With ADR disabled:

- Again based on the overview page of the RAK7268 it can be observed that a full spread of frequencies received via CFList is used and that there are no missing uplink or downlink packets observed.

[image]

On EU868:

Unfortunately, there is no status page, and observations were done purely on the TTN console. Based on observations in either ADR or no ADR modes both uplink and downlink transmissions presented no problems.

Please do let me know what else I can do or what other information I can provide to assist with debug and testing.

-- pogipo

beegee-tokyo, 2022-02-28:
Hi,
I am never using ADR, I will have to check if there is anything going wrong in the library. The LoRaMAC stack was taken without changes from the Semtech source code, it will take some time to dig through that mess.
I am using AS923-3, which is a variant of the AS923, but the code in the stack is the same for both.
My bigger problem is TTN, as they have no native support for AS923-3, I have to setup the channels manually for every device.

beegee-tokyo, 2022-02-28:
I run a test:
- I use AS923-3 instead of AS923, but the LoRaMAc stack for both is the same, only difference is a frequency shift of 6.6MHz between the two
- Used RAK7258 (I have only that gateway available) setup for AS923-3, using default channel settings
- Chirpstack LoRaWAN server

I have ADR enabled and DR is 5
[image]

I cannot see any problems and the packets arrive every 1 minute in Chirpstack, 
[image]

Then I tested the device against TTN, au1.cloud.thethings.network, and I see the same problem that you have, sporadic a message comes through, but most are lost. Same as you, I can see the messages on the gateway.

Not sure what I can do here, it works fine on LNS like Chirpstack and fails on TTN.

yourapiexpert, 2022-02-28:
Hi! Thank you so much for the effort and the trouble.

I think you have identified just how messed up TTN is.  I installed a brand new version (3.18) of TTI Stack and AS923 could not be more wrong.  This is the channel map that that RAK268 received against the CFList of the device:

[image]

Where the gateway thinks it should be operating...nobody knows. Contrast this to public TTN:

[image]

At least the CFList matches the capable frequencies of the gateway.  Just some feedback...I will install a Chirpstack here soon and I already have plans to check on AWS IoT since that is the final destination at the end of the year.

beegee-tokyo, 2022-03-01:
@yourapiexpert 
No trouble at all. These problems with TTN (specially on everything that is not US915 or EU868) was one of the reasons to use a local Chirpstack LNS or even the built-in LNS of the RAK gateways.

Good luck with your projects.

closed 2022-03-01 by beegee-tokyo

yourapiexpert, 2022-03-01:
My sincerest apologies for re-opening this but I think I might have discovered something that might help shed light, or, at least, demonstrates that the problem goes away after days of this.

Even with Chirpstack, I was able to reproduce the missing frames under ADR conditions naturally questioning, re-questioning, and consistently checking the config as it was a brand new install.

However, the penny did not drop until I saw in the logs of the network server what the RAK4631 ('the device') was rejecting NewChannelReq of seemingly valid frequencies.  I almost remember seeing the channel mask for ADR include those rejected frequencies but I am not sure and it seems a big oversight if that's the case so let's just say it's me and my memory.

Everything cleared up, instantly, when I removed this piece of code from src/mac/region/RegionAS923.cpp (line 1065):

```
	        /*
		// We are not allowed to change the frequency
		if (channelAdd->NewChannel->Frequency != Channels[id].Frequency)
		{
			freqInvalid = true;
		}
		*/
```

Suddenly on TTN the problem was no longer prevalent, on Chirpstack no longer prevalent and for the first time I am witnessing a full spread of frequencies sent via CFList + NewChannelReq:

[image]

This helped somewhat with my TTN saga but not fully so I put some extra debug into SX126X and you would not believe what happened:

[image]

For the first few packets everything seems ordinary and you can see that the frequencies line up correctly with what was in the CFList and the only other frequency to come down via NewChannelReq was 922000000.  When ADR kicks in, it all goes pear- shaped and for some or other reason the device starts sending on a frequency not only not advertised via CFList or NewChannelReq but also a frequency not even within the bounds of AS923 which caps off at 923400000 per the TTN global conf file.

_Interestingly if I shift 924600000 down by 1.8MHz then I do come up with a frequency that was advertised in CFList._

[image]

Now, to test this theory I did the ugliest and most inelegant hack in radio/sx126x.cpp at around line 440:

```
        // sorry Mr. Blocks for the ugly...
	if (frequency > 923400000)
	{
		frequency -= 1800000;
		LOG_LIB("SX126X", "Frequency adjusted to %luHz", frequency);
	}
```

As you can see, even with ADR enabled I have been able to get an amazing and consistent run with my 30-second intervals on TTN:

[image]

I don't know who, what or why these frequencies are shifting.  I would think if it's something TTN is sending then there would be mass outrage from Australia but I just have not so far been able to figure out where this is coming from.  I hope I provided enough information for you to be able to replicate but if not, you know where to find me :)

beegee-tokyo, 2022-03-02:
That looks strange, I will have to dig in a little bit deeper.
Strange enough, the 1.8MHz shift points to AS923-2 which is -1.8MHz away from the AS923-1 (or default AS923).

The allowed channels 1 to 8 are defined in RegionAS923.h and range from 923.2 to 924.6 MHz, that part comes from the Semtech SX1262 library, I didn't touch that. And these are the channels that the check fails.
[image]

There is one thing I am not 100% sure about. The LoRa Alliance Regional Parameters document states a channel data structure for 16 channels, but the Semtech library channel table has only 8 channels.
[image]

yourapiexpert, 2022-03-02:
Agreed.  I went down a rabbit hole on that one too but I have some good news.  The entire problem can be cleared up by commenting out the frequency check as well as the following in src/mac/region/RegionAS923.cpp around line 360:

```
oid RegionAS923InitDefaults(InitType_t type)
{
	switch (type)
	{
	case INIT_TYPE_INIT:
	{
		// Channels
		Channels[0] = (ChannelParams_t)AS923_LC1;
		Channels[1] = (ChannelParams_t)AS923_LC2;
		/*
		Channels[2] = (ChannelParams_t)AS923_LC3;
		Channels[3] = (ChannelParams_t)AS923_LC4;
		Channels[4] = (ChannelParams_t)AS923_LC5;
		Channels[5] = (ChannelParams_t)AS923_LC6;
		Channels[6] = (ChannelParams_t)AS923_LC7;
		Channels[7] = (ChannelParams_t)AS923_LC8;
		*/
```

This appears to correctly set channels during the LinkAdrReq.  From an earlier debug here is TTN with the default 7 channels enabled:

```<ADR> Requested channel 0(923200000Hz)
<ADR> Requested channel 1(923400000Hz)
<ADR> Requested channel 2(923600000Hz)
<ADR> Requested channel 3(923800000Hz)
<ADR> Requested channel 4(924000000Hz)
<ADR> Requested channel 5(924200000Hz)
<ADR> Requested channel 6(924400000Hz)
<ADR> Requested channel 7(924600000Hz)
<ADR> Requested channel 8(922200000Hz)
<ADR> Requested channel 9(922400000Hz)
<ADR> Requested channel 10(922600000Hz)
<ADR> Requested channel 11(922800000Hz)
<ADR> Requested channel 12(923000000Hz)
<ADR> Requested channel 13(0Hz)
<ADR> Requested channel 14(0Hz)
<ADR> Requested channel 15(0Hz)
```

Now TTN did not send that many but they got mugged in that list somehow between CFList + NewChanReq and then LinkADRReq.

When I disabled the channels and left only LC1 + LC2:

```
<ADR> Requested channel 0(923200000Hz)
<ADR> Requested channel 1(923400000Hz)
<ADR> Requested channel 2(922200000Hz)
<ADR> Requested channel 3(922400000Hz)
<ADR> Requested channel 4(922600000Hz)
<ADR> Requested channel 5(922800000Hz)
<ADR> Requested channel 6(923000000Hz)
<ADR> Requested channel 7(922000000Hz)
<ADR> Requested channel 8(0Hz)
<ADR> Requested channel 9(0Hz)
<ADR> Requested channel 10(0Hz)
<ADR> Requested channel 11(0Hz)
<ADR> Requested channel 12(0Hz)
<ADR> Requested channel 13(0Hz)
<ADR> Requested channel 14(0Hz)
<ADR> Requested channel 15(0Hz)
```

Not only did the list remain correct, but also:

[image]

an absolutely flawless TTN experience.

I don't know what the effort involved is but I see the folks at loramac-node also based theirs on  the reference implementation but have a same yet different AS923 implementation doing something different with the 4 regions.

beegee-tokyo, 2022-03-02:
@yourapiexpert 
Wow, great finding. I will do some tests.
As I said, the Mac part of the library is more or less taken from Semtech, but it is as well already 2 years old. Could be in there since ever.

Thank you very much for looking into it.

beegee-tokyo, 2022-03-02:
TTN is stupid.
I setup my device for AS923-3, which is not supported by TTN, you have to set the channel list manually:
[image]

And the CF list I get is still for AS923-1:
[image]

My Chirpstack is giving me the correct CF list.
[image]

Cannot test more against TTN, but I can say for Chirpstack your changes are working.