beegee-tokyo/SX126x-Arduino #62: AU915 Region Datarate to SF!!

Muuhoooo, 2022-04-04:
Hi Bernd, I hope you're fine.

While looking the RP002-1.0.3 LoRaWAN® Regional Parameters document, I saw that the DR to SF defined for the AU915 region is not the same as the values defined in your document. It is the same for other regions, only the AU915 is different. Could you have misspelled it while defining it? Or is there a different situation? Thanks. Best regards.

For RP002-1.0.3 - DR0-SF12/125kHz , Yours Documents -DR0-SF10/125 kHz
For RP002-1.0.3 - DR1-SF11/125kHz , Yours Documents -DR1-SF9/125 kHz
For RP002-1.0.3 - DR2-SF10/125kHz , Yours Documents -DR2-SF8/125 kHz
For RP002-1.0.3 - DR3-SF9/125kHz , Yours Documents -DR3-SF7/125 kHz
For RP002-1.0.3 - DR4-SF8/125kHz , Yours Documents -DR4-SF8/500 kHz 
For RP002-1.0.3 - DR5-SF7/125kHz , Yours Documents -DR5-RFU

https://lora-alliance.org/wp-content/uploads/2021/05/RP002-1.0.3-FINAL-1.pdf (page 48)
https://github.com/beegee-tokyo/SX126x-Arduino/blob/bg_handling/DATARATE.MD(AU915)

beegee-tokyo, 2022-04-04:
The library is based on LoRaMAC stack V1.0.2, there might have been changes or I just copied the wrong values.
I need to look into the code.

beegee-tokyo, 2022-04-04:
Copy and paste error, will correct it.

closed 2022-04-04 by beegee-tokyo