beegee-tokyo/SX126x-Arduino #63: demo code always uses a random devEui/appEui

BNNorman, 2022-04-16:
I have setup config.h using keys I have for testing and built the code using VSCODE on a Raspberry Pi4

When I build the example temperature+LED  otaa and copy the code to the pico it ignores the config.h values and generates random devEui and appEui as seen in my gateway console. The devEui and appEui are always the same but the value changes each time I powercycle the pico.

I have spent many fruitless hours trying to find a 'magic switch' to no avail. Perhaps you could tell me what I'm missing because I followed the build process in the readme document.

Many thanks

beegee-tokyo, 2022-04-17:
I am a little bit confused.
What example code are you referring to? What Pico board are you using?

BNNorman, 2022-04-17:
My bad, it looks like a link I followed from Waveshare brought me here. I should have double checked

closed 2022-04-17 by BNNorman