beegee-tokyo/SX126x-Arduino #64: Chirpstack ABP configuration

andrelmbraga, 2022-05-13:
Hello,

For us915 i would like to know these informations. I didnt find it on the lib.

Rx1 Delay, Rx1 DR offset, RX2 DR, RX2 Frequency, Factory-preset Frequecies

ABP is working but not OTA, so in ABP I need to have these values to use downlinks on chirpstack.

Also I would like to know where can I change the uplink to be confirmed. 

Thanks in advance!

beegee-tokyo, 2022-05-14:
Rx1 Delay, Rx1 DR offset, RX2 DR, RX2 Frequency, Factory-preset Frequencies are defined by the LoRaWAN regional parameters and the library has no function to change them.
I am testing with ABP and there is no need to change them manually.

If you want to mess with them, you have to look into the [RegionAU915.h](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/src/mac/region/RegionAU915.h)

closed 2022-05-14 by beegee-tokyo

andrelmbraga, 2022-05-14:
@beegee-tokyo thanks for your answer. In fact I dont wanna change the configuration on the RegionAU915.h file.  I wanna get the parameters from there.

For us915:

Rx1 Delay :  1000 or 1 ? ( chirpstack say that expected values are between 0 and 15)

/*!
 * Receive delay 1
 */
#define US915_RECEIVE_DELAY1 1000

Rx1 DR offset : 0

/*!
 * Default Rx1 receive datarate offset
 */
#define US915_DEFAULT_RX1_DR_OFFSET 0

RX2 DR : 8

/*!
 * Second reception window channel datarate definition.
 */
#define US915_RX_WND_2_DR DR_8

RX2 Frequency: 923300000

/*!
 * Second reception window channel frequency definition.
 */
#define US915_RX_WND_2_FREQ 923300000

Factory-preset Frequecies: i really dont know what they mean. 
??

If you can help I ll really appreciate, thanks for your time!

beegee-tokyo, 2022-05-15:
For the delays, Chirpstack talks in seconds, the library talks in milliseconds, so 1000 == 1
For the frequencies, US915 has [72 channels](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/CHANNELS.MD#usa-915-mhz) pre-defined. But as most gateways are only capable of 8 channels, they are put into 8 groups with 8 TX and 1 RX channel:
[image]
 
You have to set Chirpstack, the gateway and your node to the same group.
In the library you select the group with **`lmh_setSubBandChannels(subband)`** with subband counting from 1 to 9.
On the gateway, it depends on the brand. On a RAK Edge gateway (picture above) you can select the group, Dragino gateways are the same. And in Chirpstack you have to select US915 in the **`chirpstack-network-server.toml`**. Usually there is a prepared example file for Chirpstack. I found it in /etc/chirpstack-network-server and it defined the region like this:
```log
# Network-server settings.
[network_server]
# Network identifier (NetID, 3 bytes) encoded as HEX (e.g. 010203)
net_id="000000"

  # LoRaWAN regional band configuration.
  #
  # Note that you might want to consult the LoRaWAN Regional Parameters
  # specification for valid values that apply to your region.
  # See: https://www.lora-alliance.org/lorawan-for-developers
  [network_server.band]
  name="US915"
```

andrelmbraga, 2022-05-16:
Thanks for your answer. It works really well. With uplink confirmed and also donwlink confirmed.
I'm now in a task in order to indentify how can I now that my uplink was confirmed, like returning something after lmh_send, just the error status is not enough. 
Another point is be able to know if the the process of a class A for example was completed, like sending process and windows.

I saw one function called lmh_send_blocking()  however it seems to just return after the timeout looks like the lmh_mac_is_busy never stops to be busy. 

Did you guys used this function already or is in test? Thanks to much for the support. I thanks for this nice library.

beegee-tokyo, 2022-05-16:
Instead of using the blocking send function, why do you not use the normal send function and register the callbacks.
There are several of them:
```cpp
// LoRaWAN event handlers
/** LoRaWAN callback when join network finished */
static void lpwan_joined_handler(void);
/** LoRaWAN callback when join network failed */
static void lpwan_join_fail_handler(void);
/** LoRaWAN callback when data arrived */
static void lpwan_rx_handler(lmh_app_data_t *app_data);
/** LoRaWAN callback after class change request finished */
static void lpwan_class_confirm_handler(DeviceClass_t Class);
/** LoRaWAN callback after class change request finished */
static void lpwan_unconfirm_tx_finished(void);
/** LoRaWAN callback after class change request finished */
static void lpwan_confirm_tx_finished(bool result);
```
```cpp
/** Structure containing LoRaWan callback functions, needed for lmh_init() */
static lmh_callback_t lora_callbacks = {get_lora_batt, BoardGetUniqueId, BoardGetRandomSeed, lpwan_rx_handler,
										lpwan_joined_handler, lpwan_class_confirm_handler, lpwan_join_fail_handler,
										lpwan_unconfirm_tx_finished, lpwan_confirm_tx_finished};
```
If you use confirmed packages, `lpwan_confirm_tx_finished(bool result);` tells you with the parameter result if the packet was acknowledged by the server.

You can find a code sample how to use the callbacks here: https://github.com/beegee-tokyo/WisBlock-API/blob/main/src/lorawan.cpp

andrelmbraga, 2022-05-16:
Really nice, I ll try on my side in a few hours. Thank you very much.

andrelmbraga, 2022-05-23:
Its ok, in fact I mad a real nice process in order to work. Basically I`m a class A device when a send command uplink to get timestamp from the server my device goes to C and my network server active the device again o c profile and send a downlink. The error is less than 2 seconds. For my application works like a charm.

Now I get another point, is it possible to restart the lora chip?  If yes can you suggest the function? Sometimes after a while my device just keep getting error on the sending process.

beegee-tokyo, 2022-05-24:
There are two options,
a) restart only LoRaMAC state machine with `lmh_init()` calling a second time. If using ABP, it should work just fine. If using OTAA it will start a join cycle. 
b) reset the SX1262 by going through the complete initialization again
   - lora_hardware_init()
   - lmh_init()

beegee-tokyo, 2022-05-24:
But the restart should not be required. I have nodes running since months without reset without problems. 

On US915, I would try to disable ADR and disable Dutycycle with LORAWAN_ADR_OFF and LORAWAN_DUTYCYCLE_OFF
```cpp
static lmh_param_t lora_param_init = {LORAWAN_ADR_OFF, LORAWAN_DEFAULT_DATARATE, LORAWAN_PUBLIC_NETWORK, JOINREQ_NBTRIALS, LORAWAN_DEFAULT_TX_POWER, LORAWAN_DUTYCYCLE_OFF};
```

Another thing, how frequently are you sending packets? In my experience, everything faster than 20 seconds can lead to congestion somewhere in the LoRaMAC stack (not my code, that is from Semtech). I usually use 30 seconds as lowest value between two packets.