beegee-tokyo/SX126x-Arduino #65: Arduino Mbed RP2040 interrupts

bogdanpad, 2022-05-16:
Hi ,

I am trying to run the LoRaWAN (OTA) example with an Ebyte E22900M22 connected to a  Raspberry Pi Pico (Arduino Mbed OS RP2040 V3.0.1  and SX126x-Arduino  V2.0.1) . It is sending a join request and the infrastructure replays with a join accept but nothing else happens after this. Either the interrupts are not working, either the code gets stuck  somehow. 

Are the radio interrupts based on DIO1 ? ( The function called by DIO1 seems to be  "RadioOnDioIrq" which is empty ?). 

With the debug activated, the serial output is the following.

20:06:16.585 -> =====================================
20:06:16.585 -> SX126x LoRaWan test
20:06:16.585 -> =====================================
20:06:16.585 -> <TIM> Timer 0 assigned
20:06:16.585 -> <BRD> SyncWord = 2414
20:06:16.585 -> <LMH> OTAA 
20:06:16.585 -> DevEui=00-95-64-1F-DA-91-19-0B
20:06:16.585 -> DevAdd=260116F8
20:06:16.585 -> AppEui=70-B3-D5-7E-D0-02-01-E1
20:06:16.585 -> AppKey=07-C0-82-0C-30-B9-08-70-0C-0F-70-06-00-B0-BE-09
20:06:16.585 -> <TIM> Timer 0 assigned
20:06:16.585 -> <TIM> Timer 1 assigned
20:06:16.585 -> <TIM> Timer 2 assigned
20:06:16.585 -> <TIM> Timer 3 assigned
20:06:16.585 -> <TIM> Timer 4 assigned
20:06:16.632 -> <TIM> Timer 5 assigned
20:06:16.632 -> <TIM> Timer 6 assigned
20:06:16.680 -> <RADIO> RX window timeout = 0
20:06:16.680 -> <LMH> Selected subband 1
20:06:16.680 -> <LMH> Selected subband 1
20:06:16.680 -> <EU868> Channel count ch# 0, freq 868100000
20:06:16.680 -> <EU868> Set channel 0, frequency 868100000
20:06:16.680 -> <EU868> Channel count ch# 1, freq 868300000
20:06:16.680 -> <EU868> Set channel 1, frequency 868300000
20:06:16.680 -> <EU868> Channel count ch# 2, freq 868500000
20:06:16.680 -> <EU868> Set channel 2, frequency 868500000
20:06:16.680 -> <EU868> Channel count ch# 3, freq 0
20:06:16.680 -> <EU868> Channel count ch# 4, freq 0
20:06:16.680 -> <EU868> Channel count ch# 5, freq 0
20:06:16.680 -> <EU868> Channel count ch# 6, freq 0
20:06:16.680 -> <EU868> Channel count ch# 7, freq 0
20:06:16.680 -> <EU868> Using channel 1, frequency 868300000
20:06:16.680 -> <LM> dutyCycleTimeOff was = 0
20:06:16.680 -> <LM> Set send_join_now to false

I've tested the same hardware configuration with Radiolib (Peer to Peer) and all works fine. Also I've tested the Ebyte module  connected to a nRF52832 with your library and all works well, including LoRaWAN multicast with a minor modification.

I ordered some RAK11300 for my project, hopefully it will work but I would still like to understand why my test configuration (which is similar) doesn't work ?

Thank you for sharing this great library.

beegee-tokyo, 2022-05-16:
Hi @bogdanpad 

I have the library running with eByte module and a RP Pico board.
The DIO1 interrupts are calling [RadioOnDioIrq](https://github.com/beegee-tokyo/SX126x-Arduino/blob/dc15fab55f3975171959c2255153d14427d989d4/src/radio/sx126x/radio.cpp#L1248)

I think the problem is within the example code. The Arduino BSP for the RP2040 got an update recently and some of the examples failed after that. Something with the timers I used broke.

Try to replace the `TimerEvent_t appTimer;` from the example with `mbed::Ticker appTimer;` that worked in the RAK11300 example [LoRaWAN_OTAA_ABP.ino](https://github.com/RAKWireless/WisBlock/blob/master/examples/RAK11300/communications/LoRa/LoRaWAN/LoRaWAN_OTAA_ABP/LoRaWAN_OTAA_ABP.ino)

bogdanpad, 2022-05-16:
Hi @beegee-tokyo 

Sorry for saying about RadioOnDioIrq that is empty. The way my editor shown the code was rather confusing :)

Replacement of the timers haven't solved the problem.

I was able to carry on with the debugging and I found that the start_lora_task() is not called anywhere, therefore the interrupts were not handled by anything. I added start_lora_task(); right after the lora_hardware_init and it started to work.

Attaching the timer interrupt in the timer interrupt callback,  causes the Pico to block. I simply moved the appTimer.attach in the loop, (after the send_lora_frame();)  and all works good now. I will test it for a while to see if anything else occurs.

By the way, it works with  Arduino Mbed OS RP2040 V3.1.1 , updated today.

Thank you very much!

Regards,

Bogdan

beegee-tokyo, 2022-05-17:
Good to hear that you got it to work. for `start_lora_task()` I need to check, it should be started if **`ARDUINO_ARCH_RP2040`** is defined. But I tested it only with the RAKwireless BSP lately.

Will close this, but let me know if you run into other problems.

closed 2022-05-17 by beegee-tokyo