beegee-tokyo/SX126x-Arduino #67: RAK4630 join failure callback not firing, timers seem to stop?

bsodmike, 2022-06-15:
Hello!

I've setup the basic [OTA demo from the WisBlock repo](https://github.com/RAKWireless/WisBlock/blob/master/examples/RAK4630/communications/LoRa/LoRaWAN/LoRaWAN_OTAA_ABP/LoRaWAN_OTAA_ABP.ino) and started to monkey-patch is dependency directly to trace exactly where the join was failing.

I've also enabled debugging to try and get a better idea, but I have a feeling that some sort of timing is being thrown off here.

```
=====================================
Welcome to RAK4630 LoRaWan!!!
Type: OTAA
Region: EU868
=====================================
Setting up the EUIs and Keys...
Key setup done.
<LMH> OTAA
           DevEui=xxx
                                         DevAdd=00000000
                                                        AppEui=xxx
                                                                                      AppKey=xxx
Inside: TimerInit (for NRF52)
Inside: TimerInit (for NRF52)
Inside: TimerInit (for NRF52)
Inside: TimerInit (for NRF52)
Inside: TimerInit (for NRF52)
Inside: TimerInit (for NRF52)
Inside: TimerInit (for NRF52)
<LMH> Selected subband 1
Inside: TimerInit (for NRF52)
Starting Join procedure...
Trying: MLME_JOIN
Running: ResetMacParameters
Trying: MLME_JOIN -> Send
<EU868> Channel count ch# 0, freq 868100000
<EU868> Set channel 0, frequency 868100000
<EU868> Channel count ch# 1, freq 868300000
<EU868> Set channel 1, frequency 868300000
<EU868> Channel count ch# 2, freq 868500000
<EU868> Set channel 2, frequency 868500000
<EU868> Channel count ch# 3, freq 0
<EU868> Channel count ch# 4, freq 0
<EU868> Channel count ch# 5, freq 0
<EU868> Channel count ch# 6, freq 0
<EU868> Channel count ch# 7, freq 0
<EU868> Using channel 1, frequency 868300000
<LM> dutyCycleTimeOff was = 0
<LM> Set send_join_now to false
<LM> JoinRequestTrials: 1

MLME_JOIN -> Send: LORAMAC_STATUS_OK
```

So tracing through:

- `lmh_join()` calls `LoRaMacMlmeRequest()`
- This calls `Send` and that's where my additional debug ends.
- It looks like `status = ScheduleTx();` never runs as the timing stops.
- `SendFrameOnChannel` is called as `dutyCycleTimeOff = 0` (https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/src/mac/LoRaMac.cpp#L2019-L2043)
- There is a a "weak" fix for EU868 join, could this be an issue? (https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/src/mac/LoRaMac.cpp#L2019-L2043)

What I've noticed:

- Only once, the code executed differently and ran much slower, the actual join fail handle was called.  Since then I've never been able to get it to run like this.
- Running the code _now_ runs super quick, till it hits the main `loop()`.

Let me know if you'd like me to share my test repo, happy to help anytime.

Thanks, Mike.

beegee-tokyo, 2022-07-11:
Hi Mike,
Not sure why the join fail callback is not fired. I didn't see that problem so far.
Enabling the library debug is messing up the timing, I recommend to use this as little as possible.

Is this based on the **`LoRaWAN_OTAA_ABP`** from the [WisBlock Repo](https://github.com/RAKWireless/WisBlock) ?

bsodmike, 2022-07-11:
Hi Bernd, this is Mike (from RAK). I’ve got this to work with your help on a separate device on AS923MHz.

When I have time to investigate the first module I was testing with I’ll open a new ticket if I have any new feedback. Thanks

closed 2022-07-11 by bsodmike