beegee-tokyo/SX126x-Arduino #69: Ping_pong dont call timeout

promistrio, 2022-06-29:
I use ESP32 with E22-900M30S and nothing happens on Ping_pong example, but this library (https://github.com/StuartsProjects/SX12XX-LoRa) can transmit data ( but can't receive. )

I enable debug messages:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
=====================================
SX126x PingPong test
=====================================
MCU Espressif ESP32
BoardId: 00-00-94-8F-D3-D1-08-F0
<BRD> SyncWord = 2414
<BRD> LoRa Task started
Starting Radio.Rx
<RADIO> RX window timeout = 3000

```

That's all. I ran it on 2 devices
```

#ifdef ESP32
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 27;  // LORA RESET
int PIN_LORA_DIO_1 = 4; // LORA DIO_1
int PIN_LORA_BUSY = 25;  // LORA SPI BUSY

int PIN_LORA_NSS = 5;	// LORA SPI CS
int PIN_LORA_SCLK = 18;  // LORA SPI CLK
int PIN_LORA_MISO = 19;  // LORA SPI MISO
int PIN_LORA_MOSI = 23;  // LORA SPI MOSI

int RADIO_TXEN = 33;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = 32;	 // LORA ANTENNA RX ENABLE

	// Define the HW configuration between MCU and SX126x
	hwConfig.CHIP_TYPE = SX1262_CHIP;		  // Example uses an eByte E22 module with an SX1262
	hwConfig.PIN_LORA_RESET = PIN_LORA_RESET; // LORA RESET
	hwConfig.PIN_LORA_NSS = PIN_LORA_NSS;	 // LORA SPI CS
	hwConfig.PIN_LORA_SCLK = PIN_LORA_SCLK;   // LORA SPI CLK
	hwConfig.PIN_LORA_MISO = PIN_LORA_MISO;   // LORA SPI MISO
	hwConfig.PIN_LORA_DIO_1 = PIN_LORA_DIO_1; // LORA DIO_1
	hwConfig.PIN_LORA_BUSY = PIN_LORA_BUSY;   // LORA SPI BUSY
	hwConfig.PIN_LORA_MOSI = PIN_LORA_MOSI;   // LORA SPI MOSI
	hwConfig.RADIO_TXEN = RADIO_TXEN;		  // LORA ANTENNA TX ENABLE
	hwConfig.RADIO_RXEN = RADIO_RXEN;		  // LORA ANTENNA RX ENABLE
	hwConfig.USE_DIO2_ANT_SWITCH = false;	  // Example uses an CircuitRocks Alora RFM1262 which uses DIO2 pins as antenna control
	hwConfig.USE_DIO3_TCXO = false;			  // Example uses an CircuitRocks Alora RFM1262 which uses DIO3 to control oscillator voltage
	hwConfig.USE_DIO3_ANT_SWITCH = false;	 // Only Insight ISP4520 module uses DIO3 as antenna control

```

beegee-tokyo, 2022-06-29:
Hi @promistrio 

From the initial log it looks like the SPI configuration is correct, the SX1262 is found and initialized, but then it seems the RX timeout IRQ is not triggered. I don't know the E22-900M30S in detail

Could be
DIO1 is not assigned correct
TXEN/RXEN are not assigned correct
DIO2 might be used for antenna switch instead of separate TXEN/RXEN pins

promistrio, 2022-06-29:
> This is a design with the E22-900M22S module that works fine with ESP32, nRF52. It is a feather compatible board. [image]

We can connect RX_EN TX_EN to MCU. Is this library support it?

promistrio, 2022-06-30:
> Yes, and I checked, the eByte modules seems to use DIO3 to control the TCXO Your HW config should look like this:
> 
> ```c++
> hwConfig.RADIO_TXEN = hwConfig.RADIO_TXEN;              // LORA ANTENNA TX ENABLE (e.g. eByte E22 module)
> hwConfig.RADIO_RXEN = hwConfig.RADIO_RXEN              // LORA ANTENNA RX ENABLE (e.g. eByte E22 module)
> hwConfig.USE_DIO2_ANT_SWITCH = false;                         // LORA DIO2 does not control antenna
> hwConfig.USE_DIO3_TCXO = hwConfig.USE_DIO3_TCXO;  // LORA DIO3 controls oscillator voltage (e.g. eByte E22 module)
> hwConfig.USE_DIO3_ANT_SWITCH = false;                         // LORA DIO3 does not control antenna
> ```
> 
> RXEN and TXEN will be controlled by the library, you do not need to control them from our application.

Thank you! I got rx timeout message, but cad not start.

```

MCU Espressif ESP32
BoardId: 00-00-94-8F-D3-D1-08-F0
<BRD> SyncWord = 2414
<BRD> LoRa Task started
<RADIO> RX window timeout = 3000
<RADIO> RadioIrqProcess => IRQ_RX_TX_TIMEOUT
OnRxTimeout

```

promistrio, 2022-07-01:
It's so strange. I'll try to conduct more experiments. For example, on sx12xx lib by Stuart, I got successful interrupt on tx done, but DIO1 not up on rx

promistrio, 2022-07-09:
@beegee-tokyo I replaced the loop function in the ping pong code example as follows:
[image]
And after launching, I did not see the sent packets using SDR. But with the help of the Stuart library, sending works and messages are visible on the radio (but not receive)

beegee-tokyo, 2022-07-10:
Digged out my old PCB with the eByte E22-900M22S and my old Adafruit Huzzah ESP32.  As I have only one left, the PingPong partner is build with the RAKwireless WisBlock modules:
[image]
The library works fine with these two combinations:
[image]

I have really no idea why it doesn't work with the eByte E22-900M30. There is one difference, that I don't know if it needs control. The eByte E22-900M30 must have an RF booster. The eByte documentation says it is an Semtech SX1262, but the SX1262 has a max TX power of 22dBm. But  eByte claims their module has an TX power of 30dBm. That is not possible without an RF booster. Maybe eByte can share some information about it and if there is something else needed to control the RF booster.

closed 2022-10-20 by beegee-tokyo