beegee-tokyo/SX126x-Arduino #72: ADR enable not working on AS923

asupawat, 2022-09-18:
Cannot change tx-power and data-rate from gateway requested, need to remove comment below
		
// Channels[2] = (ChannelParams_t)AS923_LC3;
// Channels[3] = (ChannelParams_t)AS923_LC4;
// Channels[4] = (ChannelParams_t)AS923_LC5;
// Channels[5] = (ChannelParams_t)AS923_LC6;
// Channels[6] = (ChannelParams_t)AS923_LC7;
// Channels[7] = (ChannelParams_t)AS923_LC8;

https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/src/mac/region/RegionAS923.cpp#L369

beegee-tokyo, 2022-09-19:
This is a very bad idea. AS923 requires that during join only channel 0 and 1 are used. The LoRaWAN server has to enable the other 6 channels and send the additional channels to the node. This is defined the LoRaWAN Regional Parameters by the LoRa alliance ==> https://lora-alliance.org/resource_hub/rp2-1-0-3-lorawan-regional-parameters/
When you enable all 8 channels on the node by default, it might use any channel for join and it will fail the join because the server will ignore it. Also the enabled channels have nothing to do with ADR.

ADR works fine for me with AS923, I start here with DR7 and the server push me back to a lower DR
[image]

What LoRaWAN server are you using?

asupawat, 2022-09-27:
Thanks, I found a problem in the old version of chirpstack-network-server v3.15.0 After upgrading to v3.16.3 ADR works fine in both power and data rates.

closed 2022-09-27 by asupawat