beegee-tokyo/SX126x-Arduino #74: (xQueueGiveFromISR)- assert failed!

paddygoat, 2022-10-09:
I'm using ESP32 WROOM 32D and EBYTE E22-400M30S LoRa module. I've connected everything up on breadboard with standard wiring and getting following error at line 141:
	// Initialize the Radio
	Radio.Init(&RadioEvents);

Error message:

=====================================
SX126x PingPong test
=====================================
MCU Espressif ESP32
BoardId: 00-00-C4-A0-CC-B2-67-AC
Starting lora_hardware_init
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1296 (xQueueGiveFromISR)- assert failed!
abort() was called at PC 0x40085eec on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40085000:0x3ffbeb00 0x40085275:0x3ffbeb20 0x40085eec:0x3ffbeb40 0x40080ef3:0x3ffbeb60 0x40080f3d:0x3ffbeb80 0x40083959:0x3ffbeba0 0x4000bfed:0x3ffb1ea0 0x400870e9:0x3ffb1eb0 0x40080ed2:0x3ffb1ed0 0x400d1aca:0x3ffb1ef0 0x400d3070:0x3ffb1f10 0x400d2088:0x3ffb1f30 0x400d12ee:0x3ffb1f50 0x400d51ba:0x3ffb1fb0 0x40086285:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
=====================================
SX126x PingPong test
=====================================

Code is standard pingpong example:

#include <Arduino.h>

#include <SX126x-Arduino.h>
#include <SPI.h>

hw_config hwConfig;

#ifdef ESP32
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 4;  // LORA RESET
int PIN_LORA_DIO_1 = 21; // LORA DIO_1
int PIN_LORA_BUSY = 22;  // LORA SPI BUSY
int PIN_LORA_NSS = 5;	// LORA SPI CS
int PIN_LORA_SCLK = 18;  // LORA SPI CLK
int PIN_LORA_MISO = 19;  // LORA SPI MISO
int PIN_LORA_MOSI = 23;  // LORA SPI MOSI
int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;	 // LORA ANTENNA RX ENABLE
#endif
#ifdef ESP8266
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 0; // LORA RESET
int PIN_LORA_DIO_1 = 15;  // LORA DIO_1
int PIN_LORA_BUSY = 16;   // LORA SPI BUSY
int PIN_LORA_NSS = 2;   // LORA SPI CS
int PIN_LORA_SCLK = 14;  // LORA SPI CLK
int PIN_LORA_MISO = 12;  // LORA SPI MISO
int PIN_LORA_MOSI = 13;  // LORA SPI MOSI
int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;		 // LORA ANTENNA RX ENABLE
#endif
#ifdef NRF52_SERIES
[... 426 lines cut ...]
		{
			Serial.println("Sending a PONG in OnCadDone as Slave");
#ifdef NRF52_SERIES
			if (bleUARTisConnected)
			{
				bleuart.print("Sending a PONG in OnCadDone as Slave\n");
			}
#endif
			// Send the reply to the PONG string
			TxdBuffer[0] = 'P';
			TxdBuffer[1] = 'O';
			TxdBuffer[2] = 'N';
			TxdBuffer[3] = 'G';
		}
		// We fill the buffer with numbers for the payload
		for (int i = 4; i < BufferSize; i++)
		{
			TxdBuffer[i] = i - 4;
		}

		Radio.Send(TxdBuffer, BufferSize);
	}
}

Any ideas what might be causing this error? Thank you.

beegee-tokyo, 2022-10-10:
I have no idea what could cause the problem. I just tested your code (only changing the SX1262 pin numbers) and it works.
```log
12:30:57.951 --> ets Jun  8 2016 00:22:57
12:30:57.951 --> 
12:30:57.951 --> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:30:57.951 --> configsip: 0, SPIWP:0xee
12:30:57.951 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:30:57.951 --> mode:DIO, clock div:2
12:30:57.951 --> load:0x3fff0030,len:1184
12:30:57.951 --> load:0x40078000,len:12784
12:30:57.951 --> load:0x40080400,len:3032
12:30:57.951 --> entry 0x400805e4
12:30:58.061 --> [¢êmum•ÍÁÍ2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
12:30:58.125 --> =====================================
12:30:58.125 --> SX126x PingPong test
12:30:58.125 --> =====================================
12:30:58.125 --> MCU Espressif ESP32
12:30:58.125 --> BoardId: 00-00-4C-ED-6C-BF-71-3C
12:30:58.125 --> Starting lora_hardware_init
12:30:58.133 --> <BRD> SyncWord = 2414
12:30:58.133 --> <BRD> LoRa Task started
12:30:58.217 --> Starting Radio.Rx
12:30:58.217 --> <RADIO> RX window timeout = 3000
12:34:26.200 --> OnRxTimeout
12:34:26.710 --> CAD returned channel free after 12ms
12:34:26.710 --> Sending a PING in OnCadDone as Master
12:34:26.834 --> OnTxDone
12:34:26.834 --> <RADIO> RX window timeout = 3000
12:35:38.864 --> OnRxTimeout
12:35:39.380 --> OnRxDone
12:35:39.411 --> RssiValue=-17 dBm, SnrValue=14
12:35:39.411 --> 30 31 32 
12:35:39.411 --> <RADIO> RX window timeout = 3000
12:35:39.411 --> CAD returned channel free after 19ms
12:35:39.411 --> Sending a PING in OnCadDone as Master
12:35:39.413 --> OnTxDone
12:35:39.413 --> <RADIO> RX window timeout = 3000
12:35:52.173 --> OnRxTimeout
12:35:52.681 --> OnRxDone
12:35:52.713 --> RssiValue=-17 dBm, SnrValue=14
12:35:52.713 --> 01 74 01 6A 
12:35:52.713 --> <RADIO> RX window timeout = 3000
12:35:52.713 --> CAD returned channel free after 19ms
12:35:52.713 --> Sending a PING in OnCadDone as Master
12:35:52.716 --> OnTxDone
12:35:52.716 --> <RADIO> RX window timeout = 3000
```

Are you using ArduinoIDE or PlatformIO?
Which BSP version?
Which SX1262-Arduino library version?

paddygoat, 2022-10-10:
Are you using ArduinoIDE or PlatformIO?   .... latest Arduino 1.8.x
Which BSP version?
Which SX1262-Arduino library version? ...... current version.

I re-installed Ubuntu and installed latest legacy Arduino and it worked for a while. Then stopped working with the same error message. Then after about 5 minutes started working again, so not sure what it might be. Also, the process for setting up the device is very slow - void setup() takes about 15 seconds. I put serial printout messages for each stage and found that each stage in the radio setup is very slow compared to other devices I have used. Could be a faulty LoRa chip or bad wiring I guess.

beegee-tokyo, 2022-10-10:
Either your eByte module is not powered up or your pin assignment is wrong. Worst case, the eByte module is damaged.

- Permanent timeout when waiting for the Busy signal should not happen. I see that usually 1 or 2 times during power up.

- `SyncWord = 0000` is wrong, it should be `2414`. This means either your SPI connection is wrong or the module is not working.

paddygoat, 2022-10-10:
Looks like it was a wiring error, but it's great to have the de-bugger up and running for when the next Ebyte modules arrive. I broke the current one today - ripped off one of the pads :1st_place_medal:

beegee-tokyo, 2022-10-11:
Ok, I close this for now. Good luck with the eByte modules.

I (logically) prefer the RAKwireless WisBlock system ==> https://docs.rakwireless.com/Product-Categories/WisBlock/
RAK11200 + RAK13300 ==> ESP32 + Semtech SX1262
But no 30dBm TX power, only 22dBm.

closed 2022-10-11 by beegee-tokyo