beegee-tokyo/SX126x-Arduino #76: Change SyncWord

paddygoat, 2022-10-19:
I've got syncword showing as <BRD> SyncWord = 2414, but where is this set and what is easiest way to change it please?
Thank you.

paddygoat, 2022-10-19:
flatus@flatus:~$ grep -r "2414" /home/flatus/Arduino/libraries/SX126x-Arduino-master
/home/flatus/Arduino/libraries/SX126x-Arduino-master/docs/html/search/classes_9.js:  ['sloramaccallback_2414',['sLoRaMacCallback',['../structs_lo_ra_mac_callback.html',1,'']]],
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	// After power on the sync word should be 2414. 4434 could be possible on a restart
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	if ((readSyncWord == 0x2414) || (readSyncWord == 0x4434))
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	// After power on the sync word should be 2414. 4434 could be possible on a restart
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	if ((readSyncWord == 0x2414) || (readSyncWord == 0x4434))
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	// After power on the sync word should be 2414. 4434 could be possible on a restart
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	if ((readSyncWord == 0x2414) || (readSyncWord == 0x4434))
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	// After power on the sync word should be 2414. 4434 could be possible on a restart
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	if ((readSyncWord == 0x2414) || (readSyncWord == 0x4434))
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	// After power on the sync word should be 2414. 4434 could be possible on a restart
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	if ((readSyncWord == 0x2414) || (readSyncWord == 0x4434))
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	// After power on the sync word should be 2414. 4434 could be possible on a restart
/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/boards/mcu/board.cpp:	if ((readSyncWord == 0x2414) || (readSyncWord == 0x4434))
grep: /home/flatus/Arduino/libraries/SX126x-Arduino-master/datasheets/RP-2-1.0.3-LoRaWAN-Regional-Parameters.pdf: binary file matches
grep: /home/flatus/Arduino/libraries/SX126x-Arduino-master/datasheets/SX1261-2_V1.2.pdf: binary file matches
flatus@flatus:~$

beegee-tokyo, 2022-10-20:
The Sync Word is in a register of the SX1262.
At the moment the library supports either Public or Private network sync words. You can switch between them with **`Radio.SetPublicNetwork(bool enable)`**.
But when using LoRaWAN, this is already done when you call **`lmh_init()`** where in the lmh_param_t structure you set either **`LORAWAN_PUBLIC_NETWORK`** or **`LORAWAN_PRIVATE_NETWORK`**.

If you want to change to a different Sync Word than the predefined ones, you need to write directly into the SX1262 registers.
```cpp
uint16_t my_syncword = 0x1234;
// Change LoRa modem SyncWord
SX126xWriteRegister(REG_LR_SYNCWORD, (my_syncword  >> 8) & 0xFF);
SX126xWriteRegister(REG_LR_SYNCWORD + 1, my_syncword  & 0xFF);
```

paddygoat, 2022-10-20:
I tried that already by doing this:
`/home/flatus/Arduino/libraries/SX126x-Arduino-master/src/radio/sx126x/sx126x.h:#define LORA_MAC_PRIVATE_SYNCWORD 0x34
`
Using your suggestion above compiles ok but does not change syncword :(
I think we might be trying to write to the wrong register?

beegee-tokyo, 2022-10-20:
The register is correct, I use the same address to read it out.
```
	SX126xReadRegisters(REG_LR_SYNCWORD, (uint8_t *)&readSyncWord, 2);

	LOG_LIB("BRD", "SyncWord = %04X", readSyncWord);
```
Some commands can only be applied when the radio is in standby status. You might need a ```Radio.Standby()``` before writing the sync word.

paddygoat, 2022-10-20:
Here's the answer:
`Radio.Standby();
  
  // Required SyncWord = 0x1276.
  uint8_t my_syncword_0 = 0x76;
  uint8_t my_syncword_1 = 0x12;
  uint16_t readSyncWord = 0;

  SX126xReadRegisters(REG_LR_SYNCWORD, (uint8_t *)&readSyncWord, 0);
  Serial.print("My SyncWord 0: ");Serial.print("0x");Serial.println(readSyncWord,HEX);
  readSyncWord = 0;
  SX126xReadRegisters(REG_LR_SYNCWORD, (uint8_t *)&readSyncWord, 1);
  Serial.print("My SyncWord 1: ");Serial.print("0x");Serial.println(readSyncWord,HEX);
  readSyncWord = 0;
  SX126xReadRegisters(REG_LR_SYNCWORD, (uint8_t *)&readSyncWord, 2);
  Serial.print("My SyncWord 2: ");Serial.print("0x");Serial.println(readSyncWord,HEX);
  Serial.println();

  SX126xWriteRegister(REG_LR_SYNCWORD, my_syncword_0);
  SX126xWriteRegister(REG_LR_SYNCWORD +1, my_syncword_1);

  readSyncWord = 0;
  SX126xReadRegisters(REG_LR_SYNCWORD, (uint8_t *)&readSyncWord, 0);
  Serial.print("My SyncWord 0: ");Serial.print("0x");Serial.println(readSyncWord,HEX);
  readSyncWord = 0;
  SX126xReadRegisters(REG_LR_SYNCWORD, (uint8_t *)&readSyncWord, 1);
  Serial.print("My SyncWord 1: ");Serial.print("0x");Serial.println(readSyncWord,HEX);
  readSyncWord = 0;
  SX126xReadRegisters(REG_LR_SYNCWORD, (uint8_t *)&readSyncWord, 2);
  Serial.print("My SyncWord 2: ");Serial.print("0x");Serial.println(readSyncWord,HEX);
  `
Printout:
My SyncWord 0: 0x0
My SyncWord 1: 0x14
My SyncWord 2: 0x2414

My SyncWord 0: 0x0
My SyncWord 1: 0x76
My SyncWord 2: 0x1276

beegee-tokyo, 2022-10-21:
So problem is solved.

closed 2022-10-21 by beegee-tokyo