beegee-tokyo/SX126x-Arduino #78: does this library work with "Core1262-868M LoRa Module (Waveshare)" ?

Programmer-Bose, 2022-11-04:
does this library work with "Core1262-868M LoRa Module (Waveshare)"?[https://www.waveshare.com/core1262-868m.htm]
I tried the ping-pong code. did not work.
please help.

beegee-tokyo, 2022-11-05:
It should work, but you have to figure out the correct usage of ANT and DIO1/DIO2 pins.

cambesol, 2022-12-20:
I have the SX126x-Arduino LoRaWAN stack working with a Waveshare core1262-868M board.

beegee-tokyo, 2022-12-21:
Thanks for the feedback. Glad it worked.

mikavdwiel, 2023-07-28:
> I have the SX126x-Arduino LoRaWAN stack working with a Waveshare core1262-868M board.

Do you perhabs have the working setup with this module, i too am using this module, however the PING PONG setup doesnt work as intented. It starts up, but then only shows "Starting Radio.RX" for a while untill it sometimes starts working for a few secunds untill it gets stuck at 

CAD returned channel busy after %ldms\n", duration

beegee-tokyo, 2023-07-29:
@cambesol ☝️ 
Can you share your configuration with @mikavdwiel

cambesol, 2023-07-29:
I can take a look when I return from holiday. It will be later in the week.

mikavdwiel, 2023-08-24:
@cambesol Can you see if you can share the configuration? (ps. Hope you had a good Holiday!)

cambesol, 2023-08-24:
I believe this is what you may need:
`// Microcontroller - SX126x pin configuration
int PIN_LORA_RESET = A5;    /* LORA RESET               */
int PIN_LORA_DIO_1 = 9;     /* LORA DIO_1               */
int PIN_LORA_BUSY  = A1;     /* LORA SPI BUSY            */
int PIN_LORA_NSS   = A2;    /* LORA SPI CS maybe D13?   */
int PIN_LORA_SCLK  = S2;    /* LORA SPI CLK             */
int PIN_LORA_MISO  = S1;    /* LORA SPI MISO            */
int PIN_LORA_MOSI  = S0;    /* LORA SPI MOSI            */
int RADIO_TXEN     = 4;     /* TBD?? LORA ANTENNA TX ENABLE   */
int RADIO_RXEN     = 3;     /* TBD?? LORA ANTENNA RX ENABLE   */`

and this:
`    // Define the HW configuration between MCU and SX126x
    hwConfig.CHIP_TYPE           = SX1262_CHIP;       // Example uses an eByte E22 module with an SX1262
    hwConfig.PIN_LORA_RESET      = PIN_LORA_RESET;    // LORA RESET
    hwConfig.PIN_LORA_NSS        = PIN_LORA_NSS;      // LORA SPI CS
    hwConfig.PIN_LORA_SCLK       = PIN_LORA_SCLK;     // LORA SPI CLK
    hwConfig.PIN_LORA_MISO       = PIN_LORA_MISO;     // LORA SPI MISO
    hwConfig.PIN_LORA_DIO_1      = PIN_LORA_DIO_1;    // LORA DIO_1
    hwConfig.PIN_LORA_BUSY       = PIN_LORA_BUSY;     // LORA SPI BUSY
    hwConfig.PIN_LORA_MOSI       = PIN_LORA_MOSI;     // LORA SPI MOSI
    hwConfig.RADIO_TXEN          = RADIO_TXEN;        // LORA ANTENNA TX ENABLE
    hwConfig.RADIO_RXEN          = RADIO_RXEN;        // LORA ANTENNA RX ENABLE
    hwConfig.USE_DIO2_ANT_SWITCH = true;              // Example uses an CircuitRocks Alora RFM1262 which uses DIO2 pins as antenna control
    hwConfig.USE_DIO3_TCXO       = true;              // Example uses an CircuitRocks Alora RFM1262 which uses DIO3 to control oscillator voltage
    hwConfig.USE_DIO3_ANT_SWITCH = false;             // Only Insight ISP4520 module uses DIO3 as antenna control
    hwConfig.USE_RXEN_ANT_PWR    = false;             // eByte E22 module only.`

The microcontroller setup is specific to my platform which uses a Particle.io P2 module.

beegee-tokyo, 2023-08-25:
@cambesol 

Improved you code insert 😀