beegee-tokyo/SX126x-Arduino #80: No Join Accept rx

cambesol, 2022-12-19:
I am porting this LoRaWAN stack to an Arduino-like platform and I can see the Join Requests being received on an AWS WirelessDevice (using the AWS Network Analyzer). I can also see that the Join Accept is sent and I can see it in the logs of my gateway and captured over the air using an SDR, but the Join Accept packet is not received by the core1262 radio. I've spent a while looking at this problem and am quite familiar with how the radio layer works. I can see pre-amble detect interrupts, but not the IRQ_RX_DONE which I would expect to cause an uplink to complete the OTAA process.

I have confirmed that the DIO2 line from the sx1262 is enabling the RF antenna switch. I can see the interrupts being raised (rising edge) on DIO1. I have written a timer abstraction for the Particle.io framework that this is running on (it appears to work with continuous and one-shot timers calling the registered callbacks).  What is the best way to get support (ideas) to help me get this over the line. It is tantalisingly close, just refusing to receive downlinks. Thank you for your time and expertise!

cambesol, 2022-12-20:
I can confirm that this issue is due to the Join Accept window needing to be tuned for the platform I am running on.
I need to make further measurements but I have successfully joined the network and successfully sent many uplinks to an AWS endpoint. Thanks for all the work you have put into this stack.

beegee-tokyo, 2022-12-21:
Thanks for the feedback. Glad it worked.

closed 2022-12-21 by beegee-tokyo