beegee-tokyo/SX126x-Arduino #89: Bandwidth issue

vladbondarenko, 2023-02-07:
Hello,

How to use 31k bandwidth with your lib?

it says [0: 125 kHz, 1: 250 kHz, 2: 500 kHz, 3: Reserved] but 

SX126x-Arduino/src/radio/sx126x/sx126x.h have 10 of them

typedef enum
{
LORA_BW_500 = 6,
LORA_BW_250 = 5,
LORA_BW_125 = 4,
LORA_BW_062 = 3,
LORA_BW_041 = 10,
LORA_BW_031 = 2,
LORA_BW_020 = 9,
LORA_BW_015 = 1,
LORA_BW_010 = 8,
LORA_BW_007 = 0,
} RadioLoRaBandwidths_t;

125kHz is too much for me, is there any way to use 31k?

Thanks.

vladbondarenko, 2023-02-07:
Hmm, seems it should be just 5-th
const RadioLoRaBandwidths_t Bandwidths[] = {LORA_BW_125, LORA_BW_250, LORA_BW_500, LORA_BW_062, LORA_BW_041, LORA_BW_031, LORA_BW_020, LORA_BW_015, LORA_BW_010, LORA_BW_007};

#define LORA_BANDWIDTH 5

SDR confirmed that bandwidth is ~31k now.

beegee-tokyo, 2023-02-08:
Correct, the docs need some updates. Just have no time.

closed 2023-02-08 by beegee-tokyo