beegee-tokyo/SX126x-Arduino #90: When I try an compile PingPong for ESP32-S3 (LoRa Only) I get errors for REGION definitions.

PaulSchulz, 2023-03-07:
I get the following error (one of several ver similar ones) when compiling PingPong example. 

home/paul/Arduino/libraries/SX126x-Arduino/src/mac/Commissioning.h:40:2: error: #error "SX126x-Arduino V2.0 does support all LoRaWAN regions without definition of 'REGION_XXYYY.\n\nPlease read detailed information how to use it on https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/README_V2.md"
 #error "SX126x-Arduino V2.0 does support all LoRaWAN regions without definition of 'REGION_XXYYY.\n\nPlease read detailed information how to use it on https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/README_V2.md"
  ^~~~~

At this stage I just want to see the LoRa radio working on the Heltec Wifi LoRa 32(V3) board (ESP32-S3 based), before moving onto LoRaWAN. (Can I turn off the Region setting and use the fixed frequency?)

- Where do I set the Hardware Type? (eg. ESP32)

PaulSchulz, 2023-03-07:
I am using the library version 2.0.15 in Arduino IDE.

beegee-tokyo, 2023-03-07:
Which PingPong example exactly?
This error should not come up with the latest example codes.

beegee-tokyo, 2023-03-07:
There is no need to do hardware setup unless the BSP you are using is not declaring that it is an ESP32.

PaulSchulz, 2023-03-07:
Thanks for the message. I will try again tomorrow without any of my edits.

PaulSchulz, 2023-03-07:
This appears to be an issue with hardware selection.. the error doesn't occur if I:
- select hardware fully supported in Arduino IDE: eg.ATMageZero ESP23-S2
- use PingPong from Library examples
(Only a warning about redefinition on LED_BUILTIN)

Is there a particular board definition I should use for comple testing with the example?

The board that I want to use, but gives me the the compilere erros is:
- Heltec WiFi LoRa 32(**V3**), which has it's definitions here: https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.7/package_heltec_esp32_index.json

- The 'Heltec WiFi LoRa 32(V2)' (which uses the SX1276) also produces a similar error.

I don't understand why a 'REGION_' variable would be getting set before the call to  Commissioning.h

Thanks, Paul
 ----- 
The full error is:
In file included from /home/paul/Arduino/libraries/SX126x-Arduino/src/mac/LoRaMacHelper.h:26,
                 from /home/paul/Arduino/libraries/SX126x-Arduino/src/mac/LoRaMacHelper.cpp:22:
/home/paul/Arduino/libraries/SX126x-Arduino/src/mac/Commissioning.h:40:2: error: #error "SX126x-Arduino V2.0 does support all LoRaWAN regions without definition of 'REGION_XXYYY.\n\nPlease read detailed information how to use it on https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/README_V2.md"
 #error "SX126x-Arduino V2.0 does support all LoRaWAN regions without definition of 'REGION_XXYYY.\n\nPlease read detailed information how to use it on https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/README_V2.md"
  ^~~~~
/home/paul/Arduino/libraries/SX126x-Arduino/src/mac/Commissioning.h:48: warning: "REGION_EU868" redefined
 #define REGION_EU868

beegee-tokyo, 2023-03-08:
The **`Heltec WiFi LoRa 32(V2)`** will not work with the library, the SX1272 is not supported.

About the regions:
I checked the link to the Heltec BSP that you are using, it is not compatible with this library. They have their own [LoRaWAN implementation](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/tree/master/esp32/libraries/LoraWan102) in the BSP itself and that's where the conflict is coming from.

Nothing I can do, you have to use their LoRaWAN stack / LoRa P2P drivers.

PaulSchulz, 2023-03-08:
Thank you for looking into this. I have had more success using ESPHome to compile and use your library. (Happy to share my configuration once I have managed to get something to work.)

(I need to manually include 'SPI' and 'Ticker' libraries.)

closed 2023-03-08 by PaulSchulz

PaulSchulz, 2023-03-08:
I have managed to get a LoRa Ping/Pong program to work with your library on the Heltec WiFi LoRa 32 (V3) board. I butchered your example code but it can be found at: https://github.com/PaulSchulz/esphome-esp32s3

In particular: custom_lora.yaml, my_lora_component.h

It can be build and intalled with esphome with: 
  esphome run --device /def/ttyUSB0 custom_lora.yaml

bongo505, 2023-03-28:
I had no issues getting it to run out of the box using the PingPongIO example on a Heltec WSL V3 (no display), but had some really odd issues getting the receive timeout to fire. Sometimes it would run for 5-10 minutes before triggering the timeout.
As soon as I modified it using Paul's call to CadDone inside the loop (using a millis check similar to his) then it starts to work as soon as it is turned on.
Note that I have none of the other Heltec code loaded as it is not yet compatible with the V3 boards. Is there any reason why the Rx timeout is not firing reliably?