beegee-tokyo/SX126x-Arduino #92: Race condition between LoRaMacFlags.Bits.McpsReq test/set/clear causes failure to fire confirmed-transmission callback

avillacis, 2023-04-13:
In one of my projects using LoRaWAN confirmed transmission, I observed the following race condition on an ESP32S2 device:

```
LoRaWAN Task                                                       Arduino Main loop Task
Call to ```OnMacStateCheckTimerEvent``` in ```mac/LoRaMAC.cpp``` 
- Test that LoRaMacState == LORAMAC_IDLE --> TRUE
- Test that LoRaMacFlags.Bits.McpsReq == 1 --> TRUE
- Call MacMcpsConfirm
    - app clears confirmed-tx-pending condition
<TASK SWITCH>                                                       Retrieve next lorawan payload
                                                                    call lmh_send()
                                                                    - call LoRaMacMcpsRequest
                                                                      - sets LoRaMacFlags.Bits.McpsReq = 1
                                                                   <TASK SWITCH>
- returns from confirmed-tx callback
- set LoRaMacFlags.Bits.McpsReq = 0

```

The just-set McpsReq flag (in the main loop task) gets erroneously cleared in the LoRaWAN task, and the fact that a new confirmed-transmission was started gets lost.

I have made a quick-and-dirty fix by copying the flags to a local variable, immediately clearing McpsReq, then testing the local copy of the flags. However this is not a proper fix. A proper fix must involve a critical-section that gets entered and exited on all public methods of ```mac/LoRaMAC.cpp```. This is necessarily board dependent and I do not have other multitask-capable boards to test a proper fix.

avillacis, 2023-04-14:
Please see https://github.com/yubox-node-org/SX126x-Arduino/tree/mitigate-McpsReq-flag-race-condition for the race mitigation I mentioned when opening the bug.

beegee-tokyo, 2023-04-17:
Hello @avillacis 

I will run some tests and when I see no problems, I will merge your change.

Thanks for looking into this and providing a solution.

beegee-tokyo, 2023-04-17:
Merged with #93

Thank @avillacis for finding this.

closed 2023-04-17 by beegee-tokyo