beegee-tokyo/SX126x-Arduino #97: The gateway show SNR=0

gamma2, 2023-05-19:
HI
I have a "heltec wifi lora32 v3" card and a Rakwireless gateway RAK7268V2 (which I use with the built-in network server option) and I'm trying to send Lorawan packets after the OTAA join.
When I use the heltec library with the following platformio project "https://github.com/maxgerhardt/pio-heltec-lorawan-test/tree/main" with the correct configuration since I live in Italy and with the sketch taken from the heltec examples ( 0.0.7\esp32-0.0.7\libraries\Heltec-Example\examples\LoRaWAN\LoRaWan_downlinkdatahandle.ino) all received packets show correct SNR value.
When I use this library with a similar example (SX126x-Arduino-master-bengee\SX126x-Arduino-master\examples\LoRaWan\LoraWan.ino) after each join, after the third/fourth packet received, the gateway shows a SNR = 0 for all subsequent packages.
I wonder if it's something related to the library since this doesn't happen with the heltec one.

Can you give me some pointers?

Tank you.

beegee-tokyo, 2023-05-19:
Hi @gamma2 

If the gateway always shows SNR 0, that has nothing to do with the node. The SNR is not calculated or sent by the node.

If you look into the [RAK Forum](https://forum.rakwireless.com/c/lpwanc2ae/33) you can see others having the same problem.

E.g. this post ==> https://forum.rakwireless.com/t/snr-data-always-0-for-rak7249/8226

closed 2023-05-19 by beegee-tokyo

gamma2, 2023-05-19:
Thanks for the reply.
It would be interesting to understand why this doesn't happen with the heltec library.

beegee-tokyo, 2023-05-19:
I can't see this problem with any of my devices (all using this library) and my RAK gateways, a RAK7258 and a RAK7268-V2.
[image]
[image]