beegee-tokyo/SX126x-Arduino #99: Assistance Needed: error 1

peterverhagenresx, 2023-05-31:


peterverhagenresx, 2023-05-31:
`16:44:50.556 -> ESP-ROM:esp32c3-api1-20210207
16:44:50.588 -> Build:Feb  7 2021
16:44:50.588 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
16:44:50.588 -> SPIWP:0xee
16:44:50.588 -> mode:DIO, clock div:1
16:44:50.588 -> load:0x3fcd5810,len:0x438
16:44:50.588 -> load:0x403cc710,len:0x91c
16:44:50.588 -> load:0x403ce710,len:0x25b0
16:44:50.588 -> entry 0x403cc710
16:44:50.654 -> E (95) esp_core_dump_flash: Core dump data check failed:
16:44:50.654 -> Calculated checks=====================================
16:44:50.654 -> SX126x LoRaWan test
16:44:50.654 -> =====================================
16:44:50.687 -> lora_hardware_init failed - 1`

Using an esp32-c3 board
Does any information exist on what error 1 means?
Ta!
PV

peterverhagenresx, 2023-05-31:
/**
 * @file LoRaWan.ino
 * @author Bernd Giesecke (bernd.giesecke@rakwireless.com)
 * @brief 
 * @version 0.1
 * @date 2021-11-26
 * 
 * @copyright Copyright (c) 2021
 * 
 */

#include <Arduino.h>

#include <LoRaWan-Arduino.h>

#include <SPI.h>

#define SCHED_MAX_EVENT_DATA_SIZE APP_TIMER_SCHED_EVENT_DATA_SIZE /**< Maximum size of scheduler events. */
#define SCHED_QUEUE_SIZE 60										  /**< Maximum number of events in the scheduler queue. */

#define LORAWAN_APP_DATA_BUFF_SIZE 64  /**< Size of the data to be transmitted. */
#define LORAWAN_APP_TX_DUTYCYCLE 10000 /**< Defines the application data transmission duty cycle. 10s, value in [ms]. */
#define APP_TX_DUTYCYCLE_RND 1000	   /**< Defines a random delay for application data transmission duty cycle. 1s, value in [ms]. */
#define JOINREQ_NBTRIALS 3			   /**< Number of trials for the join request. */

hw_config hwConfig;

#ifdef ESP32
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 10;	 // LORA RESET
int PIN_LORA_NSS = 7;	 // LORA SPI CS
int PIN_LORA_SCLK = 4;	 // LORA SPI CLK
int PIN_LORA_MISO = 5;	 // LORA SPI MISO
int PIN_LORA_DIO_1 = 3; // LORA DIO_1
int PIN_LORA_BUSY = 0;	 // LORA SPI BUSY
int PIN_LORA_MOSI = 6;	 // LORA SPI MOSI
int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;	 // LORA ANTENNA RX ENABLE
#endif
#ifdef ESP8266
// ESP32 - SX126x pin configuration
int PIN_LORA_RESET = 0;	  // LORA RESET
int PIN_LORA_NSS = 2;	  // LORA SPI CS
int PIN_LORA_DIO_1 = 15;  // LORA DIO_1
int PIN_LORA_BUSY = 16;	  // LORA SPI BUSY
int PIN_LORA_SCLK = SCK;  // LORA SPI CLK
int PIN_LORA_MISO = MISO; // LORA SPI MISO
int PIN_LORA_MOSI = MOSI; // LORA SPI MOSI
int RADIO_TXEN = -1;	  // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;	  // LORA ANTENNA RX ENABLE
#endif
#ifdef NRF52_SERIES
#ifndef RAK4630
// nRF52832 - SX126x pin configuration
int PIN_LORA_RESET = 4;	 // LORA RESET
int PIN_LORA_NSS = 28;	 // LORA SPI CS
int PIN_LORA_SCLK = 12;	 // LORA SPI CLK
int PIN_LORA_MISO = 14;	 // LORA SPI MISO
int PIN_LORA_DIO_1 = 11; // LORA DIO_1
int PIN_LORA_BUSY = 29;	 // LORA SPI BUSY
int PIN_LORA_MOSI = 13;	 // LORA SPI MOSI
int RADIO_TXEN = -1;	 // LORA ANTENNA TX ENABLE
int RADIO_RXEN = -1;	 // LORA ANTENNA RX ENABLE
// Replace PIN_SPI_MISO, PIN_SPI_SCK, PIN_SPI_MOSI with your
SPIClass SPI_LORA(NRF_SPIM2, 14, 12, 13);
#endif
#endif

// Foward declaration
static void lorawan_has_joined_handler(void);
[... 262 lines cut ...]

	lmh_error_status error = lmh_send(&m_lora_app_data, LMH_UNCONFIRMED_MSG);
	Serial.printf("lmh_send result %d\n", error);
}

/**@brief Function for handling a LoRa tx timer timeout event.
*/
static void tx_lora_periodic_handler(void)
{
	TimerSetValue(&appTimer, LORAWAN_APP_TX_DUTYCYCLE);
	TimerStart(&appTimer);
	Serial.println("Sending frame");
	send_lora_frame();
}

/**@brief Function for the Timer initialization.

   @details Initializes the timer module. This creates and starts application timers.
*/
static uint32_t timers_init(void)
{
	appTimer.timerNum = 3;
	TimerInit(&appTimer, tx_lora_periodic_handler);
	return 0;
}

beegee-tokyo, 2023-05-31:
Error code 1 means simply the SX1262 transceiver was not found. 
Check your SPI and GPIO pin assignments. If you have a "flying wire" connection, check your wiring.

During initialization of the SPI communication to the SX1262 the function tries to read the syncword from the chip.
If a wrong value was returned (not 0x2414 or 0x4434) it means the communication with the SX1262 does not work.

peterverhagenresx, 2023-05-31:
Sorry for the lack of info earlier - I was running out the door.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/_images/esp32-c3-devkitc-02-v1-pinout.png
Our esp32 is the ESP32-C3-DevKitC-02_v1 from espressif;
and LoRaWAN Module is https://core-electronics.com.au/sx1262-lora-node-module-for-raspberry-pi-pico-lorawan.html

Pin out has been customised for the LoRaWAN module to line up with how we would expect the esp32 to communicate - pins outlined in the code above.

We have checked the wiring configuration a few times, there is only one ground connected to the LoRaWAN Module - would this be a factor where we need each pin on the board to be wired to ground?

Thanks
PV

beegee-tokyo, 2023-05-31:
One GND connection should be good enough.

I have no experience with the single core version of the ESP32. From the link you sent I cannot see whether the SPI bus is on the pins you selected. They are marked as JTAG pins, analog pins or generic GPIO's. However I can see they are listed as SPI pins in the pins_arduino.h for the esp32C3 board (assuming you selected that board in the Arduino IDE).

One thing you can try to get more information is to enable the libraries debug output by changing to _**`#define LIB_DEBUG 1`**_ in the [sx126x-debug.h](https://github.com/beegee-tokyo/SX126x-Arduino/blob/7be4c45af8b07d014f6d6d4551fbc8e180a07403/src/sx126x-debug.h#L18) file.

peterverhagenresx, 2023-05-31:
Have just turned on the LIB_DEBUG as discussed

`09:24:13.725 -> ESP-ROM:esp32c3-api1-20210207
09:24:13.725 -> Build:Feb  7 2021
09:24:13.725 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
09:24:13.725 -> SPIWP:0xee
09:24:13.725 -> mode:DIO, clock div:1
09:24:13.725 -> load:0x3fcd5810,len:0x438
09:24:13.725 -> load:0x403cc710,len:0x91c
09:24:13.725 -> load:0x403ce710,len:0x25b0
09:24:13.725 -> entry 0x403cc710
09:24:13.822 -> E (95) esp_core_dump_flash: Core dump data check failed:
09:24:13.822 -> Calculated checks=====================================
09:24:13.822 -> SX126x LoRaWan test
09:24:13.822 -> =====================================
09:24:13.854 -> lora_hardware_init failed - 1
09:28:19.144 -> ESP-ROM:esp32c3-api1-20210207
09:28:19.144 -> Build:Feb  7 2021
09:28:19.144 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
09:28:19.144 -> SPIWP:0xee
09:28:19.144 -> mode:DIO, clock div:1
09:28:19.144 -> load:0x3fcd5810,len:0x438
09:28:19.144 -> load:0x403cc710,len:0x91c
09:28:19.144 -> load:0x403ce710,len:0x25b0
09:28:19.144 -> entry 0x403cc710
09:28:19.187 -> E (96) esp_core_dump_flash: Core dump data check failed:
09:28:19.187 -> Calculated checks=====================================
09:28:19.187 -> SX126x LoRaWan test
09:28:19.187 -> =====================================
09:28:19.219 -> <BRD> SyncWord = 0000
09:28:19.219 -> lora_hardware_init failed - 1
09:28:19.219 -> <LMH> OTAA 
09:28:19.219 -> DevEui=00-95-64-1F-DA-91-19-0B
09:28:19.219 -> DevAdd=260116F8
09:28:19.219 -> AppEui=70-B3-D5-7E-D0-02-01-E1
09:28:19.219 -> AppKey=07-C0-82-0C-30-B9-08-70-0C-0F-70-06-00-B0-BE-09
09:28:19.251 -> <LMH> Selected subband 2
09:28:19.251 -> <LMH> Selected subband 1`

beegee-tokyo, 2023-06-01:
This one ==> _**`<BRD> SyncWord = 0000`**_ tells me that your SPI connection is not working. 
The sync word should be 0x2414 or 0x4434. If it reads 0x0000 that means the SPI is not connected to the transceiver.

You have to check your connections and your SPI settings for your specific board.

peterverhagenresx, 2023-06-01:
We're gonna try a new ESP32 module - potentially this one doesn't do SPI the way we think it does.
I ran out a script the other day for the board to tell me it's default ports, and they line up with what we have been using; so it's all very odd to me.
Do appreciate the help so far none the less!

closed 2023-06-12 by beegee-tokyo