beegee-tokyo/SX126x-Arduino #101: Set TX Power to high values.

gamma2, 2023-06-11:
Good morning, I'm doing some tests with my heltec wifi lora32 v3. How can I set the transmission power to max values (20/22) db?
I try to set it with the following statement
static lmh_param_t lora_param_init = {LORAWAN_ADR_OFF, DR_0, LORAWAN_PUBLIC_NETWORK, JOINREQ_NBTRIALS, 22, LORAWAN_DUTYCYCLE_OFF};
but it seems to have no effect.
Maybe it's not within the spec of EU868 and if so, how can I force it?

beegee-tokyo, 2023-06-11:
Hi gamma2,

Two things.

1) The LoRaWAN stack doesn't set the TX power in dBm, it uses the values 0 to 10 with 0 being the highest TX power. 
2) The LoRaWAN stack does translate the values of 0 to 10 to the allowed TX power per region.

So 
a) for the highest TX power, the value in the lmh_param_t structure is _**0**_
b) you cannot set a TX power of 22dBm when using EU868. It is limited to _**16dBm**_.

Reference [TX power per region](https://docs.rakwireless.com/RUI3/Appendix/#tx-power-by-region) which is extracted from the regional LoRaWAN parameters issued by the LoRa Alliance.

gamma2, 2023-06-12:
Thanks for the clarification.

closed 2023-06-12 by beegee-tokyo