beegee-tokyo/SX126x-Arduino #102: Range Test

gamma2, 2023-06-12:
Good morning,
I'm doing range tests with my heltec wifi lora32 v3 card and the lowest RSSI I can get is -115dBm  (BW_L = 125 kHz, SF = 12) before the communication stops.
The SX1262 specifications give a sensitivity of -133dBm (BW_L = 125 kHz, SF = 12).
Why can't I reach -133dBm?
What tests could I do to figure out what the problem is?

beegee-tokyo, 2023-06-12:
I can confirm that with a different module (RAK4631 = nRF52 + SX1262) the lowest RSSI level I can get reception is -115.

I checked the datasheet as well and the -133dBm seems to be a special test case:
[image]

Reading just above this, it seems the high sensitivity is linked to the Rx Boost, which is not enabled in LoRaWAN and about which we talked [here](https://github.com/beegee-tokyo/SX126x-Arduino/issues/100)

Testing in LoRa P2P and using **`Radio.RxBoosted()`** instead of **`Radio.Rx()`** I can get an improvement to -117dBm, which somehow matches with the comment in the Semtech Radio drivers:
```cpp
SX126xWriteRegister(REG_RX_GAIN, 0x96); // max LNA gain, increase current by ~2mA for around ~3dB in sensivity
```
But it costs you 2mA consumption.

gamma2, 2023-06-13:
Thank you for confirming my results.
However, SX1262 chip spec gives -137dBm in Rx Boosted gail

[image]

beegee-tokyo, 2023-06-13:
I guess you have to ask Semtech about this.
I cannot answer why this cannot be achieved on our devices.

gamma2, 2023-06-13:
Thanks, I'll ask Semtech.

gamma2, 2023-06-14:
Hey beegee-tokyo, this is this is the response received on the Semtech forum from Stuart:

--------------------------------------------------------------------------------------------------------
Well first you should consider where the sensitivity figure of -133dBm was obtained.

If the -133dBm was the tested sensitivty in a laboratory faraday shield then its unlikely you will get the same sensitivity figure outside in the real world where the background RF noise level can be quite high.

So measure the background RF noise level and reception at SF12 should work at around -20dBm below that.

And consider that a processor such as a ESP32 can produce quite a lot of local EMI which will affect the sensitivity of the LoRa device.
-------------------------------------------------------------------------------------------------------------

Which somehow justifies the value of -115dBm that we can reach.

beegee-tokyo, 2023-06-15:
Hi @gamma2 

I was expecting an answer like this. SPecial test environment, far from real world applications.

I will close this issue with the answer from Semtech.

closed 2023-06-15 by beegee-tokyo