beegee-tokyo/SX126x-Arduino #103: Why is the lowest bandwidth supported 125khz?

factoidforrest, 2023-08-06:
The documentation for the sx1262 says that it can handle something as low as 7.2khz ( wish I could site that now but I did read it recently) and the calculator online shows a doubled range between 125khz and 10khz bandwidth, claiming a theoretical 14km with the maximum spreading factor and minimum bandwidth. 

Any reason you don't allow a setting down around 10khz?

beegee-tokyo, 2023-08-06:
You can set any bandwidth in LoRa P2P that the SX1262 supports.
[const RadioLoRaBandwidths_t Bandwidths[]](https://github.com/beegee-tokyo/SX126x-Arduino/blob/7be4c45af8b07d014f6d6d4551fbc8e180a07403/src/radio/sx126x/radio.cpp#L440)
```cpp
const RadioLoRaBandwidths_t Bandwidths[] = {LORA_BW_125, LORA_BW_250, LORA_BW_500, LORA_BW_062, LORA_BW_041, LORA_BW_031, LORA_BW_020, LORA_BW_015, LORA_BW_010, LORA_BW_007};
```

The additional bandwidths were added in June 2021 but never made it into the documentation.

closed 2023-08-06 by beegee-tokyo

beegee-tokyo, 2023-08-06:
Added to the README ==> [LoRa Definitions](https://github.com/beegee-tokyo/SX126x-Arduino#lora-definitions)