beegee-tokyo/SX126x-Arduino #106: Arduino IDE PingPong Example

matteodallefeste, 2023-09-16:
Goodmorning.
I'm using your library for testing Lora devices with esp32.
Compiling the PingPong Example with the current release give me an error:

In file included from C:\Users\xxxx\Documents\Arduino\libraries\SX126x-Arduino\src\mac\LoRaMacHelper.h:26,
                 from C:\Users\xxxx\Documents\Arduino\libraries\SX126x-Arduino\src\mac\LoRaMacHelper.cpp:22:
C:\Users\xxxx\Documents\Arduino\libraries\SX126x-Arduino\src/mac/Commissioning.h:40:2: error: #error "**SX126x-Arduino V2.0 does support all LoRaWAN regions without definition of 'REGION_XXYYY**.\n\nPlease read detailed information how to use it on https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/README_V2.md"
 #error "SX126x-Arduino V2.0 does support all LoRaWAN regions without definition of 'REGION_XXYYY.\n\nPlease read detailed information how to use it on https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/README_V2.md"

But the ping Pong Example it's only a Lora and not LoraWAN.
Am i wrong in something or there's an issue?

Thanks
Matteo

beegee-tokyo, 2023-09-17:
What version of the SX126x library are you using?
There was a major change between version 1.x.x and 2.x.x 

Make sure you use both the library and example code from V2.x.x

matteodallefeste, 2023-09-17:
i just downloaded the latest version  [2.0.20] with Arduino IDE.
If i try to compile the example included in the library package called "PingPong" the compiler return me this error: 

*SX126x-Arduino V2.0 does support all LoRaWAN regions without definition of 'REGION_XXYYY*

But this example is not running LoraWan but only Lora
if i comment the Commissioning.h check al line 39-41 the compiler seem to run properly.

may be useful a conditional check for exclude the code of line 39-41 if we are in Lora mode and not in LoraWan mode?

thanks for your time Bernd!

matteodallefeste, 2023-09-17:
Probably it's the same problem that you reported on NOTE 3 on the file README_V2.md
Downloading the esp32 package i find many devices with the Lora module.
in these modules it's present by default the Region selection.
So in the build process it include automatically the Region info
[image]

[image]

beegee-tokyo, 2023-09-17:
That is not the problem here. The problem is the BSP for the Heltec Stick. It has LoRaWAN regions definitions that do not work with the SX126x-Arduino library V2.x.x

Install the SX126x-Arduino library V1.3.3 and it should work again (I guess).
Otherwise contact Heltec to update their BSP to support the latest version of my library.

[image]

matteodallefeste, 2023-09-17:
ok thanks Bernd!
Hava a great day

matteodallefeste, 2023-09-17:
Issue is caused by BSP and not from the library. May be useful a "Note 5" in the ReadmeV2 documentation

closed 2023-09-17 by matteodallefeste